zephyrproject-rtos/zephyr #64333: picolibc: FPU issue on x86 with timer_behavior test

teburd, 2023-10-24:
**Describe the bug**
@golowanow noted that a test case (timer_behavior) was failing on an x86 hardware platform with an FPU exception.

Re-running this test on qemu_x86 shows the same exception.

**To Reproduce**
Comment out the platform_type lines in tests/kernel/timer/timer_behavior/testcase.yaml

Run west
```west twister -K -p qemu_x86 -T tests/kernel/timer/timer_behavior -vvv```

**Expected behavior**
A non failing test case, or at least one that doesn't cause a hardware exception over FPU usage

**Logs and console output**
```
DEBUG   - run test: qemu_x86/tests/kernel/timer/timer_behavior/kernel.timer.timer
DEBUG   - Spawning QEMUHandler Thread for qemu_x86/tests/kernel/timer/timer_behavior/kernel.timer.timer
DEBUG   - Running qemu_x86/tests/kernel/timer/timer_behavior/kernel.timer.timer (qemu)
DEBUG   - Spawning QEMUHandler Thread for qemu_x86/tests/kernel/timer/timer_behavior/kernel.timer.timer
DEBUG   - QEMU (136691): SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
DEBUG   - QEMU (136691): Booting from ROM..
DEBUG   - QEMU (136691): *** Booting Zephyr OS build zephyr-v3.4.0-3252-gf0daf904bb02 ***
DEBUG   - QEMU (136691): Running TESTSUITE timer_jitter_drift
DEBUG   - QEMU (136691): ===================================================================
DEBUG   - QEMU (136691): START - test_jitter_drift_timer_period
DEBUG   - QEMU (136691): periodic timer behavior test using built-in restart mechanism
DEBUG   - QEMU (136691): collecting time samples for approx 100 seconds
DEBUG   - QEMU (136691): periodic timer samples gathered, calculating statistics
DEBUG   - QEMU (136691): E: Floating point unit device not available
DEBUG   - QEMU (136691): E: EAX: 0x05f4732e, EBX: 0x019e7c6d, ECX: 0x00000000, EDX: 0x02fa3997
DEBUG   - QEMU (136691): E: ESI: 0x033cf8d9, EDI: 0x033cf8da, EBP: 0x001286e4, ESP: 0x001286d4
DEBUG   - QEMU (136691): E: EFLAGS: 0x00000206 CS: 0x0008 CR3: 0x00138000
DEBUG   - QEMU (136691): E: call trace:
DEBUG   - QEMU (136691): E: EIP: 0x0010c173
DEBUG   - QEMU (136691): E:      0x0010c0a1 (0x7d000000)
DEBUG   - QEMU (136691): E:      0x0010335f (0x3827c5ac)
DEBUG   - QEMU (136691): E:      0x00105fcc (0x1043d8)
DEBUG   - QEMU (136691): E:      0x001043fa (0x10d088)
DEBUG   - QEMU (136691): E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
DEBUG   - QEMU (136691): E: Current thread: 0x124ac0 (test_jitter_drift_timer_period)
DEBUG   - QEMU (136691): E: Halting system
DEBUG   - QEMU (136691) complete (unexpected eof) after 0.5647683143615723 seconds
```

**Additional context**
Bisecting found this commit f0daf904bb0202c9247cdcf4d1186e0762a9db67 to be the culprit

golowanow, 2023-10-24:
FPU exception at this point in the test:

https://github.com/zephyrproject-rtos/zephyr/blob/4974bc01370d779ad67c082491d4ceecb47b20b8/tests/kernel/timer/timer_behavior/src/jitter_drift.c#L212

nashif, 2023-10-24:
CONFIG_FPU needed now?

ceolin, 2023-10-24:
> > CONFIG_FPU needed now?
> 
> it works for qemu_x86 and to fix this particular test, at least.
> 
> I wonder might it be a general problem with picolibc now as default libc and where config options have not been set explicitly like in this test case.

This test used to work with soft floating point ... Wondering if that is no longer supported

keith-packard, 2023-10-24:
> This test used to work with soft floating point ... Wondering if that is no longer supported

It sounds like the soft float bits aren't getting used in this case, which sounds like a toolchain configuration issue, or perhaps a problem with multilib aliasing for x86?

keith-packard, 2023-10-24:
Nope, just an old version of picolibc that didn't handle x86 soft float (because who even does that in 2023?). It's all fixed in the current bits which are part of pr #62882

nashif, 2023-10-24:
> Nope, just an old version of picolibc that didn't handle x86 soft float (because who even does that in 2023?). It's all fixed in the current bits which are part of pr #62882 and the related SDK pr [zephyrproject-rtos/sdk-ng#707](https://github.com/zephyrproject-rtos/sdk-ng/pull/707) (which has already been merged).

I thought we were testing sqrtf and friends in `tests/lib/c_lib/` and it is passing with the current toolchain... Are doing something different there?

golowanow, 2023-10-25:
> I thought we were testing sqrtf and friends in `tests/lib/c_lib/` and it is passing with the current toolchain... Are doing something different there?

sqrt() test is just skipped without CONFIG_FPU

https://github.com/zephyrproject-rtos/zephyr/blob/88b05b3b91ce3e7491c54544a083dd26ce2aff98/tests/lib/c_lib/src/test_sqrt.c#L178-L183

keith-packard, 2023-10-25:
> sqrt() test is just skipped without CONFIG_FPU

Fixed in #63443

PR #62882 (2023-11-20): Picolibc 1.8.5
Merge in picolibc version 1.8.5.

There is a matching SDK PR pending, https://github.com/zephyrproject-rtos/sdk-ng/pull/707

This offers two new printf variants, 'long long', which provides an integer-only mode with long long support included and a 'minimal' variant which cuts out a bunch of format modifier support to make things very small.

hello-world on cortex-m3 FLASH sizes:
```
float		13378
float inexact	10666
long long	 7950
integer		 7798
minimal		 6834
```
For comparison, here are the minimal C lib sizes:
```
float		11082
long long	 9194
integer		 8426
nano		 7386
```
W [... cut]
closed 2023-11-20 by cfriedt