zephyrproject-rtos/zephyr #64344: TINYCRYPT kconfig bug caused BT tests to fail

aescolar, 2023-10-25:
**Describe the bug**
In 713ca15224c3223e99bf045b1ab50a6917bf60ea , the TINYCRYPT kconfiguration was changed, to depend on a ZEPHYR_TINYCRYPT_MODULE which nobody sets. Which causes all builds using TINYCRYPT to produce a warning, and therefore those tests to fail.

**To Reproduce**
Steps to reproduce the behavior:
1. twister -T  tests/bluetooth/audio/ascs/
2. See error

**Expected behavior**
No kconfig warnings and test failures

**Impact**
Blocked development in main branch and specially around BT development.

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/actions/runs/6625055972/job/17995280341

**Environment (please complete the following information):**
 - OS: CI / Ubuntu 22.04
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: Any after 713ca15224c3223e99bf045b1ab50a6917bf60ea

**Additional context**
Introduced by 713ca15224c3223e99bf045b1ab50a6917bf60ea

aescolar, 2023-10-25:
CC @ceolin @Thalley @jhedberg

ppryga-nordic, 2023-10-25:
The issue blocks: https://github.com/zephyrproject-rtos/zephyr/pull/61468 and other Bluetooth subsystem PRs that are pending.

closed 2023-10-25 by carlescufi