zephyrproject-rtos/zephyr #64352: Raspberry Pi Pico: Pull Down Resistors on ADC PINs cause low input impedance

psimonte, 2023-10-25:
Here's a revised version of your text with corrected mistakes:

"When using ADC inputs with the Raspberry Pi Pico, these pins have an internal pull-down resistor of around 50 kOhm activated after a reset. This could lead to a voltage drop when measuring resistors near that value (e.g., 24 kOhm).

Deactivating the pull-up/pull-down resistor fixes that problem:

#include <hardware/gpio.h>
gpio_set_pulls(26, false, false);

I initially thought I could simply add "bias-disable" in the rpi_pico-pinctrl.dtsi file to achieve this. However, it seems that soc/arm/rpi_pico/rp2/pinctrl_soc.h does not process "bias-disable," as it assumes that this is the default state for every pin."

garw, 2024-02-03:
This issue seems to have been fixed with commit 2e848d8.

closed 2024-04-19 by github-actions[bot]

jimspruell-gm, 2024-09-18:
how can I deactivate the ADC pull down resistors in micropython?