zephyrproject-rtos/zephyr #64375: sample: mcumgr: does not work with ipv4

jeppenodgaard, 2023-10-25:
**Describe the bug**
Target crash when an image is transferred with `mcumgr image upload` when ipv4 only is used.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect the target to a ethernet port on your host PC and configure the port manually
1. Build and flash sample without ipv6:
```shell
west build -b nucleo_h563zi --sysbuild samples/subsys/mgmt/mcumgr/smp_svr/ -- -DSB_CONFIG_BOOTLOADER_MCUBOOT=y -DEXTRA_CONF_FILE=overlay-udp.conf -DCONFIG_MCUMGR_TRANSPORT_UDP_IPV6=n; west flash -r stm32cubeprogrammer --no-erase
```
```shell
west build -b mimxrt1050_evk --sysbuild samples/subsys/mgmt/mcumgr/smp_svr/ -- -DSB_CONFIG_BOOTLOADER_MCUBOOT=y -DEXTRA_CONF_FILE=overlay-udp.conf -DCONFIG_MCUMGR_TRANSPORT_UDP_IPV6=n; west flash
```
1. Transfer an image with mcumgr client:
```shell
mcumgr --conntype udp --connstring <ip_address>:1337 image upload build/smp_svr/zephyr/zephyr.signed.bin
```

**Expected behavior**

- MCUMGR_TRANSPORT_NETBUF_USER_DATA_SIZE has a default value if ipv4 is chosen, therefore custom configuration seems illogical.
- Also I would not expect a fatal error, but rather the mcumgr client to exit with and error and the terminal on the target to print an error.

**Logs and console output**
```shell
*** Booting Zephyr OS build zephyr-v3.5.0-2-g89fd2ae107b6 ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Image index: 0, Swap type: none
I: Bootloader chainload address offset: 0x10000
I: Jumping to the first image slot
*** Booting Zephyr OS build zephyr-v3.5.0-2-g89fd2ae107b6 ***
[00:00:00.001,000] <inf> smp_udp: Network disconnected
[00:00:02.001,000] <inf> smp_udp: Network connected
[00:00:02.002,000] <inf> smp_sample: build time: Oct 25 2023 15:11:33
[00:00:02.002,000] <inf> smp_udp: Started (IPv4)
[00:00:08.321,000] <inf> mcumgr_img_grp: Erased 0x1e000 bytes of image slot
[00:00:08.323,000] <inf> mcumgr_img_grp: Erased 0x2000 bytes of image slot trailer
[00:00:08.326,000] <err> os: ***** BUS FAULT *****
[00:00:08.326,000] <err> os:   Precise data bus error
[00:00:08.326,000] <err> os:   BFAR Address: 0x75516a60
[00:00:08.326,000] <err> os: r0/a1:  0x00000001  r1/a2:  0x000000bf  r2/a3:  0x0000001f
[00:00:08.326,000] <err> os: r3/a4:  0x75516a60 r12/ip:  0x00000000 r14/lr:  0x08014713
[00:00:08.326,000] <err> os:  xpsr:  0x01000000
[00:00:08.326,000] <err> os: Faulting instruction address (r15/pc): 0x0802770c
[00:00:08.326,000] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:00:08.326,000] <err> os: Current thread: 0x20001fa8 (unknown)
[00:00:08.404,000] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.3
 - Commit SHA used: d4037ce6a10c14dd3d27e10e825b8a82f98150fe

jeppenodgaard, 2023-10-25:
Increasing `MCUMGR_TRANSPORT_NETBUF_USER_DATA_SIZE` to 24 (the same value used by IPV6) fixes the issue.

thedjnK, 2023-10-25:
Do you have `CONFIG_IPV6` enabled even if MCUmgr is not using IPv6?

jeppenodgaard, 2023-10-25:
> Do you have `CONFIG_IPV6` enabled even if MCUmgr is not using IPv6?

CONFIG_NET_IPV6 is enabled. I guess that's the one you are looking for?

thedjnK, 2023-10-25:
Yes, that'd be why, the data struct would then be 24 even for IPv4 connections whereas without it 8 should be sufficient

PR #64412 (2023-10-30): mgmt: mcumgr: transport: Fix UDP user data buffer overflow
Fixes a buffer overflow issue if UDP is enabled for IPv4 only but IPv6 networking is enabled

Fixes #64375
closed 2023-10-30 by MaureenHelm