zephyrproject-rtos/zephyr #64379: Net: Hardfault when combining offloaded and non offloaded network interfaces

matt-wood-ct, 2023-10-25:
**Describe the bug**
When using multiple network interfaces some of which use `CONFIG_NET_OFFLOAD`, non offloaded interfaces cause a hardfault when attempting to open TCP connections using posix sockets (`connect` specifically). The network interfaces in my case are the eth_mcux (NXP's ethernet driver) and esp_at (the at command based wifi driver for espressif's ESP-AT firmware).

the function `net_if_is_ip_offloaded` is passed a NULL iface value which causes the code inside the function:
`return (iface->if_dev->offload != NULL);`
To attempt to accesss a member of a NULL pointer which triggers a fault.

In the fault callstack when `net_context_recv` is called NULL is passed into `net_if_is_ip_offloaded` because its argument is provided by `net_context_get_iface` which returns NULL when the specified network context has an interface value of -1 for non offloaded contexts at this stage in the posix transaction (I do not know if this is incorrect behaviour).

My current work around is to introduce NULL checks inside `net_if_is_ip_offloaded` as below:
```
if(NULL == iface) return false;
if(NULL == iface->if_dev) return false;
return (iface->if_dev->offload != NULL);
```

**To Reproduce**
As far as I am aware this bug is not possible to encounter using any of the supported board without using an overaly as none of the supported board include multiple network interfaces of differing types. In my case I am using custom hardware with a custom board file. So for this reason it is not practical to reproduce as you would need to create a setup with an NXP ethernet and some kind of offloaded interface which requires the `CONFIG_NET_OFFLOAD` config.

**Expected behavior**
POSIX connect to return success, no hard fault.

**Impact**
Preventing usage of virgin zephyr, I am currently relying on local file edits to make the firmware function, I am aware of no out of tree solution.

**Logs and console output**

Code used to produce this log attempts to perform a posix socket connect operation to a server after ~10 seconds.

```
[00:00:01.349,000] <inf> COR_NET_MAIN: Interface 'Ethernet' is up
[00:00:01.349,000] <inf> COR_NET_MAIN: Interface 'Wi-Fi' is up
Connected
[00:00:03.504,000] <inf> eth_mcux: ETH_0 enabled 10M full-duplex mode.
[00:00:03.510,000] <inf> net_dhcpv4: Received: 10.0.1.132
[00:00:03.510,000] <inf> net_config: IPv4 address: 10.0.1.132
[00:00:03.510,000] <inf> net_config: Lease time: 172800 seconds
[00:00:03.510,000] <inf> net_config: Subnet: 255.255.254.0
[00:00:03.511,000] <inf> net_config: Router: 10.0.0.1
[00:00:03.658,000] <inf> net_config: IPv4 address: 10.0.1.129
[00:00:03.658,000] <inf> net_config: Lease time: 0 seconds
[00:00:03.658,000] <inf> net_config: Subnet: 255.255.254.0
[00:00:03.658,000] <inf> net_config: Router: 10.0.0.1
[00:00:11.350,000] <err> os: ***** BUS FAULT *****
[00:00:11.350,000] <err> os:   Precise data bus error
[00:00:11.350,000] <err> os:   BFAR Address: 0xb083b548
[00:00:11.350,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x00000001  r2/a3:  0x80014938
[00:00:11.350,000] <err> os: r3/a4:  0xb083b530 r12/ip:  0x80014128 r14/lr:  0x60067bbb
[00:00:11.350,000] <err> os:  xpsr:  0x610f0000
[00:00:11.350,000] <err> os: Faulting instruction address (r15/pc): 0x60068284
[00:00:11.350,000] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:00:11.350,000] <err> os: Current thread: 0x80014938 (unknown)
[00:00:11.351,000] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK 0.16.3
 - Commit SHA or Version used: zephyr-v3.4.0 (356c8cbe63ae01b3ab438382639d25bb418a0213)

jukkar, 2023-10-25:
This is fixed by commit 32bd3aa781523009332f7e102fdd555cf740acdc in 3.5

jukkar, 2023-10-25:
The relevant PR is #61215
I added there the backport label.

matt-wood-ct, 2023-10-25:
Awesome, thanks guys, I feel a bit silly now wasting time writing up this whole thing.
Thanks for adding the backport request, but I will do my best to migrate to 3.5 anyhow.