zephyrproject-rtos/zephyr #64386: Improve usability of init_priorities.py output

kartben, 2023-10-25:
**Is your enhancement proposal related to a problem? Please describe.**

Especially since the init_priorities check stuff is so new, I think special care should be taken about making sure the logs and errors it spits out are easy to understand and actionable.

In the error message below, the ` < ` doesn't really stand out IMO and does not immediately read as "less than". I think spelling things out better would also be more aligned with the kind of error message/warning one would expect from a static analysis tool.

``ERROR: /soc/i2c@40005400/test-i2c-device@20 POST_KERNEL 1 < /soc/i2c@40005400 POST_KERNEL 4``

What's more, it feels like it could be better at indicating this is an error related to init_levels, i.e add `%(module)s` or `%(filename)s` to the format string?

**Describe the solution you'd like**

With a minimal change, just enough to make it easier to spot the issue is regarding init priorities

``ERROR: [check_init_priorities] /soc/i2c@40005400/test-i2c-device@20 POST_KERNEL 1 < /soc/i2c@40005400 POST_KERNEL 4`` 

Or, more verbose:

``ERROR: [check_init_priorities] /soc/i2c@40005400/test-i2c-device@20 has lower init priority than /soc/i2c@40005400 (POST_KERNEL 1 < POST_KERNEL 4)``  

Of course == and >  version of the messages would need updating as well.

**Describe alternatives you've considered**

I didn't put much thought into it but it could be that having the error message on two lines would make it even easier to have both a concise, machine-readable, message (first line), and a human-friendly one (second line).

**Additional context**

Related: would we consider having a colored output, to make the various portions of the error message easier to visually parse?

kartben, 2023-10-25:
@fabiobaltieri @bjarki-trackunit

fabiobaltieri, 2023-10-25:
Yeah the output is admittently a bit cryptic right now, I'll look into it.