zephyrproject-rtos/zephyr #64387: tests/lib/cmsis_dsp/filtering/libraries.cmsis_dsp.filtering.misc.fpu fails on mps3_an547

aescolar, 2023-10-25:
**Describe the bug**
tests/lib/cmsis_dsp/filtering/libraries.cmsis_dsp.filtering.misc.fpu  asserts on an mps3_an547

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p mps3_an547 -s tests/lib/cmsis_dsp/filtering/libraries.cmsis_dsp.filtering.misc.fpu 
2. See error

**Expected behavior**
No test failures

**Impact**
Main development blocked when this test is triggered in CI

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/actions/runs/6641677391/job/18044850533?pr=64134#step:13:2642

**Environment (please complete the following information):**
 - OS: CI / Ubuntu 22.04
 - Toolchain: SDK 0.16.3
 - Commit SHA or Version used: Master as of now 3f1eed17f83a93e6dcd731d2a3b77b48f985d68c

keith-packard, 2023-10-25:
Looks like the ARM conversion functions overwrite the buffer by one element in some cases. I stuck some checking in `test_arm_conv_q15` and `test_arm_conv_q7` and got this result:
```
...
START - test_arm_conv_q15_14_15
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q15_14_15 in 0.007 seconds
START - test_arm_conv_q15_14_16
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q15_14_16 in 0.007 seconds
===================================================================
START - test_arm_conv_q15_14_17
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q15_14_17 in 0.007 seconds
===================================================================
START - test_arm_conv_q15_14_18
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q15_14_18 in 0.007 seconds
...
START - test_arm_conv_q15_16_16
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q15_16_16 in 0.007 seconds
===================================================================
START - test_arm_conv_q15_16_17
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q15_16_17 in 0.007 seconds
===================================================================
START - test_arm_conv_q15_16_18
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q15_16_18 in 0.007 seconds
===================================================================
START - test_arm_conv_q15_16_33
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q15_16_33 in 0.007 seconds
===================================================================
START - test_arm_conv_q15_17_15
 PASS - test_arm_conv_q15_17_15 in 0.007 seconds
===================================================================
START - test_arm_conv_q15_17_16
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q15_17_16 in 0.007 seconds
...
START - test_arm_conv_q15_32_16
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q15_32_16 in 0.007 seconds
===================================================================
START - test_arm_conv_q15_32_17
 PASS - test_arm_conv_q15_32_17 in 0.007 seconds
===================================================================
START - test_arm_conv_q15_32_18
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q15_32_18 in 0.007 seconds
===================================================================
START - test_arm_conv_q15_32_33
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q15_32_33 in 0.007 seconds
...
START - test_arm_conv_q7_30_31
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q7_30_31 in 0.007 seconds
===================================================================
START - test_arm_conv_q7_30_32
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q7_30_32 in 0.007 seconds
===================================================================
START - test_arm_conv_q7_30_33
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q7_30_33 in 0.007 seconds
===================================================================
START - test_arm_conv_q7_30_34
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q7_30_34 in 0.007 seconds
===================================================================
START - test_arm_conv_q7_30_49
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q7_30_49 in 0.007 seconds
...
START - test_arm_conv_q7_32_32
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q7_32_32 in 0.007 seconds
===================================================================
START - test_arm_conv_q7_32_33
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q7_32_33 in 0.007 seconds
===================================================================
START - test_arm_conv_q7_32_34
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q7_32_34 in 0.007 seconds
===================================================================
START - test_arm_conv_q7_32_49
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q7_32_49 in 0.007 seconds
===================================================================
START - test_arm_conv_q7_33_31
 PASS - test_arm_conv_q7_33_31 in 0.006 seconds
===================================================================
START - test_arm_conv_q7_33_32
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q7_33_32 in 0.007 seconds
...
START - test_arm_conv_q7_48_32
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q7_48_32 in 0.007 seconds
===================================================================
START - test_arm_conv_q7_48_33
 PASS - test_arm_conv_q7_48_33 in 0.007 seconds
===================================================================
START - test_arm_conv_q7_48_34
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q7_48_34 in 0.007 seconds
===================================================================
START - test_arm_conv_q7_48_49
overwrote output 1 past end (0 instead of baad)
 PASS - test_arm_conv_q7_48_49 in 0.007 seconds
```
This fails because the Zephyr allocator is catching the buffer overrun while the newlib allocator appears to miss the fault.
We'll need to find someone who understands this code to see whether there's a fault in the code or whether the buffer simply needs to be overallocated by one element. It looks like the code is always writing `0` at this location; maybe that's of some help in debugging?

keith-packard, 2023-10-25:
I've sumitted pr #64393 which checks explicitly for buffer overrun in the hope that this will help debug the underlying cmsis issue.

aescolar, 2023-10-26:
Lowering to medium as the test has been temporarily disabled in CI for this platform.
Still, the issue should be investigated and fixes (and the temporary patch https://github.com/zephyrproject-rtos/zephyr/commit/b3de6432069dafa680c0c712bf53d51809ee961b removed)

closed 2024-01-09 by github-actions[bot]