zephyrproject-rtos/zephyr #64389: net: Add IGMP shell to manage multicast groups

krish2718, 2023-10-25:
**Is your feature request related to a problem? Please describe.**
Test multicast groups, handy to have a shell command rather than writing applications for testing.

**Describe the solution you'd like**
It’s handy to have a IGMP shell as part of net to manage and test multicast groups, multicast is essential for Matter.

Below should suffice to start with:

`net igmp join <group address>`

`net igmp leave <group address`

jukkar, 2023-10-26:
I am not sure we need separate shell commands for this. As the multicast networking does not really work without joining to IGMP (for IPv4) or MLD (for IPv6), we could modify the `net ipv4` and `net ipv6` shell commands so that when user adds a multicast address to the interface, the shell command will automatically join the multicast group.

jukkar, 2023-10-26:
I can implement this, will send a PR shortly.

krish2718, 2023-10-26:
> that when user adds a multicast address to the interface, the shell command will automatically join the multicast group.

Yes, this is how it works in Linux, adding a route will trigger IGMP join.

PR #64417 (2023-10-27): Join IPv4 or IPv6 multicast group in net-shell if needed
If user adds a multicast address to a network interface, the net-shell command will join the multicast group automatically.

Fixes #64389
commit 1467d7fa (2023-10-27): net: shell: Join IPv4 mcast group if needed

If user adds IPv4 address to the network interface, check
if the address is a multicast one and add it to multicast
group if it is.

Fixes #64389
closed 2023-10-27 by carlescufi