zephyrproject-rtos/zephyr #64396: testsuite: coverage: failed to parse coverage data on qemu_x86

golowanow, 2023-10-25:
**Describe the bug**

gcovr failed to parse coverage data on qemu_x86 after changes introduced by #63826 in bc8b7dd6dd

**To Reproduce**
Steps to reproduce the behavior:

./scripts/twister -vv -j 1 -p qemu_x86 --coverage --coverage-tool gcovr  -T samples/hello_world

**Expected behavior**
a coverage report generated

**Impact**
unable to get coverage data on qemu_x86

**Logs and console output**
```
INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 errored, 0 skipped with 0 warnings in 41.12 seconds
INFO    - In total 1 test cases were executed, 0 skipped on 1 out of total 630 platforms (0.16%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Generating coverage files...
DEBUG   - Select gcovr as the coverage tool...
DEBUG   - Working on /home/foobar/zephyrproject/zephyr/twister-out/qemu_x86/samples/hello_world/sample.basic.helloworld/handler.log
DEBUG   - Generating gcda files
DEBUG   - Gcov data captured: /home/foobar/zephyrproject/zephyr/twister-out/qemu_x86/samples/hello_world/sample.basic.helloworld/handler.log
DEBUG   - Running gcovr -r /home/foobar/zephyrproject/zephyr --gcov-executable /home/foobar/zephyr-sdk/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcov -e tests/* -e .*generated.* -e .*/tests/.* -e .*/samples/.* --json -o /home/foobar/zephyrproject/zephyr/twister-out/coverage.json /home/foobar/zephyrproject/zephyr/twister-out...
(WARNING) Unrecognized GCOV output for /home/foobar/zephyrproject/zephyr/subsys/testsuite/coverage/coverage.c
	  branch  1 taken -1
	This is indicative of a gcov output parse error.
	Please report this to the gcovr developers
	at <https://github.com/gcovr/gcovr/issues>.
(WARNING) Exception during parsing:
	NegativeHits: Got negative hit value in gcov line 'branch  1 taken -1' caused by a bug in gcov tool, see https://gcc.gnu.org/bugzilla/show_bug.cgi?id=68080. Use option --gcov-ignore-parse-errors with a value of negative_hits.warn, or negative_hits.warn_one.
(ERROR) Exiting because of parse errors.
	You can run gcovr with --gcov-ignore-parse-errors
	to continue anyway.
(ERROR) Trouble processing '/home/foobar/zephyrproject/zephyr/twister-out/qemu_x86/samples/hello_world/sample.basic.helloworld/zephyr/CMakeFiles/zephyr.dir/subsys/testsuite/coverage/coverage.c.gcda' with working directory '/home/foobar/zephyrproject/zephyr'.
Stdout of gcov was >>File '/home/foobar/zephyrproject/zephyr/subsys/testsuite/coverage/coverage.c'
Lines executed:81.67% of 120
Branches executed:100.00% of 36
Taken at least once:83.33% of 36
Calls executed:72.97% of 37
Creating 'coverage.c##19135d5d887750668d3beefb9f920d68.gcov'

Lines executed:81.67% of 120
<< End of stdout
Stderr of gcov was >><< End of stderr
Current processed gcov file was '/home/foobar/zephyrproject/zephyr/coverage.c##19135d5d887750668d3beefb9f920d68.gcov'.
Use option --verbose to get extended informations.
Traceback (most recent call last):
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/workers.py", line 80, in worker
    work(*args, **kwargs)
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/gcov.py", line 323, in process_datafile
    done = run_gcov_and_process_files(
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/gcov.py", line 564, in run_gcov_and_process_files
    process_gcov_data(fname, covdata, abs_filename, options)
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/gcov.py", line 140, in process_gcov_data
    coverage, source_lines = parse_coverage(
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/gcov_parser.py", line 355, in parse_coverage
    _report_lines_with_errors(
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/gcov_parser.py", line 521, in _report_lines_with_errors
    raise errors[0]  # guaranteed to have at least one exception
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/gcov_parser.py", line 317, in parse_coverage
    _parse_line(raw_line, ignore_parse_errors, persistent_states),
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/gcov_parser.py", line 628, in _parse_line
    tag = _parse_tag_line(line, ignore_parse_errors, persistent_states)
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/gcov_parser.py", line 745, in _parse_tag_line
    raise NegativeHits(line)
gcovr.gcov_parser.NegativeHits: Got negative hit value in gcov line 'branch  1 taken -1' caused by a bug in gcov tool, see https://gcc.gnu.org/bugzilla/show_bug.cgi?id=68080. Use option --gcov-ignore-parse-errors with a value of negative_hits.warn, or negative_hits.warn_one.
Traceback (most recent call last):
[... 39 lines cut ...]
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/gcov.py", line 564, in run_gcov_and_process_files
    process_gcov_data(fname, covdata, abs_filename, options)
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/gcov.py", line 140, in process_gcov_data
    coverage, source_lines = parse_coverage(
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/gcov_parser.py", line 355, in parse_coverage
    _report_lines_with_errors(
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/gcov_parser.py", line 521, in _report_lines_with_errors
    raise errors[0]  # guaranteed to have at least one exception
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/gcov_parser.py", line 317, in parse_coverage
    _parse_line(raw_line, ignore_parse_errors, persistent_states),
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/gcov_parser.py", line 628, in _parse_line
    tag = _parse_tag_line(line, ignore_parse_errors, persistent_states)
  File "/home/foobar/zephyrproject/.venv/lib/python3.8/site-packages/gcovr/gcov_parser.py", line 745, in _parse_tag_line
    raise NegativeHits(line)
gcovr.gcov_parser.NegativeHits: Got negative hit value in gcov line 'branch  1 taken -1' caused by a bug in gcov tool, see https://gcc.gnu.org/bugzilla/show_bug.cgi?id=68080. Use option --gcov-ignore-parse-errors with a value of negative_hits.warn, or negative_hits.warn_one.
(ERROR) Bad --add-tracefile option.
	The specified file does not exist.
DEBUG   - All coverage data processed: False

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain  Zephyr SDK
 - d8b153e00f06d

golowanow, 2023-10-25:
@nordic-piks FYI

nordic-piks, 2023-10-25:
> @nordic-piks FYI

Interesting.
I am unable to reproduce this error. Just used the command from the bug description and generated coverage for this function:
[image]

I am using:
gcovr 4.2
zephyr-sdk/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcov,
gcov (Zephyr SDK 0.16.0) 12.2.0

golowanow, 2023-10-25:
```
./zephyrproject/.venv/bin/gcovr
gcovr 6.0
```
```
~/zephyr-sdk/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcov --version
gcov (Zephyr SDK 0.16.3) 12.2.0
```

golowanow, 2023-10-25:
@nordic-piks, it looks like GCC optimized out `if { memcpy() }` branch and only `print_u32()` are left at  `write_u64()`.
My first quick fix was wrapping this function into '-O0' like this:
```
#pragma GCC push_options
#pragma GCC optimize("O0")
static inline void write_u64( ... ) {
...
}
#pragma GCC pop_options
```

nordic-piks, 2023-10-25:
> @nordic-piks, it looks like GCC optimized out `if { memcpy() }` branch and only `print_u32()` are left at `write_u64()`. My first quick fix was wrapping this function into '-O0' like this:
>
> ``` [...]

I just upgraded gcovr to 6.0 (leaving all other versions the same) and got the same error as you.
This suggests that this is not an issue with the compiler/optimization, but rather incompatibility between gcovr 6.0 and gcov 12.2.0.
Can you try downgrading to gcovr 4.2?

nordic-piks, 2023-10-25:
> I just upgraded gcovr to 6.0 (leaving all other versions the same) and got the same error as you. This suggests that this is not an issue with the compiler/optimization, but rather incompatibility [...]

Error in console suggest bug in gcov: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=68080

golowanow, 2023-10-25:
hmm, the same gcov has no problems running with b5bc088529 but fails on bc8b7dd6dd

nordic-piks, 2023-10-25:
> hmm, the same gcov has no problems running with [b5bc088](https://github.com/zephyrproject-rtos/zephyr/commit/b5bc088529269dd70210d42366f5d20e24658fc3) but fails on [bc8b7dd](https://github.com/zephyrproject-rtos/zephyr/commit/bc8b7dd6dd36e0ae6b256353bf967c3568b907a7)

As the bug describes, there are threading issues - as I understand it, at the same time something is reading counters and writing them.
This is totally true for this case as we are trying to get coverage for the code of the function that is reading those counters, thus when it is executing, it is writing them at the same time ;)

Still, my change can trigger that bug, as it adds a branch, which may influence which counters are produced or the timing when they are updated.
This means that it may be just a coincidence for specific compiler, optimization level, and code.
As I understand, the same issue may still happen in any other line in this particular file, it is just a matter of how the compiler arranges code or how execution happens.
The issue remains, counters are updated and read at the same time.

My proposal would be to follow suggestion from gcovr doc - ignore this error/bug:
https://gcovr.com/en/6.0/guide/gcov_parser.html#negative-hit-counts

What do you think?

golowanow, 2023-10-25:
> My proposal would be to follow suggestion from gcovr doc - ignore this error/bug

not sure I understand your point - how to ignore it ?
This issue happens not only for the hello_world sample, and #64397 helps to get coverage working again.

nordic-piks, 2023-10-25:
> > My proposal would be to follow suggestion from gcovr doc - ignore this error/bug
> 
> not sure I understand your point - how to ignore it ? This issue happens not only for the hello_world sample, and #64397 helps to get coverage working again.

This error can be ignored by adding an option to gcovr call: --gcov-ignore-parse-errors=negative_hits.warn
This can be added here: https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/pylib/twister/twisterlib/coverage.py#L224

I suggest that because even with your change, others, with different compilers or optimization levels can be hit by the same problem but at different locations in this file.

Additionally, this ignore option (and this error message) was introduced at gcovr 6.0.
So to add it, this needs to be updated too:
https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/requirements-build-test.txt#L13

nordic-piks, 2023-10-25:
Looking here:
https://github.com/gcovr/gcovr/issues/583#issuecomment-1340762818
LCOV is ignoring such case by default.

golowanow, 2023-10-26:
@nordic-piks, thank you for your workaround suggestion, please see it in #64406

PR #64406 (2023-11-03): scripts: coverage: gcovr to ignore negative counters
Apply a workaround for an issue introduced with bc8b7dd for code coverage collection at qemu_x86 when gcov data was broken because of optimized out memcpy() in write_u64() static inline function, so gcovr parser fails with "gcovr.gcov_parser.NegativeHits: Got negative hit value in gcov line 'branch  1 taken -1'".

Add gcovr command line option to ignore such errors (since v.6.0): "--gcov-ignore-parse-errors=negative_hits.warn_once_per_file"

See also:
https://gcovr.com/en/stable/changelog.html#march-2023
https://github.com/gcovr/gcovr/pull/7 [... cut]
closed 2023-11-03 by nashif