zephyrproject-rtos/zephyr #64405: Cellular modem API development

bjarki-trackunit, 2023-10-26:
## Introduction

We need a device driver API to configure and control cellular modems. This RFC will list every individual feature the API should cover, documenting the requirements and rationale behind every individual feature. This RFC will then be used to track the development of each individual features development, which should be in large part developed in parallel driven by a variation of developers.

### Cellular modem API progress overview
The following tasks indicate whether a feature has been successfully implemented or not.
- [ ] Setting network credentials (PIN, PUK, APN, ...)
- [ ] Getting network signal quality
- [ ] Getting network status (searching, registered, roaming)
- [ ] Setting radio access technologies (LTE, NB-IOT, used bands, ...)
- [ ] Getting supported radio access technologies and bands
- [ ] Getting modem properties (IMEI, IMSI)
- [ ] Configuring power saving

### Cellular modem API documentation
Each individual feature is described in detail here

#### Setting network credentials
Assignee: Unassigned
PR: None

#### Getting network signal quality
Assignee: Unassigned
PR: None

#### Getting network status
Assignee: Unassigned
PR: None

#### Setting radio access technologies
Assignee: Unassigned
PR: None

#### Getting supported radio access technologies and bands
Assignee: Unassigned
PR: None

#### Getting modem properties
Assignee: Unassigned
PR: None

### Status summary
The RFC is at this point only an initial proposal to which features we should implement, and what they should cover. Everything is open to change, and no actual work has begun.

jeffwelder-ellenbytech, 2023-10-26:
Looking forward to these improvements! :)

I'm interested in switching firmwares using the `AT#FWSWITCH` command for use with eSIM. In the USA there's often multiple carrier firmwares in a modem for different provider requirements: (AT&T/Verizon/Global)
Maybe this would be part of Setting Radio Access Technologies, or maybe a new category is more appropriate.

Probably could find some time to work on it once something is defined, or with some guidance. 

Also maybe a category for HW Settings such as Flow Control, Baud, etc.

naNEQ, 2023-11-17:
Very much looking forward to that and willing to help.

Small input from my side, in the "Getting modem properties" part, I think these should be included:

- Modem IMEI (AT+CGSN)
- Modem ID (AT+CGMM)
- Modem manufacturer (AT+CGMI)
- Modem FW version (AT+CGMR)
- SIM card IMSI (AT+CIMI)
- SIM card ICCID (AT+QCCID)

beriberikix, 2024-06-21:
Personally I think adopting something like this functionality for Wi-Fi would be a nice benefit. There's different protocols (ex. custom AT cmds), interfaces (SPI/UART/SDIO/USB), features (ex. co-existence) that may be similar across wi-fi vendors but are subtly different.