zephyrproject-rtos/zephyr #64436: stm32 async uart: unknown state of the module while enabling transfer

semihalf-niedzwiecki-dawid, 2023-10-26:
Actually, I'm not sure if it's a bug or a feature :-) feel free to change the label.

The UART module is enabled at `PRE_KERNEL_1` level. Let's assume the `uart_rx_enable` (enabling async transfer) function is called at `POST_KERNEL` level. It can't be called earlier, because of some dependencies. There is a window when UART module is enabled, but the async transfer not. If we receive a byte or two in this window the RXNE bit will be set (and potentially overrun). Then, when the DMA is enabled, it will copy the one byte from the DR register (because RXNE is set), but an interrupt won't be trigger, because an IDLE line is not detected. When the next transfer occurs, the interrupt will be triggered and pass all new data with the one byte from the previous transfer at the beginning. 

I think it is a bit misleading. Do you think it is intended behavior?

IMO, there are two possible solutions for that:
1. Clear the RXNE (and maybe errors) bit in the UART driver, before enabling DMA not to copy data received before enabling async transfer.
2. Do not change the current implementation and assume it is up to the driver user to read the register (and clear errors?) with `uart_poll_in` which clears the RXNE flag.

Please let me know what you think.

MaureenHelm, 2023-10-31:
@dcpleung can you take a look?

dcpleung, 2023-10-31:
I can think of two other ways to deal with this (in addition to the one mentioned above):

* Enable hardware flow control on both end, and de-assert CTS so the other end does not send anything until you are ready. Though it will still receive whatever is at the other end waiting in queue. Or,
* You can always clear out the input (e.g. by doing polling in until there is nothing to be received) before enabling async transfer.

But I think it is really up to the user of async transfer to decide what to do, as it will require both ends to agree to some handshaking before exchanging any data.

niedzwiecki-dawid, 2023-11-06:
Thanks for the response :-)

>Enable hardware flow control on both end, and de-assert CTS so the other end does not send anything until you are ready. Though it will still receive whatever is at the other end waiting in queue. Or,

Unfortunately, I can't change how the transmitter works, so enabling flow control isn't an option in my case.

 >You can always clear out the input (e.g. by doing polling in until there is nothing to be received) before enabling async transfer.

It is 2 solution from my first comment, right? Or I'm missing something.

So, you think adding clearing `RXNE` flag in the `uart_stm32_async_rx_enable` function (inside the driver) is not a good option?

dcpleung, 2023-11-06:
I am not sure about the STM32 hardware. But if there is a way to tell the hardware to discard buffered input (or even simply reset the hardware to clear out everything), that would work better/faster than clearing by doing polling-in.

Also, correct me if I am wrong, even if you clear `RXNE` in `uart_stm32_async_rx_enable`, there is still data in the FIFO, so the interrupt will still fire, right?

niedzwiecki-dawid, 2023-11-07:
The async mode always works with DMA. The `RXNE` bit is a trigger for DMA to copy data from `DR` (data register) to a provided buffer during configuration. When new data is received and RXNE is cleared by DMA, `DR` register content is just replaced with the new byte.
[image]

However, you are right. There are some STM32 chips e.g. STM32H7 that can use FIFO embedded in the UART module. When the FIFO mode is enabled, the `RXFNE` (RX FIFO not empty) is a trigger for the DMA. With FIFO disabled it is also `RXNE`. But the current STM32 driver doesn't support enabling FIFO mode, so there is no need to clear that bit at the moment.

Regarding interrupt - the async mode uses IDLE line interrupt. The interrupt is triggered after transfer, when the IDLE line is detected. The `RXNE` is used only to trigger DMA transfer, after every byte.

dcpleung, 2023-11-07:
I don't have a good generic solution...

As for your application, do you know if the transmitter has flow control enabled at all? Or does it disregard any CTS/RTS and simply keeps on sending stuff? Although you can't change the transmitting side, maybe they have something that you can use?

niedzwiecki-dawid, 2023-11-08:
Unfortunately, it is not enabled. In my opinion, clearing RXNE is a good solution. I've sent PR #64950 to present that.

msianko, 2023-11-14:
I've noticed quite similar issue. Let's say the `uart_rx_enable()` function is called (which calls 'uart_stm32_async_rx_enable()' function). Next `uart_rx_disable()` is called (which calls 'uart_stm32_async_rx_disable()'). Next some data is sent to the microcontroller via UART. In this case the subsequent bytes are overwriting the RDR. After the transfer the RDR stores the most recent byte which is copied when 'uart_stm32_async_rx_enable()' is called again. In case of this issue please pay attention that in the body of 'uart_stm32_async_rx_enable()':
```
static inline void uart_stm32_dma_rx_enable(const struct device *dev)
{
	const struct uart_stm32_config *config = dev->config;
	struct uart_stm32_data *data = dev->data;

	LL_USART_EnableDMAReq_RX(config->usart);

	data->dma_rx.enabled = true;
}
```
the 'LL_USART_EnableDMAReq_RX()' function is called while in 'uart_stm32_dma_rx_disable()'
```
static inline void uart_stm32_dma_rx_disable(const struct device *dev)
{
	struct uart_stm32_data *data = dev->data;

	data->dma_rx.enabled = false;
}
```
The DMA request is not disabled here (the bit DMAR in USART's CR3 register is not cleared). Calling 'LL_USART_DisableDMAReq_RX()' from  'uart_stm32_dma_rx_disable()' solves this problem. What do you think? @ FRASTM  @niedzwiecki-dawid

erwango, 2023-11-16:
@niedzwiecki-dawid can you comment on this ?

niedzwiecki-dawid, 2023-11-17:
Sorry, I've missed that comment.

This is a slightly different case. I was describing a situation when data are received before first  `uart_stm32_async_rx_enable` and what happens after the call. `uart_stm32_async_rx_disable` hasn't been even called when the problem occurs.

However, it is also important and I have a few notes. 

I don't know why `LL_USART_DisableDMAReq_RX` isn't called in `uart_stm32_dma_rx_disable`. At the first glance it should be. However DMA is disabled with `dma_stop`, so data are not copied after every received byte and the `DMAR` bit is enabled in the `uart_stm32_async_rx_enable` function anyway.

>After the transfer the RDR stores the most recent byte which is copied when uart_stm32_async_rx_enable() is called again.

I don't understand this point. `uart_stm32_async_rx_disable` function enables `RXNE` interrupt with `LL_USART_EnableIT_RXNE`. The ISR should clear `RXNE` bit:
```
#ifdef CONFIG_UART_ASYNC_API
	if (LL_USART_IsEnabledIT_IDLE(config->usart) &&
			LL_USART_IsActiveFlag_IDLE(config->usart)) {

		LL_USART_ClearFlag_IDLE(config->usart);

		LOG_DBG("idle interrupt occurred");

		if (data->dma_rx.timeout == 0) {
			uart_stm32_dma_rx_flush(dev);
		} else {
			/* Start the RX timer not null */
			async_timer_start(&data->dma_rx.timeout_work,
								data->dma_rx.timeout);
		}
	} else if (LL_USART_IsEnabledIT_TC(config->usart) &&
			LL_USART_IsActiveFlag_TC(config->usart)) {

		LL_USART_DisableIT_TC(config->usart);
		LL_USART_ClearFlag_TC(config->usart);
		/* Generate TX_DONE event when transmission is done */
		async_evt_tx_done(data);

#ifdef CONFIG_PM
		uart_stm32_pm_policy_state_lock_put(dev);
#endif
	} else if (LL_USART_IsEnabledIT_RXNE(config->usart) &&
			LL_USART_IsActiveFlag_RXNE(config->usart)) {
#ifdef USART_SR_RXNE
		/* clear the RXNE flag, because Rx data was not read */
		LL_USART_ClearFlag_RXNE(config->usart);
#else
		/* clear the RXNE by flushing the fifo, because Rx data was not read */
		LL_USART_RequestRxDataFlush(config->usart);
#endif /* USART_SR_RXNE */
	}

	/* Clear errors */
	uart_stm32_err_check(dev);
#endif /* CONFIG_UART_ASYNC_API */
```
`RXNE` bit triggers DMA transfer, so once it is cleared, the old byte from `RDR` should not be copied after `uart_stm32_async_rx_enable`. 

I have hypothesis why this happens:
The `DMAR` (DMA request) bit somehow changes the way how `RXNE` interrupt works - `uart_stm32_isr` is not called for `RXNE` interrupt, so `RXNE` is not cleared and it triggers DMA after `uart_stm32_async_rx_enable`. If that's the point, I don't get why `RDR` stores the most recent byte, not the first after disabling asyc transfer. According to my experience, `RDR` is not replaced with a new byte when `RXNE` bit is set, only the overrun error is signaled, but maybe it depends on a chip.

Would you be able to check if you enter this code
```
	} else if (LL_USART_IsEnabledIT_RXNE(config->usart) &&
			LL_USART_IsActiveFlag_RXNE(config->usart)) {
#ifdef USART_SR_RXNE
		/* clear the RXNE flag, because Rx data was not read */
		LL_USART_ClearFlag_RXNE(config->usart);
#else
		/* clear the RXNE by flushing the fifo, because Rx data was not read */
		LL_USART_RequestRxDataFlush(config->usart);
#endif /* USART_SR_RXNE */
```
with and without `DMAR` bit cleared.

erwango, 2023-11-17:
> I don't know why LL_USART_DisableDMAReq_RX isn't called in uart_stm32_dma_rx_disable. At the first glance it should be. However DMA is disabled with dma_stop, so data are not copied after every received byte and the DMAR bit is set in the uart_stm32_async_rx_enable function anyway.

@FRASTM can answer maybe ?

PR #65320 (2023-11-20): STM32 UART: Request RX data flush on async_rx_enable
When enabling async RX the first time after boot, there is an additional byte received with the first RX_DATA_RDY event, which seems to be caused by the RX data not being flushed before enabling the UART RX DMA.

Adding a request to flush the RX data register before enabling the RX DMA, solves the issue.

Potentially fixes #64436

To recreate the issue, connect the RX and TX UART pins on a b_u585i_iot02a board, and use the following overlay:
```
/*
 * Pins 2 and 3 must be connected to each other on the STMOD+1 connector to
 * loopback RX/TX. [... cut]
closed 2023-11-20 by MaureenHelm