zephyrproject-rtos/zephyr #64442: LE Audio: Cannot set Media Player into Inactive State

piotrnarajowski, 2023-10-26:
**Describe the bug**
While going through GMCS Server role test cases, many of them require Media Player to be initially set to Inactive State (MEDIA_PROXY_STATE_INACTIVE 0x00). Currently Media Player is initialized with hardcoded groups and tracks and is automatically set to Paused State, which makes it impossible to set it to inactive state.

**To Reproduce**
Can be reproduced with AutoPTS.

**Expected behavior**
According to [MCS spec](https://www.bluetooth.com/specifications/mcs-1-0/):
"If the media player has no current track, then the media player is inactive."
Media Player should be initialized in an inactive state without current track, or API should allow to set it into inactive state. 

**Impact**
Testcases that require Media Player in Inactive State fail.

**Logs and console output**
[GMCS_SR_MCP_BV_76_C_2023_10_26_12_44_50.zip](https://github.com/zephyrproject-rtos/zephyr/files/13176349/GMCS_SR_MCP_BV_76_C_2023_10_26_12_44_50.zip)

**Environment (please complete the following information):**
 - OS: Windows 11
 - Toolchain: Zephyr SDK 0.16.3
 - https://github.com/zephyrproject-rtos/zephyr/pull/64377

asbjornsabo, 2023-11-28:
The media player can be set to inactive statue using the MPL shell command "test_set_media_state", or the function call mpl_test_media_state_set.
(There is also an "mpl_test_unset_parent_group" that used to be required to pass PTS tests, and a few others.)
These are found in the file mpl_internal.h.
Dependent upon "CONFIG_BT_MPL_LOG_LEVEL_DBG" being configured.

piotrnarajowski, 2023-11-28:
Thanks, don't know why I missed that...

closed 2023-11-29 by Thalley