zephyrproject-rtos/zephyr #64449: LE Audio: No Broadcast Receive State notification with PA Sync State set to Synchronized to PA (0x02)

mkasenberg, 2023-10-26:
**Describe the bug**
In the  BASS/SR/CP/BV-05-C test case [(BASS.TS.p1.pdf)](https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=524502) the PTS executes an Add Source operation to the Broadcast Audio Scan Control Point characteristic of IUT (Zephyr Host + Zephyr Controller setup) and awaits Broadcast Receive State notification with `PA Sync State` set to `Synchronized to PA (0x02)`. But only the first notification with `PA Sync State` set to `Not synchronized to PA (0x00)` is sent. There is no error in IUT logs, MTU is fine, I used Ellisys sniffer and the expected notification packet is never transmitted over the air.

**To Reproduce**
Can be reproduced with nrf5340 board and AutoPTS. I will edit this to add all steps later.

**Expected behavior**
The issue appears only with Zephyr Host + Zephyr Controller setup. The setup with Zephyr Host + Packetcraft Controller works fine in this test case, see the Ellisys logs attached below.

**Impact**
The BASS/SR/CP test cases fail.

**Logs and console output**
- Zephyr Host + Zephyr Controller setup:
Ellisys Bluetooth Analyzer logs:
[BASS_zephyr.btt.txt](https://github.com/zephyrproject-rtos/zephyr/files/13177200/BASS_zephyr.btt.txt)
IUT logs:
[BASS_SR_CP_BV-05-C_iutctl.log](https://github.com/zephyrproject-rtos/zephyr/files/13177604/BASS_SR_CP_BV-05-C_iutctl.log)

- Zephyr Host + Packetcraft Controller setup:
[BASS_packetcraft.btt.txt](https://github.com/zephyrproject-rtos/zephyr/files/13177612/BASS_packetcraft.btt.txt)

**Environment:**
 - OS: Windows 11
 - Toolchain: Zephyr SDK 0.16.3
 - Commit SHA: 96c5052733be0eb73da2c47eb2c85e18ec54fca9

mkasenberg, 2023-11-16:
With @MariuszSkamra we have analyzed logs and added marks to Ellisys logs that may help to solve the issue
[rejected_conn_param_update_zephyr_stopped_responding.btt.txt](https://github.com/zephyrproject-rtos/zephyr/files/13375405/rejected_conn_param_update_zephyr_stopped_responding.btt.txt)

mkasenberg, 2023-11-16:
Something wrong happens with Zephyr controller after this:
[image]

mkasenberg, 2023-11-16:
@cvinayak

mkasenberg, 2023-12-28:
Affected PTS test cases:
BASS/SR/CP/BV-04-C
BASS/SR/CP/BV-05-C
BASS/SR/CP/BV-06-C
BASS/SR/CP/BV-07-C
BASS/SR/CP/BV-12-C
BASS/SR/CP/BV-13-C
BASS/SR/CP/BV-14-C
BASS/SR/CP/BV-15-C
BASS/SR/CP/BV-17-C
BASS/SR/CP/BV-18-C
BASS/SR/CP/BV-19-C
BASS/SR/CP/BV-20-C
BASS/SR/SPE/BI-05-C
BASS/SR/SPE/BI-06-C

Thalley, 2023-12-28:
@mkasenberg are we sure this is a controller issue? 
Have you tried running the tests with another controller to verify?

mkasenberg, 2023-12-28:
I think so, I used the Packetcraft controller to complete the bt tester support for these test cases.

Some of these test cases have other issues e.g. the BASS/SR/CP/BV-04-C has the host-related issue described here https://github.com/zephyrproject-rtos/zephyr/issues/64183.

cvinayak, 2024-02-02:
Could be fixed by https://github.com/zephyrproject-rtos/zephyr/pull/68456

mkasenberg, 2024-02-09:
Fixed, thanks

closed 2024-02-09 by mkasenberg