zephyrproject-rtos/zephyr #64466: Zephyr Release 3.2.0 Failed to Build/Link for QEMU-RISV Board Target Device Uart_Pipe

bob2oneil, 2023-10-26:
I am able to build an application targeting Zephyr 3.2.0 for the Native-POSIX board target, but the same source does not build for the QEMU-RISV board target.  That has not always been the case historically, so something has changed over the course of time.

The point of failures appears to be building the driver source file "uart_pipe.c"

from /home/user/zephyrproject/zephyr/drivers/serial/uart_pipe.c:14:

Can anyone suggest where I am might look for a solution based on the following error logging.

Is there any particular kernel or QEMU-RISV specific kernel configurations that would address this issue?

Any suggestions are welcome.

Build logging is as follows:

```
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/user/Project/0dev/software/black
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /home/user/.cache/zephyr
-- Zephyr version: 3.2.0 (/home/user/zephyrproject/zephyr)
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: qemu_riscv64
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.0 (/home/user/zephyr-sdk-0.15.0)
-- Found toolchain: zephyr 0.15.0 (/home/user/zephyr-sdk-0.15.0)
-- Found Dtc: /usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/user/zephyrproject/zephyr/boards/riscv/qemu_riscv64/qemu_riscv64.dts
-- Generated zephyr.dts: /home/user/Project/0dev/software/black/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/user/Project/0dev/software/black/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/user/Project/0dev/software/black/build/zephyr/dts.cmake
Parsing /home/user/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/user/zephyrproject/zephyr/boards/riscv/qemu_riscv64/qemu_riscv64_defconfig'
Merged configuration '/home/user/Project/0dev/software/black/prj.conf'
Merged configuration '/home/user/Project/0dev/software/black/conf/qemu.conf'
Configuration saved to '/home/user/Project/0dev/software/black/build/zephyr/.config'
Kconfig header saved to '/home/user/Project/0dev/software/black/build/zephyr/include/generated/autoconf.h'

warning: QEMU_ICOUNT_SHIFT (defined at boards/Kconfig:67) was assigned the value '6' but got the
value ''. Check these unsatisfied dependencies: QEMU_ICOUNT (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_QEMU_ICOUNT_SHIFT and/or look up
QEMU_ICOUNT_SHIFT in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: FLASH_SHELL (defined at drivers/flash/Kconfig:46) was assigned the value 'y' but got the
value 'n'. Check these unsatisfied dependencies: FLASH_PAGE_LAYOUT (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_FLASH_SHELL and/or look up FLASH_SHELL in
the menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration
Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

warning: FLASH_PAGE_LAYOUT (defined at drivers/flash/Kconfig:54) was assigned the value 'y' but got
the value 'n'. Check these unsatisfied dependencies: FLASH_HAS_PAGE_LAYOUT (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_FLASH_PAGE_LAYOUT and/or look up
FLASH_PAGE_LAYOUT in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: FILE_SYSTEM_LITTLEFS (defined at subsys/fs/Kconfig.littlefs:6) was assigned the value 'y'
but got the value 'n'. Check these unsatisfied dependencies: FLASH_MAP (=n), FLASH_PAGE_LAYOUT (=n).
See http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_FILE_SYSTEM_LITTLEFS and/or look up
FILE_SYSTEM_LITTLEFS in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

warning: FLASH_MAP (defined at subsys/storage/flash_map/Kconfig:10) was assigned the value 'y' but
got the value 'n'. Check these unsatisfied dependencies: FLASH_HAS_DRIVER_ENABLED (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_FLASH_MAP and/or look up FLASH_MAP in the
menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration Values,
and Kconfig - Tips and Best Practices sections of the manual might be helpful too.
[... 36 lines cut ...]
                 from /home/user/zephyrproject/zephyr/include/zephyr/logging/log_msg.h:11,
                 from /home/user/zephyrproject/zephyr/include/zephyr/logging/log_core.h:9,
                 from /home/user/zephyrproject/zephyr/include/zephyr/logging/log.h:11,
                 from /home/user/zephyrproject/zephyr/drivers/serial/uart_pipe.c:14:
/home/user/zephyrproject/zephyr/include/zephyr/device.h:104:39: error: '__device_dts_ord_DT_CHOSEN_zephyr_uart_pipe_ORD' undeclared here (not in a function)
  104 | #define DEVICE_NAME_GET(name) _CONCAT(__device_, name)
      |                                       ^~~~~~~~~
/home/user/zephyrproject/zephyr/include/zephyr/toolchain/common.h:132:26: note: in definition of macro '_DO_CONCAT'
  132 | #define _DO_CONCAT(x, y) x ## y
      |                          ^
/home/user/zephyrproject/zephyr/include/zephyr/device.h:104:31: note: in expansion of macro '_CONCAT'
  104 | #define DEVICE_NAME_GET(name) _CONCAT(__device_, name)
      |                               ^~~~~~~
/home/user/zephyrproject/zephyr/include/zephyr/device.h:264:37: note: in expansion of macro 'DEVICE_NAME_GET'
  264 | #define DEVICE_DT_NAME_GET(node_id) DEVICE_NAME_GET(Z_DEVICE_DT_DEV_NAME(node_id))
      |                                     ^~~~~~~~~~~~~~~
/home/user/zephyrproject/zephyr/include/zephyr/device.h:282:34: note: in expansion of macro 'DEVICE_DT_NAME_GET'
  282 | #define DEVICE_DT_GET(node_id) (&DEVICE_DT_NAME_GET(node_id))
      |                                  ^~~~~~~~~~~~~~~~~~
/home/user/zephyrproject/zephyr/drivers/serial/uart_pipe.c:25:9: note: in expansion of macro 'DEVICE_DT_GET'
   25 |         DEVICE_DT_GET(DT_CHOSEN(zephyr_uart_pipe));
      |         ^~~~~~~~~~~~~
[265/365] Linking CXX static library app/libapp.a

```

henrikbrixandersen, 2023-10-27:
@bob2oneil Zephyr v3.2.0 is no longer supported. Can you reproduce this issue on a later release - or even better on current `main`?

bob2oneil, 2023-10-30:
I am pegged to the 3.2.0 LTS version for various reasons.  I really don't need to run the application targeting the QEMU-RISV board, but just need to compile and link so I can get a ballpark figure on memory footprint and binary size.  Is there any way to simply suppress the building of the UART pipe code via some option in the kernel configuration files.  Attached as my general kernel and QEMU specific settings for reference.

----

```
# Project Configuration specific to QEMU builds

# Flash build options
# CONFIG_FLASH_SIMULATOR=y

# Random number generator
CONFIG_TEST_RANDOM_GENERATOR=y

# This setting in necessary for QEMU-RISCV, but will
# default to 'n' or external for NATIVE-POSIX
CONFIG_NEWLIB_LIBC=y

# Enable mostly-standards-compliant implementations of various POSIX (IEEE 1003.1) APIs.
# Necessary to support clock_gettime() for QEMU target
CONFIG_POSIX_API=y
```

------

```
# C++ support
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
CONFIG_STD_CPP2A=y
CONFIG_CBPRINTF_FP_SUPPORT=y

# General kernel options
CONFIG_INIT_STACKS=y

# fs_dirent structures are big
CONFIG_MAIN_STACK_SIZE=2048

# Size of heap memory pool
CONFIG_HEAP_MEM_POOL_SIZE=16384

# Co-operative threading disabled, want pre-emption
# A cooperative thread remains the current thread until
# it performs an action that makes it unready
CONFIG_NET_TC_THREAD_COOPERATIVE=n

# Base 64 conversions support for XML interface
CONFIG_BASE64=y

# Flash storage settings
# Optionally force the file system to be recreated
CONFIG_FLASH=y
CONFIG_FLASH_MAP=y
CONFIG_FLASH_PAGE_LAYOUT=y
CONFIG_FILE_SYSTEM=y
CONFIG_FILE_SYSTEM_LITTLEFS=y
CONFIG_FS_LOG_LEVEL_DBG=y
CONFIG_FLASH_SHELL=y
# CONFIG_APP_WIPE_STORAGE=y

# Networking config
# Need TCP for Telnet session for shell interface
CONFIG_NETWORKING=y
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y
CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4=n
CONFIG_NET_ARP=y
CONFIG_NET_NATIVE=y
CONFIG_NET_L2_ETHERNET=y
[... 150 lines cut ...]
# Allow for the legacy include paths (without the zephyr/ prefix)
CONFIG_LEGACY_INCLUDE_PATH=y

# Main thread priority
# Priority at which the initialization thread runs, including the start
# of the main() function. main() can then change its priority if desired.
# A cooperative thread has a negative priority value. Once it becomes the current thread,
# a cooperative thread remains the current thread until it performs an action that makes it unready.
#
# A preemptible thread has a non-negative priority value. Once it becomes the current thread, a
# preemptible thread may be supplanted at any time if a cooperative thread, or a preemptible
# thread of higher or equal priority, becomes ready.
CONFIG_MAIN_THREAD_PRIORITY=-2
# Instructs the kernel to initialize stack areas with a known value (0xaa)
CONFIG_INIT_STACKS=y

# Tweak the timing cycles per second to get approximately 1 second intervals
# For Native-POSIX: CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=1000000
#CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=1000000
CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=100000
CONFIG_SYS_CLOCK_TICKS_PER_SEC=100000

# Disable USB stack
CONFIG_USB_DEVICE_STACK=n
```

cfriedt, 2023-10-31:
> I am pegged to the 3.2.0 LTS version for various reasons. I really don't need to run the application targeting the QEMU-RISV board, but just need to compile and link so I can get a ballpark figure [...]

@bob2oneil 
`west build -p auto -b <board> <app> -- -DCONFIG_UART_PIPE=n`

cfriedt, 2023-10-31:
You can also use board-specific overlays. Create a new file `<app>/boards/<board>.conf` with `CONFIG_UART_PIPE=n`.

carriegong, 2023-10-31:
have the same issue but i do not have uart_pipe.c included in my project

cfriedt, 2023-10-31:
@carriegong - Kconfig options can be selected by default, even if it isn't enabled with e.g. `CONFIG_FOO=y`. It can definitely be *disabled* though by explicitly setting `CONFIG_FOO=n`.

To find out why a particular option is being enabled, I would suggest using menuconfig `west build -t menuconfig` (or `ninja -C build menuconfig`). This is very similar to the Linux kernel menuconfig.

Find the particular Kconfig option by pressing `/` to search.

Look up the help associated with a Kconfig option by pressing `?` after highlighting the option. The help screen shows why a particular Kconfig option is enabled or disabled in terms of Kconfig, defaults, conditions, etc.

carriegong, 2023-10-31:
my .conf only has 
CONFIG_ZTEST=y
CONFIG_ZTEST_NEW_API=y

CONFIG_EVENTS=y
CONFIG_SIDEWALK=y

cfriedt, 2023-10-31:
> my .conf only has CONFIG_ZTEST=y CONFIG_ZTEST_NEW_API=y
> 
> CONFIG_EVENTS=y CONFIG_SIDEWALK=y

Yes. Add `CONFIG_UART_PIPE=n`.

carriegong, 2023-10-31:
> > my .conf only has CONFIG_ZTEST=y CONFIG_ZTEST_NEW_API=y
> > CONFIG_EVENTS=y CONFIG_SIDEWALK=y
> 
> Yes. Add `CONFIG_UART_PIPE=n`.

same errors

`warning: QEMU_ICOUNT_SHIFT (defined at boards/Kconfig:67) was assigned the value '6' but got the
value ''. Check these unsatisfied dependencies: QEMU_ICOUNT (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_QEMU_ICOUNT_SHIFT and/or look up
QEMU_ICOUNT_SHIFT in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: FLASH_PAGE_LAYOUT (defined at drivers/flash/Kconfig:80) has direct dependencies FLASH_HAS_PAGE_LAYOUT && FLASH with value n, but is currently being y-selected by the following symbols:
 - STREAM_FLASH (defined at subsys/storage/stream/Kconfig:7), with value y, direct dependencies y (value: y)
Parsing /opt/nordic/ncs/v2.4.0/zephyr/Kconfig`

cfriedt, 2023-10-31:
Those are warnings.

carriegong, 2023-10-31:
> Those are warnings.

Is there a way to get rid of warnings and bypass? i believe i have 
`zephyr_library_compile_options(-Wno-error)`
in my Cmake

bob2oneil, 2023-11-01:
Some good suggestions, I tried both the command line suppression and adding it to the board specific configuration file, and I still get link issues and build errors following the compile as follows.

Any other ideas?

```
Executing task: west build -p always -b qemu_riscv64 /home/user/project/0dev/software/black -- -DCONFIG_UART_PIPE=n -DOVERLAY_CONFIG=/home/user/project/0dev/software/black/conf/qemu.conf 

-- west build: making build dir /home/user/project/0dev/software/black/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/user/project/0dev/software/black
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /home/user/.cache/zephyr
-- Zephyr version: 3.2.0 (/home/user/zephyrproject/zephyr)
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: qemu_riscv64
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.0 (/home/user/zephyr-sdk-0.15.0)
-- Found toolchain: zephyr 0.15.0 (/home/user/zephyr-sdk-0.15.0)
-- Found Dtc: /usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/user/zephyrproject/zephyr/boards/riscv/qemu_riscv64/qemu_riscv64.dts
-- Generated zephyr.dts: /home/user/project/0dev/software/black/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/user/project/0dev/software/black/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/user/project/0dev/software/black/build/zephyr/dts.cmake
Parsing /home/user/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/user/zephyrproject/zephyr/boards/riscv/qemu_riscv64/qemu_riscv64_defconfig'
Merged configuration '/home/user/project/0dev/software/black/prj.conf'
Merged configuration '/home/user/project/0dev/software/black/conf/qemu.conf'
Merged configuration '/home/user/project/0dev/software/black/build/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to '/home/user/project/0dev/software/black/build/zephyr/.config'
Kconfig header saved to '/home/user/project/0dev/software/black/build/zephyr/include/generated/autoconf.h'

warning: QEMU_ICOUNT_SHIFT (defined at boards/Kconfig:67) was assigned the value '6' but got the
value ''. Check these unsatisfied dependencies: QEMU_ICOUNT (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_QEMU_ICOUNT_SHIFT and/or look up
QEMU_ICOUNT_SHIFT in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: UART_PIPE (defined at drivers/serial/Kconfig:104) was assigned the value 'n' but got the
value 'y'. See http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_UART_PIPE and/or look up
UART_PIPE in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: FLASH_SHELL (defined at drivers/flash/Kconfig:46) was assigned the value 'y' but got the
value 'n'. Check these unsatisfied dependencies: FLASH_PAGE_LAYOUT (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_FLASH_SHELL and/or look up FLASH_SHELL in
the menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration
Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

warning: FLASH_PAGE_LAYOUT (defined at drivers/flash/Kconfig:54) was assigned the value 'y' but got
the value 'n'. Check these unsatisfied dependencies: FLASH_HAS_PAGE_LAYOUT (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_FLASH_PAGE_LAYOUT and/or look up
FLASH_PAGE_LAYOUT in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: FILE_SYSTEM_LITTLEFS (defined at subsys/fs/Kconfig.littlefs:6) was assigned the value 'y'
but got the value 'n'. Check these unsatisfied dependencies: FLASH_MAP (=n), FLASH_PAGE_LAYOUT (=n).
See http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_FILE_SYSTEM_LITTLEFS and/or look up
FILE_SYSTEM_LITTLEFS in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

warning: FLASH_MAP (defined at subsys/storage/flash_map/Kconfig:10) was assigned the value 'y' but
got the value 'n'. Check these unsatisfied dependencies: FLASH_HAS_DRIVER_ENABLED (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_FLASH_MAP and/or look up FLASH_MAP in the
menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration Values,
and Kconfig - Tips and Best Practices sections of the manual might be helpful too.
[... 40 lines cut ...]
/home/user/zephyrproject/zephyr/include/zephyr/device.h:104:39: error: '__device_dts_ord_DT_CHOSEN_zephyr_uart_pipe_ORD' undeclared here (not in a function)
  104 | #define DEVICE_NAME_GET(name) _CONCAT(__device_, name)
      |                                       ^~~~~~~~~
/home/user/zephyrproject/zephyr/include/zephyr/toolchain/common.h:132:26: note: in definition of macro '_DO_CONCAT'
[... cut ...]
      |         ^~~~~~~~~~~~~
[331/363] Building CXX object CMakeFiles/app.dir/src/app/app.cpp.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /snap/bin/cmake --build /home/user/project/0dev/software/black/build

 *  The terminal process "/usr/bin/bash '-c', 'west build -p always -b qemu_riscv64 /home/user/project/0dev/software/black -- -DCONFIG_UART_PIPE=n -DOVERLAY_CONFIG=/home/user/project/0dev/software/black/conf/qemu.conf'" terminated with exit code: 1. 

```

closed 2024-01-16 by github-actions[bot]