zephyrproject-rtos/zephyr #64477: RFC: Stable API Change: Rename hci_rpmsg to hci_ipc

carlescufi, 2023-10-27:
- Problem Description:
  The HCI driver in drivers/bluetooth/hci/rpmsg and the HCI sample in
  samples/bluetooth/hci_rpmsg are no longer appropriately named. Both now use
  the IPC subsystem, which can in turn use (but does not have to) RPMsg under
  the hood.
- Proposed Change (detailed):
  The proposal, as shown in Pull Request #64391, is to rename anything related
  to HCI and RPMsg to use the term IPC instead.
- Detailed RFC:
  - Function call changes
    None that are public
  - Device Tree changes (source and bindings)
    The following chose node has been changed:
    ``zephyr,bt-hci-rpmsg-ipc`` has been renamed to ``zephyr,bt-hci-ipc``
  - Kconfig option changes
    ``CONFIG_BT_RPMSG`` has been renamed to ``CONFIG_BT_HCI_IPC``
  - Sample names changes
    ``samples/bluetooth/hci_rpmsg`` has been renamed to `samples/bluetooth/hci_ipc``
- Dependencies:
  Users will need to use the new Kconfig option and Devicetree chosen node in
  their out of tree boards and applications.

koffes, 2023-10-30:
Will these changes break messaging between cores? E.g. if we flash core A after this change has been merged, and we have an older firmware on core B?

carlescufi, 2023-10-30:
> Will these changes break messaging between cores? E.g. if we flash core A after this change has been merged, and we have an older firmware on core B?

No, because there are no runtime changes at all. Everything is compile-time.

carlescufi, 2023-10-31:
Architecture WG:

- No objections

PR #64391 (2023-11-02): bluetooth: Rename rpmsg HCI driver and sample to ipc
The existing driver and sample:

- drivers/bluetooth/hci/rpmsg
- samples/bluetooth/hci_rpmsg

are no longer correctly named, since they now use the IPC subsystem to send and receive data. The IPC subsystem can use RPMsg as a transport, but that is one of several selectable backends.

I initially wanted to deprecated both the BT_RPMSG Kconfig option as well as the zephyr,bt-hci-rpmsg-ipc chosen node in Devicetree. However, this proved to be undoable in the case of the Kconfig option. This is because it's a choice option, and those have specia [... cut]
closed 2023-11-02 by jhedberg