zephyrproject-rtos/zephyr #64496: LE Audio: Reach 100% documentation coverage

Thalley, 2023-10-27:
**Is your enhancement proposal related to a problem? Please describe.**
Code/documentation coverage tracking has been enabled and can be see from https://docs.zephyrproject.org/api-coverage/latest/. 

LE Audio (https://docs.zephyrproject.org/api-coverage/latest/zephyr/bluetooth/audio/index.html) has been good coverage for most parts, but there's no reason why we wouldn't have 100% coverage in our documentation. 

**Describe the solution you'd like**
Add missing doxygen documentation for all functions, enum, defines, etc. so that we can report 100% coverage. 

**Describe alternatives you've considered**
N/A

**Additional context**
Ideally we'd be able to enable a check to ensure that we keep it at 100% afterwards.

henrikbrixandersen, 2023-10-27:
It would be nice to have all the current Doxygen warnings for LE Audio fixed at the same time.

Thalley, 2023-10-27:
> It would be nice to have all the current Doxygen warnings for LE Audio fixed at the same time.

I was not aware of any warnings :o Haven't built locally for quite a while, but I agree that any warnings should be fixed. 

Do you have any examples?

henrikbrixandersen, 2023-10-27:
> Do you have any examples?

Just a few ;)

```
WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:33: argument 'em' of command @param is not found in the argument list of BT_BAP_LC3_UNICAST_PRESET_8_1_1(_loc, _stream_context)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:33: argument 'stream_context' of command @param is not found in the argument list of BT_BAP_LC3_UNICAST_PRESET_8_1_1(_loc, _stream_context)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:34: found </em> tag without matching <em>

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:36: The following parameter of BT_BAP_LC3_UNICAST_PRESET_8_1_1(_loc, _stream_context) is not documented:

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:43: argument 'em' of command @param is not found in the argument list of BT_BAP_LC3_UNICAST_PRESET_8_2_1(_loc, _stream_context)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:43: argument 'stream_context' of command @param is not found in the argument list of BT_BAP_LC3_UNICAST_PRESET_8_2_1(_loc, _stream_context)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:44: found </em> tag without matching <em>

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:46: The following parameter of BT_BAP_LC3_UNICAST_PRESET_8_2_1(_loc, _stream_context) is not documented:

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:53: argument 'em' of command @param is not found in the argument list of BT_BAP_LC3_UNICAST_PRESET_16_1_1(_loc, _stream_context)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:53: argument 'stream_context' of command @param is not found in the argument list of BT_BAP_LC3_UNICAST_PRESET_16_1_1(_loc, _stream_context)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:54: found </em> tag without matching <em>

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:56: The following parameter of BT_BAP_LC3_UNICAST_PRESET_16_1_1(_loc, _stream_context) is not documented:

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:63: argument 'em' of command @param is not found in the argument list of BT_BAP_LC3_UNICAST_PRESET_16_2_1(_loc, _stream_context)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:63: argument 'stream_context' of command @param is not found in the argument list of BT_BAP_LC3_UNICAST_PRESET_16_2_1(_loc, _stream_context)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:66: found </em> tag without matching <em>

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:68: The following parameter of BT_BAP_LC3_UNICAST_PRESET_16_2_1(_loc, _stream_context) is not documented:

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:75: argument 'em' of command @param is not found in the argument list of BT_BAP_LC3_UNICAST_PRESET_24_1_1(_loc, _stream_context)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:75: argument 'stream_context' of command @param is not found in the argument list of BT_BAP_LC3_UNICAST_PRESET_24_1_1(_loc, _stream_context)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:76: found </em> tag without matching <em>

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:78: The following parameter of BT_BAP_LC3_UNICAST_PRESET_24_1_1(_loc, _stream_context) is not documented:

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:85: argument 'em' of command @param is not found in the argument list of BT_BAP_LC3_UNICAST_PRESET_24_2_1(_loc, _stream_context)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:85: argument 'stream_context' of command @param is not found in the argument list of BT_BAP_LC3_UNICAST_PRESET_24_2_1(_loc, _stream_context)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:88: found </em> tag without matching <em>

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:90: The following parameter of BT_BAP_LC3_UNICAST_PRESET_24_2_1(_loc, _stream_context) is not documented:

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:97: argument 'em' of command @param is not found in the argument list of BT_BAP_LC3_UNICAST_PRESET_32_1_1(_loc, _stream_context)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/audio/bap_lc3_preset.h:97: argument 'stream_context' of command @param is not found in the argument list of BT_BAP_LC3_UNICAST_PRESET_32_1_1(_loc, _stream_context)

[... cut ...]
WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/gatt.h:703: argument 'props' of command @param is not found in the argument list of BT_GATT_CHARACTERISTIC(_uuid, _props, _perm, _read, _write, _user_data)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/gatt.h:708: found </em> tag without matching <em>

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/gatt.h:717: The following parameter of BT_GATT_CHARACTERISTIC(_uuid, _props, _perm, _read, _write, _user_data) is not documented:
```

Thalley, 2023-10-27:
> Just a few ;)

Haha, yeah, those I've seen before.
2 comments:
1) Some of those are non LE Audio related:
```

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/gatt.h:703: argument 'em' of command @param is not found in the argument list of BT_GATT_CHARACTERISTIC(_uuid, _props, _perm, _read, _write, _user_data)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/gatt.h:703: argument 'props' of command @param is not found in the argument list of BT_GATT_CHARACTERISTIC(_uuid, _props, _perm, _read, _write, _user_data)

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/gatt.h:708: found </em> tag without matching <em>

WARNING: /home/hebad/Projects/zephyrproject/zephyr/include/zephyr/bluetooth/gatt.h:717: The following parameter of BT_GATT_CHARACTERISTIC(_uuid, _props, _perm, _read, _write, _user_data) is not documented:
```
2) They come from a bug in sphinx/doxygen as far as I know. It has been reported several times, but there's not much we can do about it besides waiting for a fix. The issue is when using `_<something> ` like `_props` in a macro that passes it onto a different macro. 

We _could_ remove the `_` prefix, but that would likely cause a whole lot of other issues.

henrikbrixandersen, 2023-10-27:
Thanks, I did not know that, I just see all the warnings every time I build the docs. 

I know not all of them are LE audio related, just copied the entire output in the hope someone would pay attention 😉

Thalley, 2023-10-27:
> I know not all of them are LE audio related, just copied the entire output in the hope someone would pay attention

I can't remember if there's a bug report for this somewhere or not. We've discussed it a few times in the Discord #documentation channel, but we may just have hoped that sphinx/doxygen would fix it themselves at some point...

henrikbrixandersen, 2023-11-08:
@Thalley I have opened https://github.com/zephyrproject-rtos/zephyr/pull/64964 for fixing the Doxygen warnings.

closed 2024-07-12 by nashif