zephyrproject-rtos/zephyr #64510: ESP32-S3-DevkitC not support SPIRAM

sam131208, 2023-10-27:
**Describe the bug**
  Build the project "samples\hello_world"  with CONFIG_ESP_SPIRAM=y  on ESP32-S3-DevkitC-1 board( esp32-s3-wroom-2 D3N32R8V).
  Flash and run, get message"psram: PSRAM ID read error: 0x00ffffff".

**To Reproduce**
Steps to reproduce the behavior:
1.  update to last zephyr
2.  cd zephyr
3.  west build -b esp32s3_devkitm samples\hello_world -- -DCONFIG_ESP_SPIRAM=y
4.  west flash
5.  run
6. See error

**Expected behavior**
Can read and write spiram.

**Impact**
Can't boot to zephyr os.

**Logs and console output**
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:2
load:0x3fcd0108,len:0x1650
load:0x403b6000,len:0xc68
load:0x403ba000,len:0x30e8
SHA-256 comparison failed:
Calculated: d15f82a851b9383c4e84bd0b59e9863733a138d47ff91656bdaa7301cb797dbf
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403b61f4
I (62) boot: ESP-IDF b3cb13f06-dirty 2nd stage bootloader
I (63) boot: compile time 23:37:03
I (63) boot: chip revision: 0
I (65) boot.esp32s3: Boot SPI Speed : 40MHz
I (70) boot.esp32s3: SPI Mode       : SLOW READ
I (75) boot.esp32s3: SPI Flash Size : 32MB
I (80) boot: Enabling RNG early entropy source...
I (85) boot: Partition Table:
I (89) boot: ## Label            Usage          Type ST Offset   Length
I (96) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (104) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (111) boot:  2 factory          factory app      00 00 00010000 00100000
I (119) boot: End of partition table
I (123) esp_image: segment 0: paddr=00010020 vaddr=00000020 size=0001ch (    28)                                                                             
I (131) esp_image: segment 1: paddr=00010044 vaddr=3fc8a690 size=00118h (   280)                                                                              load
I (140) esp_image: segment 2: paddr=00010164 vaddr=3fc8a7a8 size=00134h (   308)                                                                              load
I (148) esp_image: segment 3: paddr=000102a0 vaddr=40374000 size=02e54h ( 11860)                                                                              load
I (161) esp_image: segment 4: paddr=000130fc vaddr=00000000 size=0cf3ch ( 53052)                                                                             
I (184) esp_image: segment 5: paddr=00020040 vaddr=3c000040 size=019c0h (  6592)                                                                              map
I (187) esp_image: segment 6: paddr=00021a08 vaddr=00000000 size=0e5f0h ( 58864)                                                                             
I (211) esp_image: segment 7: paddr=00030000 vaddr=42030000 size=040e4h ( 16612)                                                                              map
I (218) boot: Loaded app from partition at offset 0x10000
I (219) boot: Disabling RNG early entropy source...
E (231) psram: PSRAM ID read error: 0x00ffffff
Failed to Initialize external RAM, aborting.
abort()

**Environment (please complete the following information):**
 - OS:  Windows
 - Toolchain: Zephyr SDK 0.16.3, python 3.10.11, west 1.2.0
 - last version zephyr
 - ESP32-S3-DevkitC-1 board ( esp32-s3-wroom-2 D3N32R8V)

rftafas, 2023-11-01:
Can you confirm if you are using an octal flash/ram device? This is not fully supported yet. Maybe you will need to check e-fuses.

tyalie, 2023-11-03:
What would be needed to be able to use the octal PSRAM?

closed 2024-01-17 by github-actions[bot]