zephyrproject-rtos/zephyr #64536: OpenThread Joiner fails with "Error 13: InvalidState"

koenvervloesem, 2023-10-29:
**Describe the bug**
When trying to start the Joiner role on a Thread device, it fails with "Error 13: InvalidState".

This is done with the latest Zephyr main tree, on an nRF52840 Dongle.

**To Reproduce**

I built the echo server example with the OpenThread overlay configuration, adding configuration options to enable the Joiner role and configure Mbed TLS:

```
west build -p auto -b nrf52840dongle_nrf52840 samples/net/sockets/echo_server -- -DOVERLAY_CONFIG="overlay-ot.conf" -DCONFIG_OPENTHREAD_JOINER=y -DCONFIG_MBEDTLS_HEAP_SIZE=16384 -DCONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=768
```

On my OpenThread Border router, which acts as the Commissioner, I run:

```
> commissioner start
Commissioner: petitioning
Done
Commissioner: active
> commissioner joiner add * J01NME
Done
```

After flashing the application to the nRF52840 Dongle, which acts as the Joiner, I run this on the dongle:

```
uart:~$ ot factoryreset
```

After the reset, I run:

```
uart:~$ ot ifconfig up
Done
uart:~$ ot joiner start J01NME
Error 13: InvalidState
```

I see the same "Error 13: InvalidState" error on this Joiner device when I let another board with the same firmware but with the `CONFIG_OPENTHREAD_COMMISSIONER=y` option run the Commissioner role instead of the OpenThread Border Router.

**Expected behavior**

I expected to see "Join success!" on the Joiner device and have it join the Thread network.

**Impact**
This makes commissioning a device to a Thread network non-functional.

**Environment (please complete the following information):**
- OS: Ubuntu 22.04 LTS
- Toolchain: Zephyr SDK
- Zephyr Commit SHA: [9ac8525](https://github.com/zephyrproject-rtos/zephyr/commit/9ac85254961012dd6f26962c2bc1cade8f2abd8d))

**Additional context**

Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/64222

koenvervloesem, 2023-10-30:
@pdgendt Thanks for thinking along. That doesn't make a difference, though.

pdgendt, 2023-10-30:
What does `ot role` print out?

The joiner `start` function only checks the "up" state and role:
```c++
    VerifyOrExit(Get<ThreadNetif>().IsUp() && Get<Mle::Mle>().GetRole() == Mle::kRoleDisabled,
                 error = kErrorInvalidState);
```

koenvervloesem, 2023-10-30:
Ok I did find a working procedure now:

```
uart:~$ ot thread stop
Done
[00:00:10.547,180] <inf> net_l2_openthread: State changed! Flags: 0x10002046 Current role: disabled
uart:~$ ot panid 0xffff
Done
[00:00:19.990,386] <inf> net_l2_openthread: State changed! Flags: 0x30008000 Current role: disabled
uart:~$ ot ifconfig
up
Done
uart:~$ ot state
disabled
Done
uart:~$ ot joiner start J01NME
Done
[00:00:36.621,459] <inf> net_l2_openthread: State changed! Flags: 0x0800800b Current role: disabled
[00:00:41.466,125] <inf> net_l2_openthread: State changed! Flags: 0x00008000 Current role: disabled
[00:00:42.828,308] <inf> net_l2_openthread: State changed! Flags: 0x0800c00b Current role: disabled
Join success
[00:00:47.244,079] <inf> net_l2_openthread: State changed! Flags: 0x08000000 Current role: disabled
[00:00:47.262,908] <inf> net_l2_openthread: State changed! Flags: 0x08000000 Current role: disabled
[00:00:47.325,531] <inf> net_l2_openthread: State changed! Flags: 0x18040100 Current role: disabled
[00:00:47.427,459] <inf> net_l2_openthread: State changed! Flags: 0x0800000b Current role: disabled
uart:~$ ot thread start
Done
[00:00:58.560,455] <inf> net_l2_openthread: State changed! Flags: 0x00001035 Current role: detached
[00:01:09.807,006] <inf> net_l2_openthread: State changed! Flags: 0x10000040 Current role: detached
[00:01:10.779,022] <inf> net_l2_openthread: State changed! Flags: 0x00000100 Current role: detached
[00:01:11.174,072] <inf> net_l2_openthread: State changed! Flags: 0x300b32b7 Current role: child
[00:01:11.174,468] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: child
```

Thanks for pushing me in the right direction!

rlubos, 2023-10-30:
@koenvervloesem Glad to see you've found the solution. If I understood correctly, the reason you had to "reset" the PAN ID, is that OT stack by default initializes and starts on boot, with Kconfig-provided parameters. You should likely take a look at [CONFIG_OPENTHREAD_MANUAL_START](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/l2/openthread/Kconfig#L49) Kconfig option, which prevents boot-time OT stack initialization.

koenvervloesem, 2023-10-30:
Thanks @rlubos! I'm closing the issue, it's not a bug but a configuration error.

closed 2023-10-30 by koenvervloesem