zephyrproject-rtos/zephyr #64553: PWM polarity DTS flag not working

matejanus, 2023-10-30:
**Describe the bug**

The PWM_POLARITY_INVERTED flag provided in the device tree doesn't make any difference to the output signal. 
In the example the duct cycle was set to 33%, the output  was expected to be  33% but it was 67% (inverted). The usage of the PWM_POLARITY_INVERTED flag has no effect. 

However channel A of the PWM controller works as expected. 

Device tree entry:

	aliases {
		pwm0 = &pwm_dev0;
	};

     pwmdevs {
       compatible = "pwm-leds";
       pwm_dev0: pwm_dev_0 {
        pwms = <&flexpwm1_pwm1 1 PWM_MSEC(10) PWM_POLARITY_NORMAL>;
       };
      };

    &flexpwm1_pwm1 {
	    status = "okay";
	    pinctrl-0 = <&pinmux_flexpwm1_pwm1>;
	    pinctrl-names = "default";
    };

    pinmux_flexpwm1_pwm1: pinmux_flexpwm1_pwm1 {
        group0 {
            pinmux = <&iomuxc_gpio_sd_b0_03_flexpwm1_pwmb1>;
            drive-strength = "r0-4";
            bias-pull-up;
            bias-pull-up-value = "47k";
            slew-rate = "slow";
            nxp,speed = "100-mhz";
        };
    };

**Code snipet**
    #include "zephyr/drivers/pwm.h"

    #define PWM_NODE DT_ALIAS(pwm0)
    
    #if !DT_NODE_HAS_STATUS(PWM_NODE, okay)
    #error "Wrong devicetree alias"
    #endif

    #define DUTY_CYCLE (33)
    int main() {
        pwm_dt_spec pwm = PWM_DT_SPEC_GET(PWM_NODE);
        if(!device_is_ready(pwm.dev))
            printk("device not ready");
    
        uint32_t highPulsePeriod = (pwm.period * DUTY_CYCLE) / 100U;
        if(pwm_set_dt(&pwm, pwm.period, highPulsePeriod) != 0)
            printk("pwm setting error");
    ...
    }

**To Reproduce**
Steps to reproduce the behavior:
MIMXRT1060-EVKB development board used
1. Enable PWM in the device tree
2. change the default PWM pin to e.g. iomuxc_gpio_sd_b0_03_flexpwm1_pwmb1 (anyone that uses channel B) 
3. When SD_B0_03 is used, solder R346
4. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
5. make
6. Check PWM signal with an osciloscope

**Expected behavior**
The inverted flag causes that the duty cycle gets inverted.

**Impact**
workaround in a PWM driver is needed only when the PWM channel B is used

**Logs and console output**
No errors from the PWM module.

PWM_POLARITY_NORMAL
[image]

PWM_POLARITY_INVERTED
[image]

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 23.04
 - Toolchain Zephyr SDK 0.16.1
 - Zephyr v3.4

MaureenHelm, 2023-10-31:
@mmahadevan108 @danieldegrasse fyi

danieldegrasse, 2024-01-18:
Hi @matejanus, the root cause here is due to the PWM_SetupPWM call here: https://github.com/zephyrproject-rtos/zephyr/blob/356c8cbe63ae01b3ab438382639d25bb418a0213/drivers/pwm/pwm_mcux.c#L98. You can see that the call hardcodes the `numOfChnls` parameter to `CHANNEL_COUNT`, which is defined as 2. For channel B, this will result in out of bounds array access (as only channels A and B exist). At least in my test application on Zephyr 3.4.0, it looks like the garbage data the out of bounds array access reads sets the `level` field of `chnlParams` to `kPWM_LowTrue`, so the PWM polarity for channel B is inverted here: https://github.com/zephyrproject-rtos/hal_nxp/blob/904830e8f684a9fd573751a1cdecde877ec49242/mcux/mcux-sdk/drivers/pwm/fsl_pwm.c#L468

The net result of this is what you see: the PWM polarity is always inverted for PWM channel B. This error was fixed here: 4eedd45c8. I'll go ahead and backport that fix to the 3.4 branch.

matejanus, 2024-01-20:
Hey @danieldegrasse, 

Thanks for the great work and your help

danieldegrasse, 2024-01-22:
> Thanks for the great work and your help

Sure- thanks for highlighting the issue. Since #67806 is merged, going to close this out.

closed 2024-01-22 by danieldegrasse