zephyrproject-rtos/zephyr #64554: drivers: can: mcan Filtering not possible for Regular and Fd frames for the same frame type

palisows, 2023-10-30:
*Descpription*
Can filters has been used to capture selected frames types, e.g. CAN B and CAN B FD.
CanId is not used for filtering - id and mask from can_filter are set to 0.
Filter is successfully applied with can_add_rx_filter, assigned filterId are 0 and 1.

When frame is captured, zephyr can't properly point out filt_idx in can_mcan_get_message.
It's always 0 not mater I send CAN B or CAN B FD frame.
Thus for CAN B FD frame pointed filter is for CAN B and frame is dropped.

As workaround conditions related to CAN_FRAME_FDF and CAN_FILTER_FDF could be removed from can_mcan_get_message.
It means FD flag will be ignored when filtering.

Problem is caused by HW limitation which is not able to distinguish filters when difference is only in filter flags.
Basically HW doesn't take it into consideration when matching frames.

I'm using zephyr version 3.5.0.
Platform is stm32 with can fd driver.

**To Reproduce**
Configure two filters with can_add_rx_filter and flags set to:
1. CAN_FILTER_DATA | CAN_FILTER_IDE
2. CAN_FILTER_DATA | CAN_FILTER_IDE | CAN_FILTER_FDF

id and mask are set to 0 for both cases.
Send both frame types and check **filt_idx** from **can_mcan_get_message**.
It's always 0 and points for first configured filter.

**Expected behavior**
For CAN B, filt_idx should 0,
for CAN B FD, filt_idx should 1.

**Impact**
I can workaround but at cost of lost performance.

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Toolchain: Zephyr SDK 0.16.3

**Additional context**
Order of filters can be reversed when configured, then CAN B FD frames will be accepted.

henrikbrixandersen, 2023-11-10:
@palisows Thanks, I am looking into a fix.

henrikbrixandersen, 2023-11-13:
@palisows Proposed fix provided as part of RFC #65109.

PR #65108 (2024-01-19): RFC: Breaking API change: drivers: can: remove CAN_FILTER_FDF flag
Remove the CAN_FILTER_FDF flag for filtering on classic CAN/CAN FD frames as it is not supported natively by any known CAN controller.
    
Applications can still filter on classic CAN/CAN FD frames in their receive callback functions as needed.
    
Fixes: #64554
RFC issue: #65109

~Note: This currently includes the changes from #65058 in order to remove the mix of software and hardware filtering in the Bosch M_CAN backend completely, which is what led to #64554.~
commit 75117a0d (2024-01-19): drivers: can: remove CAN_FILTER_FDF flag

Remove the CAN_FILTER_FDF flag for filtering on classic CAN/CAN FD frames
as it is not supported natively by any known CAN controller.

Applications can still filter on classic CAN/CAN FD frames in their receive
callback functions as needed.

Fixes: #64554
closed 2024-01-19 by henrikbrixandersen