zephyrproject-rtos/zephyr #64581: stm32h7: Linkable loadable extensions  test fails

FRASTM, 2023-10-30:
**Describe the bug**
Running the tests/subsys/llext  on the nucleo_h743zi fails when   `CONFIG_ARM_MPU=n`

$ west build -p auto -b nucleo_h743zi  tests/subsys/llext -- -DCONFIG_ARM_MPU=n

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_h753zi  tests/subsys/llext -- -DCONFIG_ARM_MPU=n
2.  west flash
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
```
Running TESTSUITE llext
===================================================================
START - test_llext_simple
[00:00:00.016,000] <dbg> llext: llext_load: Loading relocatable or shared elf
[00:00:00.023,000] <dbg> llext: do_llext_load: Finding ELF tables...
[00:00:00.030,000] <dbg> llext: llext_find_tables: section 0 at 268: name 0, type 0, flags 0, addr 0, size 0
[00:00:00.040,000] <dbg> llext: llext_find_tables: section 1 at 290: name 31, type 1, flags 6, addr 0, size 56
[00:00:00.051,000] <dbg> llext: llext_find_tables: section 2 at 2b8: name 27, type 9, flags 40, addr 0, size 24
[00:00:00.061,000] <dbg> llext: llext_find_tables: section 3 at 2e0: name 37, type 1, flags 3, addr 0, size 0
[00:00:00.071,000] <dbg> llext: llext_find_tables: section 4 at 308: name 43, type 8, flags 3, addr 0, size 0
[00:00:00.082,000] <dbg> llext: llext_find_tables: section 5 at 330: name 48, type 1, flags 2, addr 0, size 37
[00:00:00.092,000] <dbg> llext: llext_find_tables: section 6 at 358: name 56, type 1, flags 30, addr 0, size 33
[00:00:00.103,000] <dbg> llext: llext_find_tables: section 7 at 380: name 65, type 1879048195, flags 0, addr 0, s2
[00:00:00.114,000] <dbg> llext: llext_find_tables: section 8 at 3a8: name 1, type 2, flags 0, addr 0, size 240
[00:00:00.124,000] <dbg> llext: llext_find_tables: symtab at 8
[00:00:00.130,000] <dbg> llext: llext_find_tables: section 9 at 3d0: name 9, type 3, flags 0, addr 0, size 47
[00:00:00.141,000] <dbg> llext: llext_find_tables: strtab at 9
[00:00:00.147,000] <dbg> llext: llext_find_tables: section 10 at 3f8: name 17, type 3, flags 0, addr 0, size 81
[00:00:00.158,000] <dbg> llext: llext_find_tables: shstrtab at 10
[00:00:00.164,000] <dbg> llext: do_llext_load: Allocate and copy strings...
[00:00:00.171,000] <dbg> llext: do_llext_load: Mapping ELF sections...
[00:00:00.178,000] <dbg> llext: llext_map_sections: section 0 name 
[00:00:00.185,000] <dbg> llext: llext_map_sections: Not copied section 
[00:00:00.192,000] <dbg> llext: llext_map_sections: section 1 name .text
[00:00:00.199,000] <dbg> llext: llext_map_sections: section 2 name .rel.text
[00:00:00.207,000] <dbg> llext: llext_map_sections: Not copied section .rel.text
[00:00:00.214,000] <dbg> llext: llext_map_sections: section 3 name .data
[00:00:00.222,000] <dbg> llext: llext_map_sections: section 4 name .bss
[00:00:00.229,000] <dbg> llext: llext_map_sections: section 5 name .rodata
[00:00:00.236,000] <dbg> llext: llext_map_sections: section 6 name .comment
[00:00:00.243,000] <dbg> llext: llext_map_sections: Not copied section .comment
[00:00:00.251,000] <dbg> llext: llext_map_sections: section 7 name .ARM.attributes
[00:00:00.259,000] <dbg> llext: llext_map_sections: Not copied section .ARM.attributes
[00:00:00.267,000] <dbg> llext: llext_map_sections: section 8 name .symtab
[00:00:00.275,000] <dbg> llext: llext_map_sections: Not copied section .symtab
[00:00:00.282,000] <dbg> llext: llext_map_sections: section 9 name .strtab
[00:00:00.290,000] <dbg> llext: llext_map_sections: Not copied section .strtab
[00:00:00.297,000] <dbg> llext: llext_map_sections: section 10 name .shstrtab
[00:00:00.305,000] <dbg> llext: llext_map_sections: Not copied section .shstrtab
[00:00:00.312,000] <dbg> llext: do_llext_load: Allocate and copy sections...
[00:00:00.320,000] <dbg> llext: do_llext_load: Counting exported symbols...
[00:00:00.327,000] <dbg> llext: llext_count_export_syms: symbol count 15
[00:00:00.334,000] <dbg> llext: llext_count_export_syms: unhandled symbol 0, name , type tag 0, bind 0, sect 0
[00:00:00.345,000] <dbg> llext: llext_count_export_syms: unhandled symbol 1, name hello_world.c, type tag 4, bind1
[00:00:00.357,000] <dbg> llext: llext_count_export_syms: unhandled symbol 2, name , type tag 3, bind 0, sect 1
[00:00:00.367,000] <dbg> llext: llext_count_export_syms: unhandled symbol 3, name , type tag 3, bind 0, sect 3
[00:00:00.378,000] <dbg> llext: llext_count_export_syms: unhandled symbol 4, name , type tag 3, bind 0, sect 4
[00:00:00.388,000] <dbg> llext: llext_count_export_syms: unhandled symbol 5, name , type tag 3, bind 0, sect 5
[00:00:00.398,000] <dbg> llext: llext_count_export_syms: unhandled symbol 6, name $d, type tag 0, bind 0, sect 5
[00:00:00.409,000] <dbg> llext: llext_count_export_syms: unhandled symbol 7, name number, type tag 1, bind 0, sec5
[00:00:00.420,000] <dbg> llext: llext_count_export_syms: unhandled symbol 8, name $t, type tag 0, bind 0, sect 1
[00:00:00.431,000] <dbg> llext: llext_count_export_syms: unhandled symbol 9, name $d, type tag 0, bind 0, sect 1
[00:00:00.441,000] <dbg> llext: llext_count_export_syms: unhandled symbol 10, name $t, type tag 0, bind 0, sect 1
[00:00:00.452,000] <dbg> llext: llext_count_export_syms: unhandled symbol 11, name , type tag 3, bind 0, sect 6
[00:00:00.462,000] <dbg> llext: llext_count_export_syms: unhandled symbol 12, name , type tag 3, bind 0, sect 7
[00:00:00.473,000] <dbg> llext: llext_count_export_syms: function symbol 13, name hello_world, type tag 2, bind 11
[00:00:00.484,000] <dbg> llext: llext_count_export_syms: unhandled symbol 14, name printk, type tag 0, bind 1, se0
[00:00:00.495,000] <dbg> llext: do_llext_load: Allocating memory for symbol table...
[00:00:00.503,000] <dbg> llext: do_llext_load: Copying symbols...
[00:00:00.510,000] <dbg> llext: llext_copy_symbols: function symbol 0 name hello_world addr 0x240025cd
[00:00:00.520,000] <dbg> llext: do_llext_load: Linking ELF...
[00:00:00.526,000] <dbg> llext: llext_link: relocation section .rel.text (2) linked to section 8 has 3 relocations
[00:00:00.536,000] <dbg> llext: llext_link: relocation 2:0 info 502 (type 2, sym 5) offset 40 sym_name  sym_type 5
[00:00:00.549,000] <inf> llext: found section symbol  addr 0x24002610
[00:00:00.556,000] <inf> llext: relocating (linking) symbol  type 3 binding 0 ndx 5 offset 40 link section 8
[00:00:00.566,000] <inf> llext: writing relocation symbol  type 2 sym 5 at addr 0x240025f4 addr 0x24002610
[00:00:00.576,000] <dbg> llext: llext_link: relocation 2:1 info e02 (type 2, sym 14) offset 44 sym_name printk sy0
[00:00:00.589,000] <inf> llext: found symbol printk at 0x8006b55, updating op code 0x240025f8
[00:00:00.598,000] <inf> llext: relocating (linking) symbol printk type 0 binding 1 ndx 0 offset 44 link section 8
[00:00:00.609,000] <inf> llext: writing relocation symbol printk type 2 sym 14 at addr 0x240025f8 addr 0x8006b55
[00:00:00.620,000] <dbg> llext: llext_link: relocation 2:2 info 502 (type 2, sym 5) offset 48 sym_name  sym_type 5
[00:00:00.632,000] <inf> llext: found section symbol  addr 0x24002620
[00:00:00.639,000] <inf> llext: relocating (linking) symbol  type 3 binding 0 ndx 5 offset 48 link section 8
[00:00:00.649,000] <inf> llext: writing relocation symbol  type 2 sym 5 at addr 0x240025fc addr 0x24002620
[00:00:00.659,000] <dbg> llext: do_llext_load: loaded module, .text at 0x240025cc, .rodata at 0x2400260c
[00:00:00.669,000] <inf> llext: Loaded extension hello
[00:00:00.675,000] <err> os: ***** USAGE FAULT *****
[00:00:00.681,000] <err> os:   No coprocessor instructions
[00:00:00.687,000] <err> os: r0/a1:  0x00000024  r1/a2:  0x00000024  r2/a3:  0x00000029
[00:00:00.696,000] <err> os: r3/a4:  0xa6044730 r12/ip:  0x080021d1 r14/lr:  0x240025db
[00:00:00.704,000] <err> os:  xpsr:  0x01000000
[00:00:00.709,000] <err> os: Faulting instruction address (r15/pc): 0x24012a64
[00:00:00.717,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.725,000] <err> os: Current thread: 0x240000b0 (test_llext_simple)
[00:00:00.733,000] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

FRASTM, 2023-10-30:
when executing the samples/subsys/llext/shell_loader  
`$ west build -p auto -b nucleo_h753zi  samples/subsys/llext/shell_loader -- -DCONFIG_ARM_MPU=n`
Loading the tests/subsys/llext/hello_world/hello_world.c
does not cause exception but :
```
[00:03:55.014,000] <dbg> llext_shell: cmd_llext_load_hex: hex2bin hex len 2112, llext buf sz 8192, read 1056
[00:03:55.024,000] <dbg> llext_shell: cmd_llext_load_hex: 4 byte MAGIC
                                      7f 45 4c 46                                      |.ELF             
[00:03:55.040,000] <dbg> llext: llext_load: Loading relocatable or shared elf
[00:03:55.048,000] <dbg> llext: do_llext_load: Finding ELF tables...
[00:03:55.054,000] <dbg> llext: llext_find_tables: section 0 at 268: name 0, type 0, flags 0, addr 0, size 0
[00:03:55.065,000] <dbg> llext: llext_find_tables: section 1 at 290: name 31, type 1, flags 6, addr 0, size 56
[00:03:55.075,000] <dbg> llext: llext_find_tables: section 2 at 2b8: name 27, type 9, flags 40, addr 0, size 40
[00:03:55.086,000] <dbg> llext: llext_find_tables: section 3 at 2e0: name 37, type 1, flags 3, addr 0, size 0
[00:03:55.096,000] <dbg> llext: llext_find_tables: section 4 at 308: name 43, type 8, flags 3, addr 0, size 0
[00:03:55.106,000] <dbg> llext: llext_find_tables: section 5 at 330: name 48, type 1, flags 2, addr 0, size 37
[00:03:55.117,000] <dbg> llext: llext_find_tables: section 6 at 358: name 56, type 1, flags 30, addr 0, size 33
[00:03:55.127,000] <dbg> llext: llext_find_tables: section 7 at 380: name 65, type 1879048195, flags 0, addr 0, s2
[00:03:55.138,000] <dbg> llext: llext_find_tables: section 8 at 3a8: name 1, type 2, flags 0, addr 0, size 224
[00:03:55.149,000] <dbg> llext: llext_find_tables: symtab at 8
[00:03:55.155,000] <dbg> llext: llext_find_tables: section 9 at 3d0: name 9, type 3, flags 0, addr 0, size 47
[00:03:55.165,000] <dbg> llext: llext_find_tables: strtab at 9
[00:03:55.172,000] <dbg> llext: llext_find_tables: section 10 at 3f8: name 17, type 3, flags 0, addr 0, size 81
[00:03:55.182,000] <dbg> llext: llext_find_tables: shstrtab at 10
[00:03:55.189,000] <dbg> llext: do_llext_load: Allocate and copy strings...
[00:03:55.196,000] <dbg> llext: do_llext_load: Mapping ELF sections...
[00:03:55.203,000] <dbg> llext: llext_map_sections: section 0 name 
[00:03:55.210,000] <dbg> llext: llext_map_sections: Not copied section 
[00:03:55.217,000] <dbg> llext: llext_map_sections: section 1 name .text
[00:03:55.224,000] <dbg> llext: llext_map_sections: section 2 name .rel.text
[00:03:55.231,000] <dbg> llext: llext_map_sections: Not copied section .rel.text
[00:03:55.239,000] <dbg> llext: llext_map_sections: section 3 name .data
[00:03:55.246,000] <dbg> llext: llext_map_sections: section 4 name .bss
[00:03:55.253,000] <dbg> llext: llext_map_sections: section 5 name .rodata
[00:03:55.260,000] <dbg> llext: llext_map_sections: section 6 name .comment
[00:03:55.268,000] <dbg> llext: llext_map_sections: Not copied section .comment
[00:03:55.276,000] <dbg> llext: llext_map_sections: section 7 name .ARM.attributes
[00:03:55.284,000] <dbg> llext: llext_map_sections: Not copied section .ARM.attributes
[00:03:55.292,000] <dbg> llext: llext_map_sections: section 8 name .symtab
[00:03:55.299,000] <dbg> llext: llext_map_sections: Not copied section .symtab
[00:03:55.307,000] <dbg> llext: llext_map_sections: section 9 name .strtab
[00:03:55.314,000] <dbg> llext: llext_map_sections: Not copied section .strtab
[00:03:55.322,000] <dbg> llext: llext_map_sections: section 10 name .shstrtab
[00:03:55.329,000] <dbg> llext: llext_map_sections: Not copied section .shstrtab
[00:03:55.337,000] <dbg> llext: do_llext_load: Allocate and copy sections...
[00:03:55.345,000] <dbg> llext: do_llext_load: Counting exported symbols...
[00:03:55.352,000] <dbg> llext: llext_count_export_syms: symbol count 14
[00:03:55.359,000] <dbg> llext: llext_count_export_syms: unhandled symbol 0, name , type tag 0, bind 0, sect 0
[00:03:55.369,000] <dbg> llext: llext_count_export_syms: unhandled symbol 1, name hello_world.c, type tag 4, bind1
[00:03:55.381,000] <dbg> llext: llext_count_export_syms: unhandled symbol 2, name , type tag 3, bind 0, sect 1
[00:03:55.392,000] <dbg> llext: llext_count_export_syms: unhandled symbol 3, name , type tag 3, bind 0, sect 3
[00:03:55.402,000] <dbg> llext: llext_count_export_syms: unhandled symbol 4, name , type tag 3, bind 0, sect 4
[00:03:55.413,000] <dbg> llext: llext_count_export_syms: unhandled symbol 5, name , type tag 3, bind 0, sect 5
[00:03:55.423,000] <dbg> llext: llext_count_export_syms: unhandled symbol 6, name $d, type tag 0, bind 0, sect 5
[00:03:55.434,000] <dbg> llext: llext_count_export_syms: unhandled symbol 7, name number, type tag 1, bind 0, sec5
[00:03:55.444,000] <dbg> llext: llext_count_export_syms: unhandled symbol 8, name $a, type tag 0, bind 0, sect 1
[00:03:55.455,000] <dbg> llext: llext_count_export_syms: unhandled symbol 9, name $d, type tag 0, bind 0, sect 1
[00:03:55.466,000] <dbg> llext: llext_count_export_syms: unhandled symbol 10, name , type tag 3, bind 0, sect 6
[00:03:55.476,000] <dbg> llext: llext_count_export_syms: unhandled symbol 11, name , type tag 3, bind 0, sect 7
[00:03:55.487,000] <dbg> llext: llext_count_export_syms: function symbol 12, name hello_world, type tag 2, bind 11
[00:03:55.498,000] <dbg> llext: llext_count_export_syms: unhandled symbol 13, name printk, type tag 0, bind 1, se0
[... cut ...]
[00:03:55.648,000] <dbg> elf: arch_elf_relocate: Unsupported ARM elf relocation type 28 at address 24014bc8
[00:03:55.658,000] <dbg> llext: llext_link: relocation 2:2 info 28 (type 40, sym 0) offset 44 sym_name  sym_type 0
[00:03:55.671,000] <err> llext: Undefined symbol with no entry in symbol table , offset 44, link section 8
[00:03:55.681,000] <err> llext: Failed to link, ret -61
[00:03:55.687,000] <dbg> llext: do_llext_load: Failed to load extension, freeing memory...
Failed to load extension test, return code -61
```

MaureenHelm, 2023-10-31:
@teburd fyi

teburd, 2023-11-01:
This is new and I haven't been able to reproduce this. I don't have a stm32h7 to debug with.

My best guess is the compiler flags being used are just slightly different here *or* some particular region of sram needs to be used that isn't being used today for placing the extension.

FRASTM, 2023-11-02:
Testing with **CONFIG_DCACHE=n** makes the tests PASSED on this stm32h753 (or other stm32h7 board I tested : stm32h7b3i_dk or stm32h747i_disco_m7)

teburd, 2023-11-02:
Now that’s an interesting issue! I’ll think this over and see if I can replicate on another m7 part I have.

pillo79, 2023-11-10:
Hello @FRASTM, I was able to replicate the issue on another STM32H747 (Arduino Giga R1), but in my case `CONFIG_DCACHE=n` did not help and I kept having random errors at every `llext_call_fn`.
While I was tracking the issue, #64918 got merged and I was not able to replicate since 0bf08e5775.
Can you try to replicate on your setup again?

This issue is still worth investigating - code now runs directly from Flash in the test, so there's a major difference in cache handling.

pillo79, 2023-11-10:
@lyakh after merging the peek PR, this test actually passes fine on the H7 - but it's relocating patching a buffer _in the CPU's Flash memory range_ :open_mouth:

FRASTM, 2023-11-22:
still failing with c6bba39f4d8d4ac4148a59e5d8cdc0c26b9c9926
with the stm32h747_disco I have tested : 
west build -p auto -b stm32h747i_disco_m7 tests/subsys/llext -- -DCONFIG_ARM_MPU=n
```
*** Booting Zephyr OS build zephyr-v3.5.0-1907-gc6bba39f4d8d ***
Running TESTSUITE llext
===================================================================
START - test_llext_simple
[00:00:00.016,000] <inf> llext: found section symbol  addr 0x8009430
[00:00:00.022,000] <inf> llext: relocating (linking) symbol  type 3 binding 0 ndx 5 offset 40 link section 8
[00:00:00.033,000] <inf> llext: writing relocation symbol  type 2 sym 5 at addr 0x800941c addr 0x8009430
[00:00:00.043,000] <inf> llext: relocating (linking) symbol printk type 0 binding 1 ndx 0 offset 44 link section8
[00:00:00.053,000] <inf> llext: writing relocation symbol printk type 2 sym 14 at addr 0x8009420 addr 0x8007b65
[00:00:00.064,000] <inf> llext: found section symbol  addr 0x8009440
[00:00:00.071,000] <inf> llext: relocating (linking) symbol  type 3 binding 0 ndx 5 offset 48 link section 8
[00:00:00.081,000] <inf> llext: writing relocation symbol  type 2 sym 5 at addr 0x8009424 addr 0x8009440
[00:00:00.091,000] <inf> llext: Loaded extension hello
[00:00:00.096,000] <err> os: ***** USAGE FAULT *****
[00:00:00.102,000] <err> os:   Illegal use of the EPSR
[00:00:00.108,000] <err> os: r0/a1:  0x08009430  r1/a2:  0x08009bd9  r2/a3:  0x00000000
[00:00:00.116,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x080021a1 r14/lr:  0x08009403 
[00:00:00.125,000] <err> os:  xpsr:  0x20000000 
[00:00:00.130,000] <err> os: Faulting instruction address (r15/pc): 0x00000000 
[00:00:00.138,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0 
[00:00:00.146,000] <err> os: Current thread: 0x240000b8 (test_llext_simple) 
[00:00:00.154,000] <err> os: Halting system 
```

lyakh, 2023-11-28:
@pillo79 @FRASTM could somebody please check if https://github.com/zephyrproject-rtos/zephyr/files/13417167/40.gitdiff.txt fixes this problem for you? You might have to add your core type to the list in the patch, if it isn't one of Cortex-M3 or -M4

lyakh, 2023-11-28:
> @pillo79 @FRASTM could somebody please check if https://github.com/zephyrproject-rtos/zephyr/files/13417167/40.gitdiff.txt fixes this problem for you? You might have to add your core type to the list in the patch, if it isn't one of Cortex-M3 or -M4

in fact I think I know how to make this work again. First of all `CONFIG_ARM_MPU` needs to be disabled for this test, but you already know that. Secondly either the image has to be made writable, i.e. the `const` qualifier has to be removed as per https://github.com/zephyrproject-rtos/zephyr/pull/62433/commits/88cfae6f4c56f63dab9cb9751d0397fcb90b2cd5 or we need to copy the image as in the original implementation into a writable memory, which can be achieved by using a modified version of that diff: 
[47.gitdiff.txt](https://github.com/zephyrproject-rtos/zephyr/files/13487544/47.gitdiff.txt)

FRASTM, 2023-11-28:
thanks @lyakh, I confirm that removing the _const_ at 
`static uint8_t hello_world_elf[] __aligned(4) = {`
makes the test PASSED  (with CONFIG_ARM_MPU=n)

lyakh, 2023-11-28:
@FRASTM thanks for testing! I've now put that fix with a bit more flexibility as https://github.com/zephyrproject-rtos/zephyr/pull/62433/commits/94e5a1a97506996ed33bbc95a4d133dc146525a5 . ATM it's a part of #62433 but if we want to have it quickly, I can also push it as a stand-alone fix.

lyakh, 2023-12-04:
@FRASTM now that #62433 has been merged, can this be closed?

FRASTM, 2023-12-05:
Like the tests/subsys/llext/testcase.yaml does CONFIG_LLEXT_STORAGE_WRITABLE=y  for _xtensa_, platform, I expect same condition to apply to the stm32h7/f7

pillo79, 2023-12-05:
@FRASTM it is possible that applying #66144 + enabling `CONFIG_CACHE_MANAGEMENT` might solve this (it does in my case on STM32H7). Can you try if that PR helps?

FRASTM, 2023-12-20:
Closing now that PR https://github.com/zephyrproject-rtos/zephyr/pull/66476. is merged.

Running the test on https://github.com/zephyrproject-rtos/zephyr/commit/eb2cd314075bb3b28581c72bb777eee7ec5fd4bc
test is PASSED 
$ west build -p auto -b stm32h747i_disco_m7 tests/subsys/llext/ -- -DCONFIG_ARM_MPU=n
```

===================================================================                                   
TESTSUITE llext succeeded                                                                             
                                                                                                      
------ TESTSUITE SUMMARY START ------                                                                 
                                                                                                      
SUITE PASS - 100.00% [llext]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.868 seconds        
 - PASS - [llext.test_llext_simple] duration = 0.868 seconds                                          
                                                                                                      
------ TESTSUITE SUMMARY END ------                                                                   
                                                                                                      
===================================================================                                   
PROJECT EXECUTION SUCCESSFUL
```

closed 2023-12-20 by FRASTM