zephyrproject-rtos/zephyr #64605: [doc] Documentation issue in 'develop/application/index'

Asireesha, 2023-10-31:
**Describe the bug**

<< Please describe the issue here >>
<< You may also want to update the automatically generated issue title above. >>

**Environment**

* Page: `develop/application/index`
* Version: main
* SHA-1: 8f01cb81471f90bd60e274455583df62aec139fe

Asireesha, 2023-10-31:
tflite library is unable to add to zepyhr project

Asireesha, 2023-10-31:
i want to ai run  inference on microcontroller

nordicjm, 2023-10-31:
What issue is there in the documentation page you have linked to?

henrikbrixandersen, 2023-10-31:
> i want to ai run inference on microcontroller

Please see the following samples:
- https://docs.zephyrproject.org/latest/samples/modules/tflite-micro/hello_world/README.html
- https://docs.zephyrproject.org/latest/samples/modules/tflite-micro/magic_wand/README.html

closed 2023-10-31 by henrikbrixandersen