zephyrproject-rtos/zephyr #64606: twister: test: pytest -s scripts/tests/twister/test_runner.py fails

hakehuang, 2023-10-31:
**Describe the bug**
pytest -s scripts/tests/twister/test_runner.py fails

**To Reproduce**
Steps to reproduce the behavior:
1.  run below
 pytest -s scripts/tests/twister/test_runner.py

**Expected behavior**
test pass

**Impact**
CI

**Logs and console output**

```
scripts/pylib/twister/twisterlib/runner.py:268: TypeError
---------------------------------------------------------------------------------------------- Captured log call -----------------------------------------------------------------------------------------------DEBUG    twister:runner.py:247 Building <Mock name='mock.testsuite.source_dir' id='140469417446368'> for <Mock name='mock.platform.name' id='140469417456064'>
=========================================================================================== short test summary info ============================================================================================FAILED scripts/tests/twister/test_runner.py::test_projectbuilder_gather_metrics[none] - TypeError: cannot unpack non-iterable Mock object
FAILED scripts/tests/twister/test_runner.py::test_projectbuilder_gather_metrics[size_report] - TypeError: cannot unpack non-iterable Mock object
FAILED scripts/tests/twister/test_runner.py::test_projectbuilder_gather_metrics[cmake] - TypeError: cannot unpack non-iterable Mock object
FAILED scripts/tests/twister/test_runner.py::test_projectbuilder_gather_metrics[size_report+cmake] - TypeError: cannot unpack non-iterable Mock object
======================================================================================== 4 failed, 128 passed in 1.91s =========================================================================================ubuntu@ubuntu-OptiPlex-7050:/home/shared/disk/zephyr_project/zephyr_test/zephyr$ 

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v3.5.0-770-gaa3e79a88a

yperess, 2023-11-07:
I'm still seeing this in ToT

yperess, 2023-11-07:
Actually, looking at the logs, the error is different, creating a separate issue.

closed 2023-11-07 by yperess

yperess, 2023-11-07:
https://github.com/zephyrproject-rtos/zephyr/issues/64940

PR #64615 (2023-10-31): twister: unittests: Fix unittest test_projectbuilder_gather_metrics
Fix #64606 by initializing mock object.
commit 0958abd2 (2023-10-31): twister: unittests: Fix unittest test_projectbuilder_gather_metrics

Fix #64606 by initializing mock object.