zephyrproject-rtos/zephyr #64633: fixes for mmu compilation warnings when memory size is 0

ycsin, 2023-10-31:
**Describe the bug**

The following code would generate compilation warnings if:
- `CONFIG_SRAM_BASE_ADDRESS` is 0, or
- `CONFIG_SRAM_BASE_ADDRESS` is 0 and `CONFIG_SRAM_SIZE` is 0, or
- `CONFIG_KERNEL_VM_BASE` is 0, or
- `CONFIG_KERNEL_VM_BASE` is 0 and `CONFIG_KERNEL_VM_SIZE` is 0

as the assert test would be essentially comparing unsigned int against 0, which will be escalated into error in Twister, see [PR63172](https://github.com/zephyrproject-rtos/zephyr/actions/runs/6705912230/job/18221509407?pr=63172).

https://github.com/zephyrproject-rtos/zephyr/blob/67676d8fbcf5d742bdc6bbf0635d79499fa66062/include/zephyr/sys/mem_manage.h#L193-L199

https://github.com/zephyrproject-rtos/zephyr/blob/67676d8fbcf5d742bdc6bbf0635d79499fa66062/include/zephyr/sys/mem_manage.h#L202-L209

https://github.com/zephyrproject-rtos/zephyr/blob/67676d8fbcf5d742bdc6bbf0635d79499fa66062/include/zephyr/sys/mem_manage.h#L226-L232

**To Reproduce**
Steps to reproduce the behavior:
1. See https://github.com/zephyrproject-rtos/zephyr/actions/runs/6705912230/job/18221509407?pr=63172

**Expected behavior**
Should compile without error nor warnings regardless of the configuration

**Impact**
Blocking #63172

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 639d1f803b

MaureenHelm, 2023-10-31:
@andyross please raise the priority if needed

ycsin, 2023-10-31:
Update:

This issue only exists in #63172 and can't repro on `main`, see https://github.com/zephyrproject-rtos/zephyr/pull/64630#issuecomment-1787639513

nashif, 2023-10-31:
> Update:
> 
> This issue only exists in #63172 and can't repro on `main`, see [#64630 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/64630#issuecomment-1787639513)

then we should close this as a non bug

ycsin, 2023-11-01:
> > Update:
> > This issue only exists in #63172 and can't repro on `main`, see [#64630 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/64630#issuecomment-1787639513)
> 
> then we should close this as a non bug

This might still be a bug, #63172 probably just uncovered it - somehow the compiler decides to compile that part of code when `sw_isr_tables.h` includes `device.h`

PR #64630 (2023-11-02): kernel: mmu: fix compilation warnings when memory size is 0
When these memory sizes are defined to be zero, the assertions test will be comparing an unsigned int against zero which result in compilation warning, and will be raised to error in Twister.

Fix them with more conditional compilations, see also #63080

fixes #64633
PR #64736 (2023-11-10): [Backport v3.5-branch] kernel: mmu: fix compilation warnings when memory size is 0
Backport 3ed8c4d64b362df00372f44d15f44ee827ccd710 from #64630.

Fixes #64633
closed 2023-11-02 by carlescufi