zephyrproject-rtos/zephyr #64643: Sample sample.llext.shell causes bss overflow on nrf5340dk_nrf5340_cpunet

de-nordic, 2023-10-31:
**Describe the bug**
Build of the mentioned sample causes huge RAM overflow of 56960 bytes.
```
west build -p -b nrf5340dk_nrf5340_cpunet zephyr/samples/subsys/llext/shell_loader -T sample.llext.shell
...
[1/161] Preparing syscall dependency handling

[3/161] Generating include/generated/version.h
-- Zephyr version: 3.5.99 (/repos/doer/zephyrproject-clean/zephyr), build: zephyr-v3.4.0-6171-g482416db01e5
[156/161] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /repos/doer/zephyrproject-clean/build/zephyr/zephyr_pre0.map
: && ccache /home/doer/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/repos/doer/zephyrproject-clean/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/arch/arch/arm/core/cortex_m/cmse/libarch__arm__core__cortex_m__cmse.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/lib/posix/liblib__posix.a  zephyr/subsys/llext/libsubsys__llext.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/mbox/libdrivers__mbox.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/doer/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v8-m.main/nofp"  -L/repos/doer/zephyrproject-clean/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m33+nodsp  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -DPICOLIBC_INTEGER_PRINTF_SCANF  --specs=picolibc.specs  -lc  -lgcc && cd /repos/doer/zephyrproject-clean/build/zephyr && /home/doer/.local/lib/python3.10/site-packages/cmake/data/bin/cmake -E true
/home/doer/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/zephyr_pre0.elf section `bss' will not fit in region `RAM'
/home/doer/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: region `RAM' overflowed by 56960 bytes
```

**To Reproduce**
Steps to reproduce the behavior:
1. Get into wherever you can see the Zephyr repo dir
2. west build -p -b nrf5340dk_nrf5340_cpunet zephyr/samples/subsys/llext/shell_loader -T sample.llext.shell
3. be baffled

**Expected behavior**
No overflow

**Impact**
Showstopper

**Logs and console output**
Can not run this.

**Environment (please complete the following information):**
 - OS: Linux local-bm 5.15.0-87-generic #97-Ubuntu SMP Mon Oct 2 21:09:21 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
 - Toolchain sdk 0.16.1
 - Commit SHA 482416db01e5b8d3e8b91c52bb9daa6eb3d8cddd

teburd, 2023-11-01:
I did make the heap/stack sizes quite large. This also prevents the sample from running on a mimxrt1010evk board I have which is also pretty sram limited. I'll try adjusting things a bit and running on there.

teburd, 2023-11-07:
was this fixed with #64707 ? I don't have this hardware to test on

closed 2023-11-08 by carlescufi