zephyrproject-rtos/zephyr #64646: k_thread_abort() can deadlock on SMP systems

peter-mitsis, 2023-10-31:
**Describe the bug**

There is a corner case in SMP where a deadlock can occur when terminating threads via k_thread_abort() from an ISR. 

**To Reproduce**

It requires two cores both simultaneously executing at ISR level and attempting to terminate the other core's interrupted thread. Below is the sequence of events. (This sequence will need to be updated slightly after PR #59400 is merged, but it would still result in a deadlock).

1. CPU0 executing thread0, CPU1 executing thread1
2. CPU0 enters ISR, CPU1 executing thread1
3. CPU0 ISR doing stuff, CPU1 enters ISR
4. CPU0 ISR issues k_thread_abort(thread1), CPU1 ISR issues k_thread_abort(thread0)
5. (Each CPU is now trying to abort the other's interrupted thread)
6. CPU0 ISR gets sched_lock, CPU1 ISR spins on sched_lock
7. CPU0 ISR determines thread1 is not aborting, CPU1 still spinning on sched_lock
8. CPU0 ISR sets thread1's _THREAD_ABORTING, CPU1 still spinning on sched_lock
9. CPU0 ISR determines that thread1 is aborting and it's not _current, CPU1 still spinning on sched_lock
10. CPU0 ISR determines it is in ISR, CPU1 still spinning on sched_lock
11. CPU0 ISR unlocks sched_lock, CPU1 gets sched_lock
12. CPU0 ISR spins while thread1's _THREAD_ABORTING is set, CPU1 ISR determines thread0 is not aborting
13. CPU0 ISR continues to spin, CPU1 ISR sets thread0's _THREAD_ABORTING
14. CPU0 ISR continues to spin, CPU1 ISR determines that thread0 is aborting and it's not _current
15. CPU0 ISR continues to spin, CPU1 ISR determines that it is in ISR
16. CPU0 ISR continues to spin, CPU1 ISR unlocks sched_lock
17. CPU0 ISR continues to spin, CPU1 ISR spins on thread0's _THREAD_ABORTING

Both CPU0 ISR and CPU1 ISR are deadlocked waiting for threads to abort

**Expected behavior**

Deadlock should not occur.

One possible solution to this would be to add some code to detect the deadlock and then act accordingly. This could be modifying the k_thread_abort() API to return -EDEADLOCK and let the caller handle it. Or it could be offloading the k_thread_abort() from ISR level to a high priority/metaIRQ thread that terminates it.

There may be yet another alternate solution.

**Impact**

Although this could hang the system, the likelihood is very low as it requires very specific circumstances (SMP + two or more cores in an ISR + each of those ISRs trying to terminate the other core's interrupted current thread)

It does not affect uniprocessor systems.

andyross, 2023-10-31:
Cool, thanks.  Just commenting to register my opinion that this isn't actually possible given that the aborting flag is only set/checked from within the scheduler lock, which serializes access to the various k_thread_abort() invocations.   But who knows, this stuff is hard.

FWIW it wouldn't be too terribly difficult to write a stress test as a reproducer: two threads sit in a loop spawning/joining one other "victim" thread each.  The victims then (1) do a irq_offload() call (or some other trap, but this is easiest) to an ISR handler that is responsible for aborting the other victim thread then (2) exit.

Run this for a few hours overnight across a bunch of instances (half as many as you have host CPU cores -- it does need to be reliably getting two CPUs for SMP).  If there's a deadlock, it should show up.  And if not we can probably write it off as "so rare it's below the noise level of other failures", which is something at least.

peter-mitsis, 2023-10-31:
Attaching a test case to replicate the issue. This test spawns two threads (one for each core). Each thread enters an ISR via irq_offload(). The two ISRs sync and then issue a call to k_thread_abort() to abort the other core's interrupted thread.

[Debug.zip](https://github.com/zephyrproject-rtos/zephyr/files/13220182/Debug.zip)

Once the project has been unzipped ...

```
$ cd <project>
$ west build -p -b qemu_x86_64 ./ -t run
```

Observed output ...

```
*** Booting Zephyr OS build zephyr-v3.5.0-728-g9fd0185c3b40 ***
isr_var1: 0x11e26d 0x10a000 0x11e26c
Spawned thread 0x10a360 is waiting
IRQ is syncing 0x11e26d
IRQ is waiting on 0x11e26c
isr_var2: 0x11e26c 0x10a360 0x11e26d
Spawned thread 0x10a000 is waiting
IRQ is syncing 0x11e26c
IRQ is waiting on 0x11e26d
IRQ is killing 0x10a360
IRQ is killing 0x10a000

```

Additional output is not observed. In a success case, we are expecting to see additional output that is something like ...

```
IRQ is done with 0x10a360
IRQ is done with 0x10a000
Final message!

```

If additional printk() statements are added into sched.c, we can see that we deadlock in z_thread_abort() -- soon to be z_thread_halt().

peter-mitsis, 2023-11-08:
Fixed by PR #59400

peter-mitsis, 2023-11-08:
Was thinking this was something else. Re-opening.

peter-mitsis, 2024-01-08:
Not forgotten ... just haven't got around to it yet.

peter-mitsis, 2024-04-23:
This issue is fixed by #71185.

commit 61c70626 (2024-04-25): kernel/sched: Fix free-memory write when ISRs abort _current

After a k_thread_abort(), the resulting thread struct is documented as
unused/free memory that may be re-used (for example, to respawn a new
thread).

But in the special case of aborting the current thread from within an
ISR, that wasn't quite happening.  The scheduler cleanup would
complete, but the architecture layer would still try to context switch
away from the aborted thread on exit, and that can include writes to
the now-reused thread struct!  The specifics will depend on
architecture (some do a full context save on entry, mo [... cut]
commit f0fd54cb (2024-05-02): kernel/sched: Fix free-memory write when ISRs abort _current

After a k_thread_abort(), the resulting thread struct is documented as
unused/free memory that may be re-used (for example, to respawn a new
thread).

But in the special case of aborting the current thread from within an
ISR, that wasn't quite happening.  The scheduler cleanup would
complete, but the architecture layer would still try to context switch
away from the aborted thread on exit, and that can include writes to
the now-reused thread struct!  The specifics will depend on
architecture (some do a full context save on entry, mo [... cut]
closed 2024-04-25 by carlescufi