zephyrproject-rtos/zephyr #64649: Bluetooth: Host: bt_att: already processing a REQ/RSP on chan xxx

axelnxp, 2023-10-31:
**Describe the bug**
We are currently evaluating Zephyr BLE on an internal platform and we just rebased on 3.5.0 tag from 3.4.0.
Since then, we are facing a regression on BLE, related to the recent rework of the buffer reuse for the att rsp.

Here is a log when running the peripheral_ht sample:
```
Bluetooth initialized
no temperature device; using simulated data
Advertising successfully started
Connected
[00:21:05.993,860] <err> bt_att: already processing a REQ/RSP on chan 0x30046d80
[00:21:05.993,874] <err> bt_att: already processing a REQ/RSP on chan 0x30046d80
[00:21:05.993,880] <err> bt_att: unable to allocate buf for error response
```

We've been able to identify the commit causing the regression on our side: 828be80986405c13d46d8fdfeaea8c87081861a4 

As a workaround, disabling the buffer re-use in `bt_att_chan_create_pdu` solves the problem.
```
        switch (att_op_get_type(op)) {
	case ATT_RESPONSE:
		/* Use a timeout only when responding */
		timeout = BT_ATT_TIMEOUT;
		// re_use = true;
		break;
	case ATT_CONFIRMATION:
		timeout = BT_ATT_TIMEOUT;
		break;
	default:
		timeout = K_FOREVER;
	}
```

Is there anything we may have missed ? A new configuration or requirement to make it work properly ?

Thanks !

kruithofa, 2023-11-02:
@axelnxp can you provide the steps required to reproduce this issue? In particular: are both sides Zephyr

carlescufi, 2023-11-16:
@yeaissa could you verify that https://github.com/zephyrproject-rtos/zephyr/pull/65272 fixes this issue as well?

yeaissa, 2023-11-16:
Yes it fixes the issue

PR #65272 (2023-11-17): Bluetooth: ATT: Revert re-use of RX buffer
The idea seemed good at the time :/

Reverting because:
- It doesn't work when `ATT_ENFORCE_FLOW=n`, and more controllers than I thought are actually using this. See #64967.
- It can result in a deadlock when not using `BT_RECV_BLOCKING` and the controller doesn't give events out of order. This is the case of the Softdevice controller
- It does create even more complexity, and I think we already have too much.

So instead, I will work on fixing the root cause, which is allowing the user to more finely control the allocation of buffers, in combination  [... cut]
closed 2023-11-17 by carlescufi