zephyrproject-rtos/zephyr #64661: drivers: console can leave characters in UART buffer if using server hooks

mgolu, 2023-10-31:
When using `CONFIG_UART_CONSOLE_DEBUG_SERVER_HOOKS`, if there are multiple characters in the buffer when the ISR is called, not all characters are processed if the server hook returns non-zero (i.e. indicates that the console shouldn't process this character).

I believe this line, which currently is `return;` should be `continue;` so that the `while` loop continues with the next character:

https://github.com/zephyrproject-rtos/zephyr/blob/d89938f4b978c3a9a7f8f711b0b65c7cca983d3e/drivers/console/uart_console.c#L477

henrikbrixandersen, 2023-11-07:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

mgolu, 2023-11-07:
Thank you for pointing out the template, I updated the issue.

henrikbrixandersen, 2023-11-07:
Thank you. I see you are referencing NCS v2.5.0. Can you reproduce this issue on mainline Zephyr? If not, please report it against NCS using the Nordic DevZone.

mgolu, 2023-11-07:
Yes, the issue is also in main:
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/console/uart_console.c#L477

henrikbrixandersen, 2023-11-07:
@anangl Please raise the priority (and perhaps reassign) if this affects multiple platforms.

carlescufi, 2023-11-08:
@mgolu could you please open a Pull Request with the change you suggest in the description (i.e. replace `return` with `continue`)? Thanks!

PR #64963 (2023-11-09): drivers: console: fix dropped characters when using debug hooks
When using console debug server hooks, not all characters are processed if the server hook returns non-zero for one character while there are other characters in the buffer. This is seen when using a fast console (like USB) where multiple characters come in before the ISR is called. Fix it by continuing to process characters instead of returning from the ISR with characters still in the buffer.

Fixes: #64661
commit 45f664ab (2023-11-09): drivers: console: fix dropped characters when using debug hooks

When using console debug server hooks, not all characters are
processed if the server hook returns non-zero for one character
while there are other characters in the buffer. This is seen
when using a fast console (like USB) where multiple characters
come in before the ISR is called. Fix it by continuing to
process characters instead of returning from the ISR with
characters still in the buffer.

Fixes: #64661
closed 2023-11-09 by carlescufi