zephyrproject-rtos/zephyr #64671: Logging: Build warning on subsys/logging/backends/log_multidomain_backend.c with gcc12

aescolar, 2023-11-01:
**Describe the bug**
log_multidomain_backend.c builds with warnings on gcc12
`log_multidomain_backend.c:41:57: warning: taking address of packed member of ‘union log_multidomain_msg_data’ may result in an unaligned pointer value `

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf5340bsim_nrf5340_cpuapp -DAPP_DIR=../samples/subsys/logging/multidomain/ ../share/sysbuild/
3. ninja
4. See warning

**Expected behavior**
No build warnings

**Impact**
* Twister failures when using gcc 12 or newer when this sample is built for this target
* Build warnings when the multidomain logger is used.

**Logs and console output**
```
[43/158] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/logging/backends/log_multidomain_backend.c.obj
zephyr/subsys/logging/backends/log_multidomain_backend.c: In function ‘process’:
zephyr/subsys/logging/backends/log_multidomain_backend.c:41:57: warning: taking address of packed member of ‘union log_multidomain_msg_data’ may result in an unaligned pointer value [-Waddress-of-packed-member]
   41 |         struct log_msg *out_log_msg = (struct log_msg *)out_msg->data.log_msg.data;
      |                                                         ^~~~~~~
```

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Toolchain (e.g Zephyr SDK, ...): Host gcc 12.3.0
 - Commit SHA or Version used: main as of today d89938f4b978c3a9a7f8f711b0b65c7cca983d3e

lopsided98, 2024-04-13:
This was fixed by #70272

aescolar, 2024-04-15:
Thanks @lopsided98 closing as fixed.

closed 2024-04-15 by aescolar