zephyrproject-rtos/zephyr #64675: [riscv-privileged] PLIC is not compliant with the Spec

amr-sc, 2023-11-01:
**Describe the bug**

RISC-V PLIC driver no more conforms to the Spec after the 8db1a5a, see https://github.com/zephyrproject-rtos/zephyr/pull/63273

Trigger Type register is not defined in the recent PLIC Spec and so is not compliant with RISC-V Privileged Standard: https://github.com/riscv/riscv-plic-spec/blob/master/riscv-plic-1.0.0.pdf
IMHO this is a miss in the Spec and it results in a situation when all the vendors who support Triggering type configuration implement it in a proprietary way, using the custom offsets and custom codes for the triggering types.

So until the Trigger Type becomes a standard, it would be nice if we leave a room for flexibility here: by default PLIC should be compliant with the Spec while the Trigger Type register could be optionally enabled  as an extension for particular platforms. 

**Environment**
Zephyr v3.5

* Page: `project/issues`
* Version: main
* SHA-1: 8db1a5add221b5e47ad4e01dfcb3b4064a8299f4

ycsin, 2023-11-01:
This is currently being escalated, see https://github.com/riscv/riscv-plic-spec/issues/52, are you using some PLIC that has a different offset for the trigger type registers?

amr-sc, 2023-11-01:
Hi @ycsin , 

> This is currently being escalated, see [riscv/riscv-plic-spec#52](https://github.com/riscv/riscv-plic-spec/issues/52),

Thanks! Nice to know there is already a ticket for the RISC-V Spec.

>  are you using some PLIC that has a different offset for the trigger type registers?

Right, we have some similar register called IRQ triggering mode and it has different offset: 
https://github.com/syntacore/zephyr/blob/sc/main-3.4/soc/riscv/riscv-privileged/syntacore-scr/common/plic.c

We implemented it as a proprietary SoC-specific extension for a regular standard PLIC. 
It's not yet upstreamed but we are working in this direction

PR #65761 (2023-12-08): drivers: intc: plic: set edge-triggered register address using compat
Define the edge-trigger register base address based on whether the PLIC node in the devicetree has an additional compatible that supports edge-triggered interrupt.

Limited the implementation to Andes NCEPLIC100 only, updated the devicetree binding of `andes_v5_ae350` accordingly.

**Caveat**: Per-instance trigger register address isn't supported, if one of the PLIC instances has `"andestech,plic"`, it assumes that all PLIC instances have the same type

Fixes #64675
closed 2023-12-08 by cfriedt