zephyrproject-rtos/zephyr #64680: bt_shell:  bt scan on then bt disable without bt scan off, it causes CPU hang

songnxp, 2023-11-01:
Describe the bug
bt scan on then bt disable without bt scan off, it causes CPU hang

To Reproduce
execute the following steps:

Open tera term
Reset the DUT
Enter 'help' command
Enter 'bt'
Start bluetooth initilization 'bt init'
Start bt scan --> 'bt scan on'
Perform bluetooth disable --> 'bt disable'
Expected behavior
CPU should not hang since some customer doesn't have a standard operation step. they may directly disable bt without scan off
Impact
CPU stuck

Logs and console output
uart:$ bt init
Bluetooth initialized
Settings Loaded
[00:00:04.993,249] fs_nvs: 8 Sectors of 4096 bytes
[00:00:04.993,264] fs_nvs: alloc wra: 0, f88
[00:00:04.993,268] fs_nvs: data wra: 0, 52
[00:00:05.253,252] bt_hci_core: No ID address. App must call settings_load()
[00:00:05.256,851] bt_hci_core: Identity: C0:95:DA:01:1A:50 (public)
[00:00:05.256,874] bt_hci_core: HCI: version 5.1 (0x0a) revision 0x8300, manufacturer 0x0025
[00:00:05.256,888] bt_hci_core: LMP: version 5.3 (0x0c) subver 0x144e
uart:$ bt scan on
Bluetooth active scan enabled
[DEVICE]: FD:E1:E9:46:B4:BA (random), AD evt type 0, RSSI -91 C:1 S:1 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 us), SID: 0xff
[DEVICE]: FD:E1:E9:46:B4:BA (random), AD evt type 4, RSSI -91 Seos C:1 S:1 D:0 SR:1 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 us), SID: 0xff
[DEVICE]: 3E:D3:14:7D:1D:E1 (random), AD evt type 3, RSSI -57 C:0 S:0 D:0 SR:0 E:0 Prim: LE 1M, Secn: No packets, Interval: 0x0000 (0 us), SID: 0xff
uart:~$ bt disable
[DEV

Environment (please complete the following information):

OS: (e.g. Linux, MacOS, Windows)
Toolchain (e.g Zephyr SDK, ...)
Commit SHA or Version used: zephyr-v3.4.0-3926-g4ba97c22555e

cvinayak, 2023-11-03:
@songnxp Please provide the details of the board used, and steps to build.

zhaynxp, 2023-11-06:
@cvinayak , board being used was NXP Redfinch(RW61X) device, issue was easy to recreate: bt init --> bt scan on --> bt disable, then DUT hang occured, almost 100% reproducible. 
We're using AOSP Shell app of Zephyr, it should be not board issue. Could you please try with a board you have handy?
FYI, no hang issue issue seen if enter bt scan off cmd before bt disable: bt init --> bt scan on --> bt scan off --> bt disable.
Could you please let us know root cause of this Hang issue, and solution to fix it?

closed 2024-01-21 by github-actions[bot]