zephyrproject-rtos/zephyr #64691: can_sam.c fails to compile when CONFIG_NETWORKING is set

kendag, 2023-11-01:
**Describe the bug**
Attempting to build an application with both CONFIG_CAN and CONFIG_NETWORKING enabled produces a compilation error.

**To Reproduce**
`west build -b sam_e70_xplained tests/drivers/can/api/ -- -DCONFIG_NETWORKING=y
`
**Expected behavior**
The test harness should build without error.

**Impact**
Currently unable to add CAN functionality to my application which also uses networking.

**Logs and console output**
```
drivers/can/can_sam.c:169:44: error: section of 'can_sam_mram_0' conflicts with previous declaration
  169 |         CAN_MCAN_DT_INST_MRAM_DEFINE(inst, can_sam_mram_##inst);        \
      |                                            ^~~~~~~~~~~~~
```

**Environment (please complete the following information):**
 - OS: Linux 6.3.12-200.fc38.x86_64
 - Toolchain: Zephyr SDK 0.16.3
 - Version: zephyr-v3.5.0

**Additional context**

henrikbrixandersen, 2023-11-07:
@nandojve FYI

henrikbrixandersen, 2023-11-14:
This is a clash between `__no_init` and `__no_cache` resulting in different sections with `CONFIG_NOCACHE_MEMORY=y`. Looking into a solution.

henrikbrixandersen, 2023-11-14:
@kendag Please test the proposed fix in #65169 and report back if it solves your issue. It compiles here and should work, but I do not have the hardware for testing.

PR #65169 (2023-11-16): linker: allow tagging variables with __nocache_noinit and update Bosch M_CAN driver to use it
- Allow tagging variables with `__nocach_noinit`. With `CONFIG_NOCACHE_MEMORY=y`, this will resolve to `__nocache`, which implies `__noinit`. With `CONFIG_NOCACHE_MEMORY=n`, this simply resolves to `__noinit`.
- Use `__nocache_noinit` for the Bosch M_CAN MRAM data variables on SoCs without dedicated MRAM.

Fixes: #64691
commit 58e1963c (2023-11-16): drivers: can: mcan: use __nocache_noinit for MRAM data variables

Use __nocache_noinit for the Bosch M_CAN MRAM data variables on SoCs
without dedicated MRAM.

Fixes: #64691
PR #65277 (2023-11-21): [Backport v3.5-branch] linker: allow tagging variables with __nocache_noinit and update Bosch M_CAN driver to use it
Backport 58e1963c6b7a7d6966244be9b89fb099aa6d66d7~2..58e1963c6b7a7d6966244be9b89fb099aa6d66d7 from #65169.

Fixes: #64691
commit 8415778a (2023-11-21): drivers: can: mcan: use __nocache_noinit for MRAM data variables

Use __nocache_noinit for the Bosch M_CAN MRAM data variables on SoCs
without dedicated MRAM.

Fixes: #64691

(cherry picked from commit 58e1963c6b7a7d6966244be9b89fb099aa6d66d7)
closed 2023-11-16 by carlescufi