zephyrproject-rtos/zephyr #64705: PM subsystem bug when using Meta IRQ threads

dennisgr102, 2023-11-01:
**Describe the bug**

Meta IRQ threads can be scheduled even if the scheduler is locked via `k_sched_lock()`, which is exactly what the PM subsystem uses before putting the system to sleep. Because of this, the idle thread which is used to put the system to sleep becomes considered non-preemptable via `base.sched_locked` going negative. When a non-preemptable thread gets preempted by a Meta IRQ a pointer to the thread is stored in the `metairq_preempted` field of the cpu structure. Upon the Meta IRQ thread completing, the stored thread in `metairq_preempted` will be scheduled regardless of the other threads which are ready to run. If `metairq_preempted` happens to be the idle thread the system will just go back sleep even though other threads may be ready to run.

The question here becomes "How does a Meta IRQ thread get scheduled in between the PM subsystem, which is called from the idle thread, calling `k_sched_lock()` and `k_sched_unlock()` given interrupts should be disabled?" Taking a look at the [sequence](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/pm/pm.c#L234) in `pm_system_suspend()` we see the following (I added some comments):

```
	k_sched_lock(); //<--- Scheduler is locked and the idle thread becomes non-preemptable
	pm_stats_start();
	/* Enter power state */
	pm_state_notify(true);
	atomic_set_bit(z_post_ops_required, id);
	pm_state_set(z_cpus_pm_state[id].state, z_cpus_pm_state[id].substate_id); 
	pm_stats_stop();

	/* Wake up sequence starts here */
#if defined(CONFIG_PM_DEVICE) && !defined(CONFIG_PM_DEVICE_RUNTIME_EXCLUSIVE)
	if (atomic_add(&_cpus_active, 1) == 0) {
		pm_resume_devices();
	}
#endif
	pm_stats_update(z_cpus_pm_state[id].state);
	pm_system_resume(); // <--- Point of interest discussed below
	k_sched_unlock(); // <--- Scheduler is unlocked and the idle thread becomes preemptable once again
```

[`pm_system_resume`](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/pm/pm.c#L134) calls `pm_state_exit_post_ops` which is where interrupts are reenabled by the power code specific to SOCs. See some examples here:

https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arm/nxp_imx/rt5xx/power.c#L79
https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arm/nxp_imx/rt6xx/power.c#L55
https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arm/st_stm32/stm32g0/power.c#L80
https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arm/microchip_mec/mec172x/power.c#L174

If an interrupt is pending it will be called in between `pm_system_resume()` and `k_sched_unlock`and thus it is possible for a Meta IRQ thread to become ready to run and scheduled before the idle thread has called `k_sched_unlock()` resulting in scenario where the idle thread is guaranteed to run after the Meta IRQ thread completes.

**To Reproduce**

Can provide an application which I used to reproduce on the NXP rt685 EVK board if helpful.

**Expected behavior**

Idle thread should not be rescheduled following Meta IRQ completion if there are higher priority threads which are ready to run. I think this is mostly a consequence of the PM subsystem enabling interrupts prematurely.

**Impact**

System will go to sleep when there are threads which are ready to run.

jhedberg, 2025-01-22:
It looks like this is still an issue. We just noticed that after taking MetaIRQs into use with #84130 the Silabs HCI driver is now broken with PM enabled.

jhedberg, 2025-01-22:
@npitre @andyross I think it's good for you to be in the loop for this one too, since it relates to the scheduler.

This issue, together with #80574 means that we now have to bugs with the MetaIRQ that make it unusable in practice. Purely from an upstream project perspective I feel like there are only two acceptable options: either prioritize fixing these for the next release, or remove the MetaIRQ feature completely. I don't think it's good for the project to ship a feature that's broken like this.

PR #84436 (2025-01-27): kernel: sched: Fix meta-IRQ preemption tracking for the idle thread
When the PM subsystem is enabled, the idle thread locks the scheduler for the duration the system is suspended. If a meta-IRQ preempts the idle thread in this state, the idle thread is tracked in `metairq_preempted`. However, when returning from the preemption, the idle thread is not removed from `metairq_preempted`, unlike all the other threads. As a result, the scheduler keeps running the idle thread even if there are higher priority threads ready to run.

This change treats the idle thread the same way as all other threads  [... cut]
commit a2eb78c0 (2025-01-27): kernel: sched: Fix meta-IRQ preemption tracking for the idle thread

When the PM subsystem is enabled, the idle thread locks the scheduler for
the duration the system is suspended. If a meta-IRQ preempts the idle
thread in this state, the idle thread is tracked in `metairq_preempted`.
However, when returning from the preemption, the idle thread is not removed
from `metairq_preempted`, unlike all the other threads. As a result, the
scheduler keeps running the idle thread even if there are higher priority
threads ready to run.

This change treats the idle thread the same way as all other threads [... cut]
closed 2025-01-27 by kartben