zephyrproject-rtos/zephyr #64713: USB DFU Example not working on nrf52840

Steampunkery, 2023-11-02:
**Describe the bug**
Following the guide [here](https://docs.zephyrproject.org/latest/samples/subsys/usb/dfu/README.html) I am unable to update the device image using DFU. I have successfully flashed MCUBoot and the usb-dfu example. However, when I attempt to use `dfu-util` to update the image to a signed blinky example, I get the following output:

```
thomasokeeffe@Wirecat:~/Documents/school/drexel/research/CTS_sensor/zephyrproject$ dfu-util --alt 1 --download build
-blinky-signed/zephyr/zephyr.signed.bin                                                                             
dfu-util 0.9                                                                                                        
                                                                                                                    
Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.                                                  
Copyright 2010-2016 Tormod Volden and Stefan Schmidt                                                                This program is Free Software and has ABSOLUTELY NO WARRANTY                                                        
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/                                                    
                                                                                                                    
dfu-util: Invalid DFU suffix signature                                                                              
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!            
dfu-util: No DFU capable USB device available
```

Please also mention any information which could help others to understand
the problem you're facing:
 - Target platform: nrf52840dk_nrf52840

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
I expected the device to be discovered by `dfu-util` and update the image successfully.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04, kernel 6.5.0
 - Toolchain: Zephy SDK 0.16.3
 - Commit SHA: 6597a65b23ff87ee526644c407dbbff7d257743a

JPHutchins, 2023-11-03:
Hi @Steampunkery!  Try the patch I mentioned here and report back with your findings: https://github.com/zephyrproject-rtos/zephyr/issues/64756

Independent from that, try the CONFIG_USB_DFU_WILL_DETACH=y setting.

Also, I suggest `dfu-tool` over `dfu-util`: `sudo apt install fwupd`

jfischer-no, 2023-11-03:
@Steampunkery I cannot reproduce it. Tested on an OS not much different from yours. Note that when you follow the sample documentation, you will need to reflash the USB DFU sample, as it is gone after downloading the hello-world image and resetting the board.

JPHutchins, 2023-11-03:
@Steampunkery Can you check the values of `CONFIG_USB_DEVICE_PID` and `CONFIG_USB_DEVICE_DFU_PID` in `build/zephyr/.config`?

Steampunkery, 2023-11-04:
Yes, I'll try all of these solutions when I get a chance this upcoming week.

JPHutchins, 2023-11-04:
A few more thoughts

1) do you see the device listed with command lsusb?
2) how about dfu-util -l

Steampunkery, 2023-11-07:
> A few more thoughts
> 
>     1. do you see the device listed with command lsusb?
> 
>     2. how about dfu-util -l

Heres the outputs:
```
$ dfu-util -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 009: ID 5986:9102 Acer, Inc BisonCam,NB Pro
Bus 003 Device 002: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 013: ID 1366:1051 SEGGER J-Link
Bus 001 Device 002: ID 04f3:0c63 Elan Microelectronics Corp. ELAN:Fingerprint
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Steampunkery, 2023-11-07:
@JPHutchins I check on the values of the configs you asked about in my dfu example build. Here are the results:

```
CONFIG_USB_DFU_WILL_DETACH=y
CONFIG_USB_DEVICE_PID=0x0005
CONFIG_USB_DEVICE_DFU_PID0xFFFF
```

JPHutchins, 2023-11-07:
@Steampunkery do you see your board in lsusb?

Keep in mind that the example project uses MCUBoot to swap images, not to do the USB DFU.  The example does not enable USB DFU from within MCUBoot, though MCUBoot does support it and I’ve tested it.

that said, the output that you are seeing is probably from the application, after MCUBoot has loaded it.  USB DFU is performed from the application, not MCUBoot.

JPHutchins, 2023-11-07:
@Steampunkery If you are on a Nordic DK make sure that the port labeled “nrf USB” is plugged in.  That is the 52840 USB PHY.  The other USB port is a JLink VCOM convenience thing.

Likewise if you are on custom HW make sure that your USB PHY is wired correctly and connected to the host OS.

Steampunkery, 2023-11-07:
Ok, I might have been connecting my microusb to the wrong port. Connecting it to the nRF USB Port, I now see `Bus 001 Device 015: ID 2fe3:0005 NordicSemiconductor Zephyr DFU sample` in `lsusb`. Using `sudo dfu-util -l` I can see `Found DFU: [2fe3:0005] ver=0305, devnum=15, cfg=1, intf=0, path="1-4", alt=0, name="UNKNOWN", serial="3989A5C43971E08B"`. However, the output of `sudo dfu-util --alt 1 --download build-blinky-signed/zephyr/zephyr.signed.bin` is still:

```
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
```

JPHutchins, 2023-11-07:
@Steampunkery good to plug both in at once so that you have logs.

I think you’ll get this working.  If dfu-util found the DFU device then the next command should have still found it!

jfischer-no, 2023-11-08:
> Ok, I was connecting my microusb to the wrong port. Connecting it to the nRF USB Port, I now see `Bus 001 Device 015: ID 2fe3:0005 NordicSemiconductor Zephyr DFU sample` in `lsusb`. Using `sudo dfu-util -l` I can see `Found DFU: [2fe3:0005] ver=0305, devnum=15, cfg=1, intf=0, path="1-4", alt=0, name="UNKNOWN", serial="3989A5C43971E08B"`. However, the output of `sudo dfu-util --alt 1 --download build-blinky-signed/zephyr/zephyr.signed.bin` is still:

Please include a full overview of the commands you use to build/flash what samples and upload/download signed images.

Steampunkery, 2023-11-08:
Here are the build commands for mcuboot, the dfu example, and the blinky project respectively:

```
west build -b nrf52840dk_nrf52840 bootloader/mcuboot/boot/zephyr -d build-mcuboot
west build -b nrf52840dk_nrf52840 zephyr/samples/subsys/usb/dfu -d build-dfu -- -DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE=\"bootloader/mcuboot/root-rsa-2048.pem\"
west build -b nrf52840dk_nrf52840 zephyr/samples/basic/blinky -d build-blinky-signed/ -- -DCONFIG_BOOTLOADER_MCUBOOT=y '-DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE="bootloader/mcuboot/root-rsa-2048.pem"'
```

Here are the flashing commands for mcuboot and the dfu example:

```
west flash -d build-mcuboot
west flash -d build-dfu/
```

Here are the commands that I have tried for DFU once I plugged the microUSB into the correct port:

```
sudo dfu-tool write-alt build-blinky-signed/zephyr/zephyr.signed.hex 1
sudo dfu-util --alt 1 --download build-blinky-signed/zephyr/zephyr.signed.bin
```

All building and flashing commands were successful.

JPHutchins, 2023-11-08:
> All building and flashing commands were successful.

Is this issue resolved?

Steampunkery, 2023-11-08:
No, I still get the following error when trying to use `dfu-util`:

```
$ sudo dfu-util --alt 1 --download build-blinky-signed/zephyr/zephyr.signed.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
```

```
$ sudo dfu-util -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [2fe3:0005] ver=0305, devnum=5, cfg=1, intf=0, path="1-3", alt=0, name="UNKNOWN", serial="3989A5C43971E08B"
```
A shot in the dark, trying to use dfu to alt 0:

```
$ sudo dfu-util --alt 0 --download build-blinky-signed/zephyr/zephyr.signed.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 2fe3:0005
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 128
Copying data from PC to DFU device
Download        [                         ]   0%            0 bytes failed!
state(10) = dfuERROR, status(3) = Device is unable to write memory
```

JPHutchins, 2023-11-08:
@Steampunkery That's a good error cause I don't think we'd want to overwrite the running slot in a dual slot config!

Still, very weird that it only finds the device with the --alt 0.  Can you look through my notes here and see if anything stands out to you? https://github.com/zephyrproject-rtos/zephyr/discussions/64767

jfischer-no, 2023-11-09:
> west flash -d build-mcuboot
> west flash -d build-dfu/

Please add logs from the Zephyr RTOS side, you see after each of the image programming.

Steampunkery, 2023-11-10:
Hi, I rebuilt MCUBoot with the `-p` (`--pristine`) flag, and it is working now. I think the problem was that I applied the patch and rebuilt, but for some reason the build system couldn't tell that the dil had been changed.

closed 2023-11-10 by jfischer-no

JPHutchins, 2023-11-13:
@Steampunkery Could you kindly suggest what you believe resolved the issue!  Thanks!

Steampunkery, 2023-11-13:
> Hi, I rebuilt MCUBoot with the `-p` (`--pristine`) flag, and it is working now. I think the problem was that I applied the patch and rebuilt, but for some reason the build system couldn't tell that the dil had been changed.

@JPHutchins This is what I believe solved the issue.

JPHutchins, 2023-11-13:
@Steampunkery From that comment it is unclear whether you are building with or without the patch.

Steampunkery, 2023-11-13:
I applied the patch, then rebuilt. It works now after applying the patch. I had to do a pristine build.