zephyrproject-rtos/zephyr #64728: Error when trying to install requirements.txt

AhmedHaisaraQuasar, 2023-11-02:
**Describe the bug**
I get an error when I run the command present in the getting started tutorial :pip install -r %HOMEPATH%\zephyrproject\zephyr\scripts\requirements.txt
The python hidapi refuse to build.
[image]

 - I am on a windows 10 pc and only try to buil the blinky example for a STM32 L4R5ZI-P
 - i try to install the hidapi alone with : pip install hidapi but i got the same error

**To Reproduce**
Steps to reproduce the behavior:
1. Only follow the différent step of the getting started guide
2. See de error

**Expected behavior**
simply work

**Impact**
i can't lanch the flash with jlink

**Logs and console output**
Building wheels for collected packages: hidapi
  Building wheel for hidapi (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for hidapi (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [9 lines of output]
      running bdist_wheel
      running build
      running build_ext
      C:\Users\Haidara\AppData\Local\Temp\pip-build-env-eaffgs92\normal\Lib\site-packages\Cython\Compiler\Main.py:381: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: C:\Users\Haidara\AppData\Local\Temp\pip-install-z5my6fqu\hidapi_3bfe4b8bfc9f4548949669b0dc48d845\hid.pyx
        tree = Parsing.p_module(s, pxd, full_module_name)
      Compiling hid.pyx because it changed.
      [1/1] Cythonizing hid.pyx
      building 'hid' extension
      error: unknown file type '.pxd' (from 'chid.pxd')
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for hidapi
Failed to build hidapi
ERROR: Could not build wheels for hidapi, which is required to install pyproject.toml-based projects

 - OS: ( Windows)
 - Toolchain (e.g Zephyr SDK,)

nordicjm, 2023-11-02:
Switch to python 3.11

kartben, 2023-11-02:
@AhmedHaisaraQuasar Thanks for taking the time to report the issue, and sorry that you had trouble with the recently released version of Python 3.12. Unfortunately some binary packages are not available from pypi and make for a troublesome install experience. As stated by @nordicjm, please use Python 3.11 for now.

```diff
-            choco install ninja gperf python git dtc-msys2 wget 7zip
+            choco install ninja gperf python311 git dtc-msys2 wget 7zip
```

PR #64744 will be updating the instructions to have folks install python311 instead of just "python".

closed 2023-11-03 by carlescufi