zephyrproject-rtos/zephyr #64757: ESP32S3: Octal SPI RAM support

tyalie, 2023-11-03:
Currently it is not possible to use the ESP32S3 variants with 8MB PSRAM. See issue #64510.
For me it breaks the use of PSRAM on the XIAO ESP32S3 boards I want to use.

The solution would be to be able to use PSRAM with octal spi.

lgl88911, 2023-11-12:
Any plans to support PSRAM for ESP32-S3? It's an important feature for this platform.

rftafas, 2023-11-29:
@lgl88911 Octal PSRAM should be available when Zephyr starts supporting [hal 5.1](https://github.com/zephyrproject-rtos/hal_espressif/tree/hal_v5.1) - we are taking care of some regressions to make it usable as 'pre-alfa' in order to share a WIP / Draft PR.

laurin, 2024-01-11:
Hey, is this still the latest status?

rftafas, 2024-01-11:
@laurin yes, it is. We are wrapping up the last issues on hal5. You can try the current wip: https://github.com/zephyrproject-rtos/zephyr/pull/65942

jadejr, 2024-07-18:
This is the PR for hal 5.1 that ended up being merged.

So, is this issue still valid? If not, then if you were using 3.7, how would you enable it?

rftafas, 2024-08-08:
PTAL `CONFIG_SPIRAM_MODE_OCT`

https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_SPIRAM_MODE_OCT

You also need to enable PSRAM normally and do all other proper configurations.

Even though the implementation has bugs, I think this can be closed.

EricNRS, 2025-01-08:
I am actively using octal SPIRAM on an ESP32-S3 N8R16 module using the `xiao_esp32s3/esp32s3/procpu` target and it works fine, so I think this issue can be closed now that HAL 5.x has been integrated.

FacepalmMute, 2025-01-10:
I have tested this with mcuboot as 2nd bootloader on my ESP32S3-N32R8V and got this error:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:2
load:0x3fcb5400,len:0x2bc8
load:0x403ba400,len:0xa91c
load:0x403c6400,len:0x15f4
entry 0x403bd89c
I (93) boot: MCUboot 2nd stage bootloader
I (93) boot: compile time Jan 10 2025 12:40:29
W (93) boot: Unicore bootloader
W (93) spi_flash: Octal flash chip is using but dio mode is selected, will automatically swich to Octal mode
I (102) spi_flash: detected chip: mxic (opi)
I (106) spi_flash: flash io: opi_str
W (109) spi_flash: Detected size(32768k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (122) boot: chip revision: v0.1
I (125) boot.esp32s3: Boot SPI Speed : 40MHz
I (129) boot.esp32s3: SPI Mode       : FAST READ
I (133) boot.esp32s3: SPI Flash Size : 32MB
I (137) boot: Enabling RNG early entropy source...
W (175) spi_flash: Octal flash chip is using but dio mode is selected, will automatically swich to Octal mode
E (176) spi_flash: failed to get chip size
```

Can anyone confirm this?

EricNRS, 2025-01-10:
That may be an octal flash issue and not PSRAM.  A similar issue was opened yesterday here:  https://github.com/zephyrproject-rtos/zephyr/issues/83743

PR #85625 (2025-02-14): soc: espressif: rework code startup to improve flash and spiram loader
This PR reworks all Espressif's SOC initialization code to fix booting issues regaring OCTAL FLASH/SPIRAM.
This also updates bootloader (either Simple Boot and MCUBoot) to use ROM functions during flash read/write operations, ensuring no dependencies of the flash driver initialization.
Runner handler is now common to all SoCs and fixes flash mode and frequency configuration, required for some flash memories.
Esptool is also updated to v4.8.1.

Fixes #64757
Fixes #85246
Fixes #83743
Fixes #82244