zephyrproject-rtos/zephyr #64758: arm64: Crash when mapping read-only area

gramsay0, 2023-11-03:
The following example crashes on the `phycore_am62x_a53` board, although it seems to work on `qemu_cortex_a53` board...

Example:

    static uint8_t rw_section[CONFIG_MMU_PAGE_SIZE] __aligned(CONFIG_MMU_PAGE_SIZE);
    uintptr_t rw_virt = 0;
    printk("Mapping RW section %p %zu\n", rw_section, sizeof(rw_section));
    z_phys_map((uint8_t **)&rw_virt,
               (uintptr_t)rw_section, sizeof(rw_section),
               K_MEM_CACHE_WB | K_MEM_PERM_RW);
    printk("Mapped RW section to %p\n", (void *)rw_virt);

    static uint8_t ro_section[CONFIG_MMU_PAGE_SIZE] __aligned(CONFIG_MMU_PAGE_SIZE);
    uintptr_t ro_virt = 0;
    printk("Mapping RO section %p %zu\n", ro_section, sizeof(ro_section));
    z_phys_map((uint8_t **)&ro_virt,
               (uintptr_t)ro_section, sizeof(ro_section),
               K_MEM_CACHE_WB);
    printk("Mapped RO section to %p\n", (void *)ro_virt);

Output:

	Mapping RW section 0xfc00b000 4096
	Mapped RW section to 0xfc7ed000
	Mapping RO section 0xfc00a000 4096
	[00:00:00.015,000] <err> os: ELR_ELn: 0x00000000fbffb7d4
	[00:00:00.021,000] <err> os: ESR_ELn: 0x000000009600014f
	[00:00:00.027,000] <err> os:   EC:  0x25 (Data Abort taken without a change in Exception level)
	[00:00:00.036,000] <err> os:   IL:  0x1
	[00:00:00.040,000] <err> os:   ISS: 0x14f
	[00:00:00.045,000] <err> os: FAR_ELn: 0x00000000fc7ec000
	[00:00:00.051,000] <err> os: TPIDRRO: 0x01000000fc015398
	[00:00:00.057,000] <err> os: x0:  0x00000000fc7ec000  x1:  0x00000000fc7ed000
	[00:00:00.065,000] <err> os: x2:  0x0000000000000040  x3:  0x000000000000003f
	[00:00:00.073,000] <err> os: x4:  0x00000000fc7ec000  x5:  0x0000000000000174
	[00:00:00.080,000] <err> os: x6:  0x0000000000000004  x7:  0x0000000000000000
	[00:00:00.088,000] <err> os: x8:  0xaaaaaaaaaaaaaaaa  x9:  0xaaaaaaaaaaaaaaaa
	[00:00:00.096,000] <err> os: x10: 0xaaaaaaaaaaaaaaaa  x11: 0x0000000000000000
	[00:00:00.104,000] <err> os: x12: 0xaaaaaaaaaaaaaaaa  x13: 0xaaaaaaaaaaaaaaaa
	[00:00:00.112,000] <err> os: x14: 0xaaaaaaaaaaaaaaaa  x15: 0xaaaaaaaaaaaaaaaa
	[00:00:00.119,000] <err> os: x16: 0xaaaaaaaaaaaaaaaa  x17: 0xaaaaaaaaaaaaaaaa
	[00:00:00.127,000] <err> os: x18: 0xaaaaaaaaaaaaaaaa  lr:  0x00000000fbffc018
	[00:00:00.135,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
	[00:00:00.143,000] <err> os: Current thread: 0xfc014700 (main)
	[00:00:00.149,000] <err> os: Halting system
	
This appears to have started happening after this commit https://github.com/zephyrproject-rtos/zephyr/commit/c9b534c4ebca8c3d02bc7d783c79771f1409d096.

Crashes here: `z_phys_map` -> `arch_mem_map` -> `inv_dcache_after_map_helper` -> `arm64_dcache_range` -> `dc_ops("ivac", start_addr);`.

Maybe caused by this:

> DC IVAC, requires write permission or else a permission fault is generated

https://developer.arm.com/documentation/den0024/a/Caches/Cache-maintenance

carlocaione, 2023-11-03:
@xakep-amatop can you take care of this?

xakep-amatop, 2023-11-03:
> @xakep-amatop can you take care of this?

yep, I'll fix it

PR #64779 (2023-11-09): arch: arm64: avoid invalidating of RO mem after mem map
The Cortex ARM documentation states that the DC IVAC instruction requires write access permission to the virtual address (VA); otherwise, it may generate a permission fault.

Therefore, it is needed to avoid invalidating read-only memory after the memory map operation.

This issue has been produced by commit c9b534c.
This commit resolves the issue #64758.

Fixes: #64758
PR #65009 (2023-11-13): [Backport v3.5-branch] arch: arm64: avoid invalidating of RO mem after mem map
Backport c4ffadb0b61fb4c7e5d0cdb517bbd21ca6791484 from #64779.

Fixes: #64758