zephyrproject-rtos/zephyr #64763: [doc] Documentation issue in 'services/logging/index'

adleris, 2023-11-03:
This issue covers some grammatical fixes to the logging page, which will be addressed in the linked pull request.

I did notice that previous changes to the logging page removed the text about the `CONFIG_LOG_PROCESS_THREAD` Kconfig option. Was there a particular reason for that? If not, I am happy to make another change to add some text about using that option.

**Environment**

* Page: `services/logging/index`
* Version: main
* SHA-1: 503422b91e7bb3826231cea55602c7efe47ed4f9

kartben, 2023-11-03:
> I did notice that previous changes to the logging page removed the text about the `CONFIG_LOG_PROCESS_THREAD` Kconfig option.

I am confused by this statement? Your PR is actually proposing changes to the text, so it's there?

adleris, 2023-11-21:
I probably should have flagged the commit. I had been specifically thinking of [this change](https://github.com/zephyrproject-rtos/zephyr/commit/9a3d762ee21#diff-a667a25747e6112bf1322597ccb775675afe4fb28150733f948e20f3bc8e12fbL324). I think it was nice to have the additional mention that Zephyr can handle the processing thread for you rather than you needed to handle it yourself using `LOG_PROCESS()`.

PR #64764 (2023-11-21): doc: logging: Fix some grammar issues
Fixes some grammar issues in the documentation.

Fixes #64763.
closed 2023-11-21 by carlescufi