zephyrproject-rtos/zephyr #64766: Driver for ST LPS28DFW Pressure Sensor

jremmert-phytec-iot, 2023-11-03:
**Is your feature request related to a problem? Please describe.**
We need a driver to get pressure samples from the ST LPS28DFW pressure sensor.

**Describe the solution you'd like**
I plan to develop the driver, based on the ST HAL "stmemsc", similar to the existing LPS22hh driver. The driver could support I2C and I3C. SPI is not supported by the LPS28DFW sensor. Currently, I only have a board to test I2C.

**Describe alternatives you've considered**
- Use the ST HAL or not? When looking in recent driver contributions for ST Sensors I saw that most are build on top of the ST HAL.
- There are 11 different ST pressure sensors, all similar. Zephyr currently supports 3 of them (LPS22hh, LPS22hb, LPS25hb) via individual drivers. I was considering if a unified driver is feasible for this sensor family. Since the sensors are similar but not register compatible, I dropped this idea.

**Additional context**
Register comparison of some ST pressure sensors.
[image]

erwango, 2023-11-03:
@avisconti ^^

avisconti, 2023-11-06:
@jremmert-phytec-iot 

I think you can start from lps22df in #62028 (24a0a8e34349c46ced077a3e909dd85989b8080a), and substitute the APIs with the 28dfw one. I think it is better to replicate here.

jremmert-phytec-iot, 2023-11-06:
Thanks for the pointer @avisconti! Somehow I missed this open PR.

lps28dfw has no SPI but has dual full-scale. There is one extra register to implement the dual full-scale setting, otherwise they look register compatible. Once my sample hardware reached, I will test and open a PR.

avisconti, 2023-11-06:
I'm going to overwrite that PR anyway with one containing only lps22df in order to speed up the review/merging. I'm not changing the lps22df code though.

closed 2023-12-02 by jremmert-phytec-iot