zephyrproject-rtos/zephyr #64775: LwM2M client implicit IPSO object version reporting

mlasch, 2023-11-03:
IPSO object versions are not reported correctly according to the LwM2M specification ([1.0.2](https://www.openmobilealliance.org/release/LightweightM2M/V1_0_2-20180209-A/OMA-TS-LightweightM2M-V1_0_2-20180209-A.pdf) or [1.1.1](https://www.openmobilealliance.org/release/LightweightM2M/V1_1_1-20190617-A/HTML-Version/OMA-TS-LightweightM2M_Core-V1_1_1-20190617-A.html#7-2-0-72-Object-Versioning)). Unfortunately, there is some ambiguity in the LwM2M specification about how to correctly report object versions from the client to a server during registration.

This is my understanding, which is backed from discussions on the [OMA developer Github repository](https://github.com/OpenMobileAlliance/OMA_LwM2M_for_Developers/issues/561).

The IPSO object version can be reported explicitly in the registration message in CoRE link format with the `ver=` attribute. In order to optimize for message size, the client is allowed (MAY) to omit the version attribute. However for this to work, both server and client implementations have to have the same understanding on how to omit and reconstruct this information.

The specification distinguishes between core and non-core objects. Core objects are defined in [Appendix E](https://www.openmobilealliance.org/release/LightweightM2M/V1_1_1-20190617-A/HTML-Version/OMA-TS-LightweightM2M_Core-V1_1_1-20190617-A.html#13-0-Appendix-E-LwM2M-Objects-defined-by-OMA-Normative). This is implemented in Zephyr.

**Current behavior in Zephyr LwM2M client 1.0 and 1.1**
For core objects (e.g. server object)
- Client version 1.0 and core object version 1.0 -> do not report version
- Client version 1.0 and core object version 1.1 -> report version
- Client version 1.1 and core object version 1.0 -> report version
- Client version 1.1 and core object version 1.1 -> do not report version

Connectivity monitoring object behaves differently:
- Client version 1.1 and core object version 1.2 -> reports version (even if it shouldn't)

For non-core objects:

- Client version 1.0 and non-core object version 1.0 -> do not report version
- Client version 1.0 and non-core object version 1.1 -> report version
- Client version 1.1 and non-core object version 1.0 -> do not report version
- Client version 1.1 and non-core object version 1.1 -> report version

**Expected behavior**
- **For core objects:**
  Clients must respect the core objects according to their LwM2M client version and should omit the object version in the registration and discover message. Allowed core object and client version combinations are defined in the specification Appendix E.

- **For non-core objects**:
  Omit the object version if **1.0** report otherwise.

AFAIK, this is how [Leshan](https://github.com/eclipse-leshan/leshan) has implemented it and how [Wakaama](https://github.com/eclipse/wakaama) should also do it. 

**Impact**
Picky servers might reject messages because of object version mismatches.

rlubos, 2023-11-06:
Hello @mlasch, thanks for the report. After reading though spec/linked issues it seems that indeed the core object version reporting is broken for LwM2M 1.1  (since we cannot simply compare object version with LwM2M version, which worked for LwM2M 1.0).

To clarify things - you refer to IPSO objects, is that really what you meant, or did you mean LwM2M objects in general? IPSO objects is just a subset of non-core objects, registered by IPSO alliance.

Anyway, to summarize:
* for **non-core** objects (IPSO etc.), it seems that no action is needed? The current behavior described by you does not deviate from the expected behavior. The code responsible for decision for non-core objects is pretty straightforward: https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/lwm2m/lwm2m_registry.c#L1983
* for **core** objects, we cannot simply rely on the LwM2M version to make a decision whether to include version or not. I think we should introduce a table with a default object version based on Appendix E of the respective LwM2M specification, and compare against it instead. I'll try to propose something.

FYI @SeppoTakalo

rlubos, 2023-11-06:
@mlasch I've opened https://github.com/zephyrproject-rtos/zephyr/pull/64870 with fix propsal, briefly tested with Leshan and it seems object versions are reported correctly now, please have a look.

mlasch, 2023-11-06:
> To clarify things - you refer to IPSO objects, is that really what you meant, or did you mean LwM2M objects in general? IPSO objects is just a subset of non-core objects, registered by IPSO alliance.

Thanks for pointing this out. You are right, I wasn't very precise. I meant LwM2M objects in general.

mlasch, 2023-11-06:
> @mlasch I've opened https://github.com/zephyrproject-rtos/zephyr/pull/64870 with fix propsal, briefly tested with Leshan and it seems object versions are reported correctly now, please have a look.

Tested and works for me 👍🏼

PR #64870 (2023-11-13): net: lwm2m: Fix core objects version reporting
Core objects version reporting was broken for LwM2M version 1.1, as the default object version not necessarily matches the LwM2M version. Therefore, implement a table with default object versions for particular LwM2M version, which can be looked up when determining whether it's needed to include object version or not during Registration/Discovery.

Fixes #64775
closed 2023-11-13 by carlescufi