zephyrproject-rtos/zephyr #64784: NXP: hal:

ithinuel, 2023-11-03:
**Describe the bug**
A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

dleach02, 2024-01-09:
@ithinuel  can you post the actual error? When I run the build for tests/lib/cpp/cxx I get an error on the linker step where something doesn't fit.

ithinuel, 2024-01-09:
Using `west twister -n -p warp7_m4 -T tests/lib/cpp/cxx` on 

```diff
diff --git a/tests/lib/cpp/cxx/src/main.cpp b/tests/lib/cpp/cxx/src/main.cpp
index 43bc14f65a..a31ea7299f 100644
--- a/tests/lib/cpp/cxx/src/main.cpp
+++ b/tests/lib/cpp/cxx/src/main.cpp
@@ -85,6 +85,8 @@

 #include <zephyr/ztest.h>

+#include <soc.h>
+
 class foo_class {
 public:
        foo_class(int foo) : foo(foo) {}
```

[result.txt](https://github.com/zephyrproject-rtos/zephyr/files/13877136/result.txt)
[build.log](https://github.com/zephyrproject-rtos/zephyr/files/13877145/build.log)
[build.log](https://github.com/zephyrproject-rtos/zephyr/files/13877148/build.log)

dleach02, 2024-02-02:
@ithinuel can you provide a PR to the HAL_NXP with your recommended fix?

ithinuel, 2024-02-02:
Unfortunately no, I do not have the expertise of C++ to do that.
This is why I reported this as an issue rather than creating a PR to fix it directly 🙁

jlh-makeen, 2024-04-16:
Experiencing the same issue, prior to 

To reproduce:
```
cd ~/zephyrproject/zephyr/samples/cpp/cpp_synchronization/
west build -p always -b colibri_imx7/mcimx7d/m4
```

Was compiling at commit (last time of my fork rebasin):
2b549daabc9d2cddda3e22c6ec1f5b27ea1997ad

If someone knows what happened to the build config files or toolchain in the meantime that would probably help solving the issue.

JiafeiPan, 2024-04-28:
Hi, @ithinuel and @jlh-makeen, I created one PR to fix this issue: https://github.com/zephyrproject-rtos/zephyr/pull/72047
But I have no imx7d board, so would you please help to verify the patch works although this PR has fixed building errors, but let's make sure it actually works on real board. thanks.

decsny, 2024-06-04:
it looks like this issue was not linked to the PR that fixed it, I believe it was fixed, reopen if that is not the case

closed 2024-06-04 by decsny