zephyrproject-rtos/zephyr #64793: using both retram and mcusram for code on stm32mp1 platforms

shlomow, 2023-11-03:
Currently, zephyr uses retram for code for stm32mp1 platforms, which is limited to 64 KB.
Is there a way to add also mcusram for code?

erwango, 2023-11-06:
^^@arnopo

arnopo, 2023-11-06:
@shlomow 
The main issue is that the RETRAM and the MCU SRAM are not contigious
RETRAM @ address 0, size 64 kB
MCU SRAM address 0x10000000 (aliased at 0x30000000) size 384 kB

The 2nd constraint is that the Cortex-M4 boots at address 0 of the RETRAM
so a

shlomow, 2023-11-06:
Thanks @arnopo!
Can you please share where in zephyr linker scripts I need to make this change?

arnopo, 2023-11-06:
I created a POC a long time ago. That define the vectore table in the RETRAM and then use only MCU SRAM for code and data. Perhaps it could inspire you. https://github.com/arnopo/zephyr_stm32mp1/commits/linker_poc
This POC is probably not up to date but I have nothing more recent.

Another approach is to use --enable-non-contiguous-regions option that has been introduced in GNU GCC to support of non-contiguous regions.
I did not test it but should help to split code between RETRAM and MCU SRAM:
https://manpages.ubuntu.com/manpages/jammy/man1/ld.1.html#:~:text=%2D%2Denable%2Dnon%2Dcontiguous%2D,output%20section%20is%20large%20enough.

shlomow, 2023-11-12:
Thanks @arnopo! Your branch from 2019 does exactly the changes I need.
The only change I made is renaming `_image_rom_start` to `__rom_region_start`.
Thanks again!

closed 2023-11-12 by shlomow

Peyton-McKee, 2025-06-12:
@mbober1 were you able to get @andrmuel solution working, I've tried and cant seem to get the IRQ handler to start.

mbober1, 2025-06-13:
I haven't spent much time testing this but it seems to work. You can try out my [code](https://github.com/mbober1/STM32MP1_ZephyrOS)