zephyrproject-rtos/zephyr #64811: esp32_devkitc_wrover board error

knagendrababu, 2023-11-04:
I have encountered an issue with the Zephyr ESP32 support page. After following the provided steps [esp32_devkitc_wrover](https://docs.zephyrproject.org/latest/boards/xtensa/esp32_devkitc_wrover/doc/index.html) and choosing the 'esp32_devkitc_wroover' board, I encountered an 'Invalid Board' error during the build process. It's worth noting that I have successfully built projects with 'nucleo_f446re' and 'esp32' without any issues.

I would appreciate any assistance in resolving this matter and understanding why the 'esp32_devkitc_wroover' option is not appearing as expected.

Command : west build -b esp32_devkitc_wrover --sysbuild samples/hello_world -p 

Error: 
xtensa:
  esp32
  esp32_ethernet_kit
  esp32_net
  esp32s2_franzininho
  esp32s2_saola
  esp32s3_devkitm
  esp_wrover_kit
  heltec_wifi_lora32_v2
  intel_adsp_ace15_mtpm
  intel_adsp_ace20_lnl
  intel_adsp_cavs25
  intel_adsp_cavs25_tgph
  m5stickc_plus
  nxp_adsp_imx8
  nxp_adsp_imx8m
  nxp_adsp_imx8x
  odroid_go
  olimex_esp32_evb
  qemu_xtensa
  xiao_esp32s3
  xt-sim
CMake Error at /home/nagendra/zephyrproject/zephyr/cmake/modules/boards.cmake:163 (message):
  Invalid BOARD; see above.
Call Stack (most recent call first):
  /home/nagendra/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:75 (include)
  /home/nagendra/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:23 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/nagendra/zephyrproject/zephyr/build -GNinja -DBOARD=esp32_devkitc_wrover -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -S/home/nagendra/zephyrproject/zephyr/share/sysbuild -DAPP_DIR:PATH=/home/nagendra/zephyrproject/zephyr/samples/hello_world

henrikbrixandersen, 2023-11-04:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

We especially need to know which version of Zephyr, you are using.

sylvioalves, 2023-11-05:
@knagendrababu, which Zephyr revision you have been working with?  `esp32_devkitc_wrover` is available only in v3.5.0 revision. Before that, I think it is just `esp32`or `esp_wrover_kit`.

knagendrababu, 2023-11-08:
It is 3.4.99
~/zephyrproject/zephyr$ **cat VERSION**
VERSION_MAJOR = 3
VERSION_MINOR = 4
PATCHLEVEL = 99
VERSION_TWEAK = 0
EXTRAVERSION =

sylvioalves, 2023-11-08:
Would you try compiling for board `esp_wrover_kit`?

knagendrababu, 2023-11-08:
Fresh installation working well.
**west build -b esp32_devkitc_wrover --sysbuild samples/hello_world**
Understand that the **esp32_devkitc_wrover** board was added in the **Zephyr 3.5 version** only.

@sylvioalves Thanks for your support

sylvioalves, 2023-11-08:
Great it worked. Feel free to ask.

closed 2023-11-08 by sylvioalves