zephyrproject-rtos/zephyr #64819: ESP32-devkitc-v4  ASSERTION FAIL [(cfg->port_pin_mask & (gpio_port_pins_t)(1UL << (pin))) != 0U] @ WEST_TOPDIR/zephyr/include/zephyr/drivers/gpio.h:1001

nefen, 2023-11-05:
firmware is build and flashed succesfully.
I tried to build blinky sample. It does not support ESP32-Wroom, but I made the minimum adaptations.
Assertion fails with a message for unsupported pin.

Board: ESP32_devkitC_v4 with chip ESP32_WROOM_32D

esp32_devkitc_wroom.overlay:
```
/ {
	aliases {
		led0 = &led0;
	};
    leds {
		compatible = "gpio-leds";
		led0: led_0 {
			gpios = <&gpio0 25 GPIO_ACTIVE_HIGH>;
		};
	};
};
```

main.c:
```
#include` <zephyr/kernel.h>
#include <zephyr/drivers/gpio.h>

#define SLEEP_TIME_MS   1000
#define LED0_NODE DT_ALIAS(led0)
static const struct gpio_dt_spec led = GPIO_DT_SPEC_GET(LED0_NODE, gpios);
int main(void)
{
	int ret;
	if (!gpio_is_ready_dt(&led)) {
		printk("Device is not ready\n");		
		return 0;
	}
	printk("Device is ready\n");		
	ret = gpio_pin_configure_dt(&led, GPIO_OUTPUT_ACTIVE);
	if (ret < 0) {
		printk("Device cannot be configured\n");		

		return 0;
	}
	printk("Device configured!\n");		
	while (1) {
		k_msleep(SLEEP_TIME_MS);
		printk("I am awake\n");
	}
	return 0;
}
```
CMakeLists.txt:
```
Set(BOARD esp32_devkitc_wroom)

cmake_minimum_required(VERSION 3.20.0)
find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(blinky)

target_sources(app PRIVATE src/main.c)
```
prj.conf:
```
CONFIG_GPIO=y
CONFIG_PRINTK=y
CONFIG_ASSERT=y
```

**To Reproduce**
west build
west flash!

**Logs and console output**
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6604
ho 0 tail 12 room 4
load:0x40078000,len:15224
ho 0 tail 12 room 4
load:0x40080400,len:3992
entry 0x40080634
I (30) boot: ESP-IDF 31fc5758f 2nd stage bootloader
I (30) boot: compile time 08:36:30
I (30) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=00000020 size=0001ch (    28)
I (113) esp_image: segment 1: paddr=00010044 vaddr=3ffb0000 size=002d0h (   720) load
I (121) esp_image: segment 2: paddr=0001031c vaddr=3ffb02d0 size=0013ch (   316) load
I (130) esp_image: segment 3: paddr=00010460 vaddr=40080000 size=03810h ( 14352) load
I (144) esp_image: segment 4: paddr=00013c78 vaddr=00000000 size=0c3c0h ( 50112)
I (164) esp_image: segment 5: paddr=00020040 vaddr=3f400040 size=01dc0h (  7616) map
I (167) esp_image: segment 6: paddr=00021e08 vaddr=00000000 size=0e210h ( 57872)
I (191) esp_image: segment 7: paddr=00030020 vaddr=400d0020 size=03bd8h ( 15320) map
I (198) boot: Loaded app from partition at offset 0x10000
I (199) boot: Disabling RNG early entropy source...
*** Booting Zephyr OS build zephyr-v3.5.0-232-g7a27431a0c63 ***
Device is ready
ASSERTION FAIL [(cfg->port_pin_mask & (gpio_port_pins_t)(1UL << (pin))) != 0U] @ WEST_TOPDIR/zephyr/include/zephyr/drivers/gpio.h:1001
        Unsupported pin

Backtrace:0x40080b20:0x3ffe65b0 0x400d097d:0x3ffe65c0 0x400d0475:0x3ffe65e0 0x400812d3:0x3ffe6600 0x400d04d7:0x3ffe6620
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain Zephyr SDK 0.16.3

sylvioalves, 2023-11-05:
@nefen, I've just tested and it works. IO25 is avaliable as GPIO. GPIO24 fails as you mentioned above. Can you make sure that GPIO25 is indeed used?  
Have you added the `esp32_devkitc_wroom.overlay` in a `boards\` folder?

nefen, 2023-11-06:
@sylvioalves, thanks for your effort!
1. For me it does not work not even for GPIO25. 
2. Overlay file is in the root folder not in the boards directory and it is well read: 
-- Found devicetree overlay: C:/Users/nefen/zephyrproject/develop/f063_palatisia/blinky_vs/esp32_devkitc_wroom.overlay
3. I tried to debug the error:

**_ASSERTION FAIL [(cfg->port_pin_mask & (gpio_port_pins_t)(1UL << (pin))) != 0U] @ WEST_TOPDIR/zephyr/include/zephyr/drivers/gpio.h:1001_**

So, I just added the following code ( I also added a handle for device sw0 just for comparison):

```
      const struct gpio_driver_config *const led_cfg = (const struct gpio_driver_config *)led.port->config;
      const struct gpio_driver_config *const but_cfg = (const struct gpio_driver_config *)but.port->config;
      printk(" led port_pin_mask: %ld, but port_pin_mask: %ld\n",led_cfg->port_pin_mask, but_cfg->port_pin_mask);
```

Off course, port pin mask was the same for led and but:
 led port_pin_mask: 917567, but port_pin_mask: 917567
The problem is that 971567 = 1110 0000 0000 0011 1111
So for assertion to pass, the pin number should be smaller than 19! 

It looks like zephyr does not follow the prescribed:
```
gpio0: gpio@3ff44000 {
				compatible = "espressif,esp32-gpio";
				gpio-controller;
				#gpio-cells = < 0x2 >;
				reg = < 0x3ff44000 0x800 >;
				interrupts = < 0x16 >;
				interrupt-parent = < &intc >;
				ngpios = < 0x20 >;
				gpio-reserved-ranges = < 0x6 0xb >, < 0x14 >, < 0x18 >, < 0x1c 0x1f >;
				status = "okay";
				phandle = < 0x7 >;
			};

```
According to the above, I expected the port pin mask to be something like this:
1111 1100 0011 1011 1010 0000 0011 1111
 and not             1110 0000 0000 0011 1111

Anyhow, off course, if a pick for port number any bit that it is 1 in the port pin maskunsigned int, the fimware works without the Assertion error.
Ie for: 			gpios = <&gpio0 0x12 0x1>;
everything is fine
_*** Booting Zephyr OS build zephyr-v3.5.0-232-g7a27431a0c63 ***
 led port_pin_mask: 917567, but port_pin_mask: 917567
Device is ready
Device configured!_

But it looks like a bug is somewhere.

I made a little more research:

I also added a port on gpio1 (led1: led_1 {gpios = <&gpio1 0x1 0x1>;status="okay";	};
The port pin mask for gpio1 is 0011 1111.
This is exactly according to: gpio1 configuration:
				ngpios = < 0x8 >;
				gpio-reserved-ranges = < 0x6 >, < 0x7 >;

nefen, 2023-11-06:
I searched dts and dtsi definitions. Main reservations of pins is declared in esp32_common.dtsi.

Other reserved ports I couldn't find. So from pins 0..31, if we remove those reserved we get the available pins as stated in row "port pin mask".

Accordig to my debugging, shown on row "917567", these are the pins that zephyr believes are available. Pins 12..16, 21..23, and 25..27 are not available, although they should.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/nefen/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/nefen/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

  |   | 31 | 30 | 29 | 28 | 27 | 26 | 25 | 24 | 23 | 22 | 21 | 20 | 19 | 18 | 17 | 16 | 15 | 14 | 13 | 12 | 11 | 10 | 9 | 8 | 7 | 6 | 5 | 4 | 3 | 2 | 1 | 0
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
esp32_common.dtsi reserved |   | 1 | 1 | 1 | 1 |   |   |   | 1 |   |   |   | 1 |   |   |   |   |   |   |   |   | 1 | 1 | 1 | 1 | 1 | 1 |   |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
  |   | 1 | 1 | 1 | 1 | 0 | 0 | 0 | 1 | 0 | 0 | 0 | 1 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 1 | 1 | 1 | 1 | 1 | 1 | 0 | 0 | 0 | 0 | 0 | 0
port pin mask |   | 0 | 0 | 0 | 0 | 1 | 1 | 1 | 0 | 1 | 1 | 1 | 0 | 1 | 1 | 1 | 1 | 1 | 1 | 1 | 1 | 0 | 0 | 0 | 0 | 0 | 0 | 1 | 1 | 1 | 1 | 1 | 1
  |   |   |   |   |   |   |   |   |   |   | i2c0 | i2c0 |   |   |   |   | spim2 | spim2 | spim2 | spim2 |   |   |   |   |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
pins present on module board |   |   |   |   |   | 1 | 1 | 1 |   | 1 | 1 | 1 |   | 1 | 1 | 1 | 1 | 1 | 1 | 1 | 1 |   |   |   |   |   |   | 1 | 1 | 1 | 1 | 1 | 1
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   | U0RXD | U0TXD
917567 |   |   |   |   |   |   |   |   |   |   |   |   |   | 1 | 1 | 1 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 1 | 1 | 1 | 1 | 1 | 1

</body>

</html>

sylvioalves, 2023-11-06:
@nefen, I still don't get it. I've tested the same code as posted and I get no ASSERT failure using GPIO25. Humm, let me try another approach using blinkly sample instead of custom app..

nefen, 2023-11-07:
@sylvioalves,

to you test.a devkitc wroom 32d?

It looks like the problem lies somewhere in the compilation of the
devicetree of the specific board.

Can uou print pin port mask for gpio0?

sylvioalves, 2023-11-08:
@nefen, sure I did with `esp32_devkitc_wroom` (32D).  Here is my output using IO25 as your suggested overlay:
```c
*** Booting Zephyr OS build zephyr-v3.5.0-1144-g4df07224f8dc ***
Device is ready
Device configured!
 led port_pin_mask: 250605631
I am awake
I am awake
I am awake
```  
I am starting to wonder whether the environment could be the issue (Windows?). I just can't reproduce the issue.

nefen, 2023-11-09:
I know. It is hard to miss!
[image]

Except of the missing 6 pins (21, 22, 23, 25, 26, 27), which I can live with, might I face additional issues?

I had used this module several times programmed through ESP-IDF.

I am just too old to keep developping in two different RTOS and I need Zephyr for nRFs.

rftafas, 2023-11-13:
> might I face additional issues?

Yes. There are several specific for Rev1 compiler, linker and code changes and workarounds specific for Rev1 that we did not brought to Zephyr (and on our part, never will). It can behave randomly, fall intop invalid states and crash with no apparent reason (yet the reason is Rev1 Silicon). It becomes worse with XIP and PSRAM. Besides, there are security flaws in Rev1 already mitigated in Rev3.

You will be fine with ESP32 Rev3 or newer devices.

nefen, 2023-11-16:
@rftafas thank you for your detailed information.

closed 2024-01-30 by github-actions[bot]