zephyrproject-rtos/zephyr #64825: Allow robot scripts without renode

mkschreder, 2023-11-05:
**Is your feature request related to a problem? Please describe.**
It is common to run end to end tests against the device using robot framework. With the new better pytest integration it should be possible to allow robot scripts to run in device testing mode. Currently it seems that only renode is supported when running the robot script as part of robot test harness in twister. 

**Describe the solution you'd like**
Ability to specify a robot harness and run in in device testing mode. Having it have access to the same "dut" interface as is available in pytest. 

**Describe alternatives you've considered**
Running robot scripts manually which is not very practical since twister has better knowledge of the test environment and can provide for example executable path, tty etc.

PerMac, 2023-11-09:
Interesting request. Is it something that you would like to implement?

MP-StefanKraus, 2023-11-24:
Hey,

I have time and motivation to implement this!

Is there any opinion on backwards compatibility, in case it is not perfectly transparent that things changed?

mkschreder, 2023-11-26:
> Hey,
> 
> I have time and motivation to implement this!
> 
> Is there any opinion on backwards compatibility, in case it is not perfectly transparent that things changed?

No. I think limiting robot framework to just renode is quite unnecessary anyway. It should be able to orchestrate the hardware abstractions that are exposed to pytest through twister and work as automation that runs along side of the test on native_posix, qemu and on device testing.

nashif, 2023-12-15:
@PiotrZierhoffer  FYI