zephyrproject-rtos/zephyr #64828: Documentation on documentation generation may need some fixes or additions

dottspina, 2023-11-06:
Bellow are a couple of findings about Zephyr documentation generation. I'm willing to write a PR if it's worth it.

## Linux Requirements (Fedora)

The additional requirements for Fedora should also include the LaTeX packages bellow (Fedora 38):

```  
texlive-cmap texlive-fncychap texlive-wrapfig texlive-capt-of texlive-framed
texlive-upquote texlive-needspace texlive-tabulary texlive-parskip 
texlive-oberdiek texlive-sectsty texlive-xindy
```

They are required for generating the PDF version of the documentation (provided by `texlive-latex-extra` on Ubuntu). 

## RAM Usage

Unsurprisingly, generating the documentation requires quite a bit of RAM: may be the Zephyr documentation could warn users that building the documentation will use around 16 GB (more or less, PDF/HTML).

I've also found (RHEL 9, Fedora 38) that executing successively `make html` and `make pdf` (in any order) will run out of memory, as in a recursion which would never meet its stopping condition: `sphinx-build` processes are OOM-killed after exhausting my 48 GB of RAM plus 24 GB swap.
 
Zephyr documentation could warn users to run `make clean` between the `html` and `pdf` targets (this *solves* the OOM issue).

I don't think there's much more Zephyr can do about it (if this actually relates to some bug, this more likely happens in Doxygen or Sphinx).

Thanks.

nordicjm, 2023-11-06:
> Unsurprisingly, generating the documentation requires quite a bit of RAM: may be the Zephyr documentation could warn users that building the documentation will use around 16 GB (more or less, PDF/HTML).

It can be stated it uses a lot but the amount it uses depends on number of CPU cores, generally about 2GB per core, so in your case 16GB of RAM indicates 4 cores with hyperthreading (or equivalent). I don't think it's documented but when I have to build documentation, I edit the cmake file to change `-j auto` to `-j7` otherwise it fails due to RAM usage

kartben, 2023-11-06:
> ## Linux Requirements (Fedora)
> The additional requirements for Fedora should also include the LaTeX packages bellow (Fedora 38):
> 
> [...]
> 
> They are required for generating the PDF version of the documentation (provided by `texlive-latex-extra` on Ubuntu).

Wouldn't `texlive-collection-latexextra` be the equivalent package for Fedora? Looks like maybe `texlive-collection-latexrecommended` is needed too for bringing in cmap? At the risk of having people download quite a few extra packages, I'd rather we depend on such "meta" packages upfront to make this easier to maintain and minimize the risk of the instructions becoming stale.

dottspina, 2023-11-06:
> Wouldn't `texlive-collection-latexextra` be the equivalent package for Fedora? Looks like maybe `texlive-collection-latexrecommended` is needed too for bringing in cmap? 

Yes, indeed, `texlive-collection-latexextra` seems to provide Fedora with most of what `texlive-latex-extra` provides Ubuntu with. And what could remain (e.g. cmap) is provided by `texlive-collection-latexrecommended`.    

> At the risk of having people download quite a few extra packages, I'd rather we depend on such "meta" packages upfront to make this easier to maintain and minimize the risk of the instructions becoming stale.

I fully agree: I had to find the missing packages with trial and error, running `latexmk` then `dnf provides */foobar.sty`, so installing *a few extra packages* to *minimize the risk of the instructions becoming stale* really seems the good approach to me.

Thanks.

dottspina, 2023-11-07:
> It can be stated it uses a lot but the amount it uses depends on number of CPU cores, generally about 2GB per core, so in your case 16GB of RAM indicates 4 cores with hyperthreading (or equivalent).

That's the first thing I thought of, after being able to confirm that `sphinx-build` processes were OOM-killed: less processes should mean less RAM. So I tried `make -j1 pdf`, without luck:

- I could still see 12 (6 cores x hyperthreading) `sphinx-build` processes
- after the Doxygen stage, when "reading sources", about at 33%, most of these processes will take more that 4GB each, up to more than 8GB for the most insane, eventually exhausting 48 GB RAM plus 24 GB swap

Again, this happens only if successively running `make html` then `make pdf`, in any order, **without** `make clean` in between: cleaning the `_build` directory one way or another, between these build targets, seems necessary to prevent Sphinx loosing itself I don't know where. And when everything is fine, as you said, the `sphinx-build` processes won't eat more than 2G each (even less).

>I don't think it's documented but when I have to build documentation, I edit the cmake file to change `-j auto` to `-j7` otherwise it fails due to RAM usage

Do you mean something like this ?

``` diff
--- a/doc/CMakeLists.txt
+++ b/doc/CMakeLists.txt
@@ -12,7 +12,7 @@ message(STATUS "Zephyr base: ${ZEPHYR_BASE}")
 #-------------------------------------------------------------------------------
 # Options

-set(SPHINXOPTS "-j auto -W --keep-going -T" CACHE STRING "Default Sphinx Options")
+set(SPHINXOPTS "-j1 -W --keep-going -T" CACHE STRING "Default Sphinx Options")
```

I've also tried, but still 12 `sphinx-build` processes: what am I missing ?

Thanks.

nordicjm, 2023-11-07:
Yes, but because you are using make you need to change `Makefile` instead, the cmake file is only if you use ninja.

closed 2024-05-27 by github-actions[bot]