zephyrproject-rtos/zephyr #64849: Windows building of ztest test cases does not work.

dabekjakub, 2023-11-06:
When trying to build a basic test case on Windows with west I get the following error:
```

C:\repos\converged>west build -p -b unit_testing -T zephyr\tests\unit\base64\utilities.base64 --build-dir twister_build
-- west build: generating a build system
-- The C compiler identification is GNU 11.3.0
-- The CXX compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/usr/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/msys64/usr/bin/c++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Loading Zephyr module(s) (Zephyr base): unittest
-- Found Python3: C:/Program Files/Python310/python.exe (found suitable version "3.10.8", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: C:/repos/converged/zephyr/.cache
-- Board: unit_testing
Parsing C:/repos/converged/zephyr/Kconfig
Loaded configuration 'C:/repos/converged/zephyr/subsys/testsuite/boards/unit_testing/unit_testing/unit_testing_defconfig'
Merged configuration 'C:/repos/converged/zephyr/tests/unit/base64/prj.conf'
Configuration saved to 'C:/repos/converged/twister_build/zephyr/.config'
Kconfig header saved to 'C:/repos/converged/twister_build/zephyr/include/generated/autoconf.h'
-- Found toolchain: host (gcc/ld)
-- Found GnuLd: C:/msys64/usr/bin/ld.bfd.exe (found version "2.37")
-- The ASM compiler identification is GNU
-- Found assembler: C:/msys64/usr/bin/gcc.exe
-- Configuring done
-- Generating done
-- Build files have been written to: C:/repos/converged/twister_build
←[92m-- west build: building application
[2/7] Building C object CMakeFiles/testbinary.dir/C_/repos/converged/zephyr/subsys/testsuite/ztest/src/ztest_new.c.obj
FAILED: CMakeFiles/testbinary.dir/C_/repos/converged/zephyr/subsys/testsuite/ztest/src/ztest_new.c.obj
C:\msys64\usr\bin\gcc.exe -DZTEST_UNITTEST -IC:/repos/converged/zephyr/subsys/testsuite/ztest/include/zephyr -IC:/repos/converged/zephyr/subsys/testsuite/include/zephyr -IC:/repos/converged/zephyr/subsys/testsuite/ztest/include -IC:/repos/converged/zephyr/subsys/testsuite/include -IC:/repos/converged/zephyr/include/zephyr -IC:/repos/converged/zephyr/include -IC:/repos/converged/zephyr/. -m32 -imacros C:/repos/converged/twister_build/zephyr/include/generated/autoconf.h -Wall -I C:/repos/converged/twister_build/zephyr/include/generated -MD -MT CMakeFiles/testbinary.dir/C_/repos/converged/zephyr/subsys/testsuite/ztest/src/ztest_new.c.obj -MF CMakeFiles\testbinary.dir\C_\repos\converged\zephyr\subsys\testsuite\ztest\src\ztest_new.c.obj.d -o CMakeFiles/testbinary.dir/C_/repos/converged/zephyr/subsys/testsuite/ztest/src/ztest_new.c.obj -c C:/repos/converged/zephyr/subsys/testsuite/ztest/src/ztest_new.c
<command-line>: fatal error: C:/repos/converged/twister_build/zephyr/include/generated/autoconf.h: No such file or directory
compilation terminated.

```
The generated file in question 
``` 
C:/repos/converged/twister_build/zephyr/include/generated/autoconf.h 
```

exists and is available to be opened.  

Tried different terminal options with and without privileges.
This seems to be either west or west extension issue.

I am not sure if this should end up here so please correct me if so.

fabiobaltieri, 2023-11-07:
Hey, tried to build that test in CI on Windows and it failed there as well, though in a completely different way: https://github.com/zephyrproject-rtos/zephyr/actions/runs/6789937592/job/18458222380?pr=64802. Failed on macOS as well.

Is this even expected to work with the native compiler? In your case:

```
-- Check for working C compiler: C:/msys64/usr/bin/cc.exe - skipped
```

in my CI run:

```
-- Check for working C compiler: C:/mingw64/bin/gcc.exe - skipped
```

@dabekjakub do you happen to know if it ever worked?

dabekjakub, 2023-11-08:
Do not know. 
However it seems to ignore the variables that designate the toolchain to use. I have gcc installed locally but use ZEPHYR_TOOLCHAIN_VARIANT=xt-clang for thesofprojec build with zephyr - which works. Have it set as environmental and is ignored for this build.

fabiobaltieri, 2023-11-08:
> Do not know. However it seems to ignore the variables that designate the toolchain to use. I have gcc installed locally but use ZEPHYR_TOOLCHAIN_VARIANT=xt-clang for thesofprojec build with zephyr - which works. Have it set as environmental and is ignored for this build.

Ok, I suspect unit testing does its own thing and not care about `ZEPHYR_TOOLCHAIN_VARIANT` since it's using the host tools, can you try setting `CC`? Like `CC=gcc-11 west build -p -b unit_testing tests/unit/base64` seems to work for me. I've no idea if this ever worked but I don't see why it would not.

@tejlmand do you have a setup to test this? I don't have a Windows machine (or any interest to get one :-), can tweak the experimental CI I've been working on to add a unit testing run I suppose if you folks manage to make it work but that's a fairly inconvenient setup to do troubleshooting.

fabiobaltieri, 2023-11-08:
@dleach02 maybe you could give it a try as well? `west build -p -b unit_testing tests/unit/base64`

kartben, 2023-11-09:
> Hey, tried to build that test in CI on Windows and it failed there as well, though in a completely different way: https://github.com/zephyrproject-rtos/zephyr/actions/runs/6789937592/job/18458222380?pr=64802. Failed on macOS as well.

I was able to reproduce your issue @fabiobaltieri. This fixes it: ` west build -p -b unit_testing tests/unit/base64 -- -DM64_MODE=1`

not quite sure about the root cause but this looks suspicious? https://github.com/zephyrproject-rtos/zephyr/blob/9bb44b8e5fc42d6d638660ef9d42edd11aa63623/cmake/modules/unittest.cmake#L70-L81

this M64_MODE flag seems to be set by some unit tests in their yaml file

kartben, 2023-11-09:
So yet another type of error from me, running the test with ` west build -p -b unit_testing tests/unit/base64 -- -DM64_MODE=1`

```
In file included from C:/Users/kartben/zephyrproject/zephyr/include/zephyr/toolchain.h:50,
                 from C:/Users/kartben/zephyrproject/zephyr/include/zephyr/sys/printk.h:11,
                 from C:/Users/kartben/zephyrproject/zephyr/subsys/testsuite/ztest/include/zephyr/ztest.h:38:
C:/Users/kartben/zephyrproject/zephyr/include/zephyr/kernel.h: At top level:
C:/Users/kartben/zephyrproject/zephyr/include/zephyr/toolchain/gcc.h:87:36: error: static assertion failed: ""
   87 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert(EXPR, "" MSG)
      |                                    ^~~~~~~~~~~~~~
C:/Users/kartben/zephyrproject/zephyr/include/zephyr/kernel.h:36:1: note: in expansion of macro 'BUILD_ASSERT'
   36 | BUILD_ASSERT(sizeof(intptr_t) == sizeof(long));
      | ^~~~~~~~~~~~
```

fabiobaltieri, 2023-11-09:
@kartben yeah I bumped into that setting as well, which seems to imply that unit test runs used to work on macOS, guess we broke it with the ZTEST conversion since mach-o binaries do not seem to like our custom sections, so that's another issue (or not, guess it depends whether we want to support that use case).

Let me try that M64_MODE in the github CI run see what happens.

fabiobaltieri, 2023-11-09:
Yeah same as yours I think: https://github.com/zephyrproject-rtos/zephyr/actions/runs/6810700736/job/18519520610?pr=64802

```
D:/a/zephyr/zephyr/zephyr/include/zephyr/kernel.h:36:1: note: in expansion of macro 'BUILD_ASSERT'
   36 | BUILD_ASSERT(sizeof(intptr_t) == sizeof(long));
      | ^~~~~~~~~~~~
```

Plus I found https://github.com/zephyrproject-rtos/zephyr/issues/53123 for macOS.

kartben, 2023-11-09:
Oh, interesting.
Commenting out the assert (probably a terrible idea?) gets things stuck further:

```
[7/7] Linking C executable testbinary.exe
FAILED: testbinary.exe
cmd.exe /C "cd . && C:\ProgramData\mingw64\mingw64\bin\gcc.exe -m64 -gdwarf-4 -T C:/Users/kartben/zephyrproject/zephyr/subsys/testsuite/include/zephyr/ztest_unittest.ld CMakeFiles/testbinary.dir/C_/Users/kartben/zephyrproject/zephyr/subsys/testsuite/ztest/src/ztest.c.obj CMakeFiles/testbinary.dir/C_/Users/kartben/zephyrproject/zephyr/subsys/testsuite/ztest/src/ztest_mock.c.obj CMakeFiles/testbinary.dir/C_/Users/kartben/zephyrproject/zephyr/subsys/testsuite/ztest/src/ztest_rules.c.obj CMakeFiles/testbinary.dir/C_/Users/kartben/zephyrproject/zephyr/subsys/testsuite/ztest/src/ztest_defaults.c.obj CMakeFiles/testbinary.dir/main.c.obj -o testbinary.exe -Wl,--out-implib,libtestbinary.dll.a -Wl,--major-image-version,0,--minor-image-version,0  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."
`_ztest_expected_result_entry_list_end' referenced in section `.rdata$.refptr._ztest_expected_result_entry_list_end[.refptr._ztest_expected_result_entry_list_end]' of CMakeFiles/testbinary.dir/C_/Users/kartben/zephyrproject/zephyr/subsys/testsuite/ztest/src/ztest.c.obj: defined in discarded section `.data.ztest_expected_result_area' of testbinary.exe
`_ztest_expected_result_entry_list_start' referenced in section `.rdata$.refptr._ztest_expected_result_entry_list_start[.refptr._ztest_expected_result_entry_list_start]' of CMakeFiles/testbinary.dir/C_/Users/kartben/zephyrproject/zephyr/subsys/testsuite/ztest/src/ztest.c.obj: defined in discarded section `.data.ztest_expected_result_area' of testbinary.exe
`_ztest_test_rule_list_end' referenced in section `.rdata$.refptr._ztest_test_rule_list_end[.refptr._ztest_test_rule_list_end]' of CMakeFiles/testbinary.dir/C_/Users/kartben/zephyrproject/zephyr/subsys/testsuite/ztest/src/ztest.c.obj: defined in discarded section `.data.ztest_test_rule_area' of testbinary.exe
`_ztest_test_rule_list_start' referenced in section `.rdata$.refptr._ztest_test_rule_list_start[.refptr._ztest_test_rule_list_start]' of CMakeFiles/testbinary.dir/C_/Users/kartben/zephyrproject/zephyr/subsys/testsuite/ztest/src/ztest.c.obj: defined in discarded section `.data.ztest_test_rule_area' of testbinary.exe
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\kartben\zephyrproject\zephyr\build'
```

fabiobaltieri, 2023-11-09:
@kartben ok now it's complaining that it does not know what to do with those sections, may  need some tweak to https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/testsuite/include/zephyr/ztest_unittest.ld (if it's using it at all but it probably is)

kartben, 2023-11-09:
> @kartben ok now it's complaining that it does not know what to do with those sections, may need some tweak to https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/testsuite/include/zephyr/ztest_unittest.ld (if it's using it at all but it probably is)
> 
> cc @carlescufi

Ya it is using it -- tried hacking it but didn't go very far. It seems like the KEEP sections for the iterables are pretty similar to what can be found in the regular linker scripts, so I'm not sure what's causing the issue.

fabiobaltieri, 2023-11-09:
@kartben can you check out how those ends up being named in the object files?

```console
$ gdb build/CMakeFiles/testbinary.dir/main.c.o
...
(gdb) info files 
...
        0x00001540 - 0x00001554 is ._ztest_unit_test.static.z_ztest_unit_test__lib_base64__test_base64_codec_
        0x00001554 - 0x00001570 is ._ztest_suite_node.static.z_ztest_test_node_lib_base64_
```

maybe they got messed up

kartben, 2023-11-09:
> @kartben can you check out how those ends up being named in the object files?
> 
> ```
> $ gdb build/CMakeFiles/testbinary.dir/main.c.o
> ...
> (gdb) info files 
> ...
>         0x00001540 - 0x00001554 is ._ztest_unit_test.static.z_ztest_unit_test__lib_base64__test_base64_codec_
>         0x00001554 - 0x00001570 is ._ztest_suite_node.static.z_ztest_test_node_lib_base64_
> ```
> 
> maybe they got messed up

```
 ~#@❯  gdb .\build\CMakeFiles\testbinary.dir\main.c.obj                                                                  ❮  9s 66ms   
GNU gdb (GDB) 11.2
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-w64-mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from .\build\CMakeFiles\testbinary.dir\main.c.obj...
(No debugging symbols found in .\build\CMakeFiles\testbinary.dir\main.c.obj)
(gdb) info files
Symbols from "C:\Users\kartben\zephyrproject\zephyr\build\CMakeFiles\testbinary.dir\main.c.obj".
Local exec file:
        `C:\Users\kartben\zephyrproject\zephyr\build\CMakeFiles\testbinary.dir\main.c.obj', file type pe-x86-64.
        Entry point: 0x0
        0x0000000000000000 - 0x00000000000010e0 is .text
        0x00000000000010e0 - 0x00000000000010e0 is .data
        0x00000000000010e0 - 0x0000000000001110 is .bss
        0x0000000000001120 - 0x0000000000001800 is .rdata
        0x0000000000001800 - 0x000000000000183c is .xdata
        0x000000000000183c - 0x0000000000001878 is .pdata
        0x0000000000001878 - 0x00000000000018a0 is ._ztest_unit_test.static.z_ztest_unit_test__lib_base64__test_base64_codec_
        0x00000000000018a0 - 0x00000000000018d8 is ._ztest_suite_node.static.z_ztest_test_node_lib_base64_
        0x00000000000018e0 - 0x0000000000001930 is .rdata$zzz
(gdb)
```

fabiobaltieri, 2023-11-09:
Oh wait

```
`_ztest_expected_result_entry_list_end' referenced in section `.rdata$.refptr._ztest_expected_result_entry_list_end[.refptr._ztest_expected_result_entry_list_end]' of CMakeFiles/testbinary.dir/C_/Users/kartben/zephyrproject/zephyr/subsys/testsuite/ztest/src/ztest.c.obj: defined in discarded section `.data.ztest_expected_result_area' of testbinary.exe
`_ztest_expected_result_entry_list_start' referenced in section `.rdata$.refptr._ztest_expected_result_entry_list_start[.refptr._ztest_expected_result_entry_list_start]' of CMakeFiles/testbinary.dir/C_/Users/kartben/zephyrproject/zephyr/subsys/testsuite/ztest/src/ztest.c.obj: defined in discarded section `.data.ztest_expected_result_area' of testbinary.exe
`_ztest_test_rule_list_end' referenced in section `.rdata$.refptr._ztest_test_rule_list_end[.refptr._ztest_test_rule_list_end]' of CMakeFiles/testbinary.dir/C_/Users/kartben/zephyrproject/zephyr/subsys/testsuite/ztest/src/ztest.c.obj: defined in discarded section `.data.ztest_test_rule_area' of testbinary.exe
`_ztest_test_rule_list_start' referenced in section `.rdata$.refptr._ztest_test_rule_list_start[.refptr._ztest_test_rule_list_start]' of CMakeFiles/testbinary.dir/C_/Users/kartben/zephyrproject/zephyr/subsys/testsuite/ztest/src/ztest.c.obj: defined in discarded section `.data.ztest_test_rule_area' of testbinary.exe
```

this is complaining about empty sections right? i.e. NOT the one above, how do we tell it to NOT discard the section?

fabiobaltieri, 2023-11-09:
```console
$ gdb build/testbinary
...
(gdb) info files 
...
        0x00004000 - 0x00004c7d is .rodata
        0x00007030 - 0x0000705c is .data
        0x0000705c - 0x00007078 is .data.ztest_suite_node_area
        0x00007078 - 0x0000708c is .data.ztest_unit_test_area
(gdb) p (void *)_ztest_expected_result_entry_list_start 
$1 = (void *) 0x48d9
(gdb) p (void *)_ztest_expected_result_entry_list_end 
$2 = (void *) 0x48d9
```

looks like here the section got ditched but it kept the symbols somewhere in `.data`

fabiobaltieri, 2023-11-09:
@kartben can you try this:

```diff
index 4a3e7d783c..f215a1c53f 100644
--- a/subsys/testsuite/include/zephyr/ztest_unittest.ld
+++ b/subsys/testsuite/include/zephyr/ztest_unittest.ld
@@ -8,24 +8,28 @@ SECTIONS
 {
        .data.ztest_expected_result_area : ALIGN(4)
        {
+               . = ALIGN(4);
                _ztest_expected_result_entry_list_start = .;
                KEEP(*(SORT_BY_NAME(._ztest_expected_result_entry.static.*)))
                _ztest_expected_result_entry_list_end = .;
        }
        .data.ztest_suite_node_area : ALIGN(4)
        {
+               . = ALIGN(4);
                _ztest_suite_node_list_start = .;
                KEEP(*(SORT_BY_NAME(._ztest_suite_node.static.*)))
                _ztest_suite_node_list_end = .;
        }
        .data.ztest_unit_test_area : ALIGN(4)
        {
+               . = ALIGN(4);
                _ztest_unit_test_list_start = .;
                KEEP(*(SORT_BY_NAME(._ztest_unit_test.static.*)))
                _ztest_unit_test_list_end = .;
        }
        .data.ztest_test_rule_area : ALIGN(4)
        {
+               . = ALIGN(4);
                _ztest_test_rule_list_start = .;
                KEEP(*(SORT_BY_NAME(._ztest_test_rule.static.*)))
                _ztest_test_rule_list_end = .;
```

https://sourceware.org/binutils/docs/ld/Output-Section-Discarding.html

kartben, 2023-11-09:
@fabiobaltieri This did the trick! Can't seem to run the executable though, so I suspect there still some 32 vs 64bit issues. 

```
 ~#@❯  .\build\testbinary.exe                                                                                           ❮  7s 873ms   
ResourceUnavailable: Program 'testbinary.exe' failed to run: An error occurred trying to start process 'C:\Users\kartben\zephyrproject\zephyr\build\testbinary.exe' with working directory 'C:\Users\kartben\zephyrproject\zephyr'. The specified executable is not a valid application for this OS platform.At line:1 char:1
+ .\build\testbinary.exe
```

fabiobaltieri, 2023-11-09:
Cool, more to troubleshoot I guess, btw I think `. = .;` would have worked as well.

Tried to hack something for macOS as well but bumped into this: https://github.com/zephyrproject-rtos/zephyr/issues/53123#issuecomment-1803707388, may be fixable for unit tests but probably not for `native_posix`.

tejlmand, 2023-11-09:
this is an interesting issue, and not something I would expect to work.

First part is this line:
```
-- Check for working C compiler: C:/msys64/usr/bin/cc.exe - skipped
```
indicating to me that msys is used, however support for msys was explicitly removed here:
https://github.com/zephyrproject-rtos/zephyr/pull/11300

But this was mainly when considering cross-compilation, which the `unit_testing` board / ztest is not.
The unit_testing board in this case is more similar to the `native_posix` target which is not supported on Windows, ref: https://github.com/zephyrproject-rtos/zephyr/issues/43737#issuecomment-1066524245
Also see the documentation for `native_posix`: https://docs.zephyrproject.org/2.6.0/boards/posix/native_posix/doc/index.html#overview

Until a solution can be found for the `unit_testing` board, then I believe we should improve our documentation regarding the `unit_testing` board.
Of course if a simple solution can be found for Windows, that would be best, but i'm not optimistic on this.

nashif, 2023-11-09:
> Until a solution can be found for the `unit_testing` board, then I believe we should improve our documentation regarding the `unit_testing` board.

there is no unit_testing board, this is just a dummy definition, it does build a very basic binary and is in no way comparable to native_posix. I do not remember this every working on windows however.

nashif, 2023-11-09:
ok, it used to work on mac and it broke at some point...

```
> uname
Darwin
> west build -b unit_testing tests/unit/list
> anashif@~/zephyr/zephyrproject/zephyr [v2.7-branch]> ./build/testbinary
Running test suite dlist
===================================================================
START - test_dlist
 PASS - test_dlist in 0.0 seconds
===================================================================
START - test_dlist2
 PASS - test_dlist2 in 0.0 seconds
===================================================================
START - test_slist
 PASS - test_slist in 0.0 seconds
===================================================================
START - test_sflist
 PASS - test_sflist in 0.0 seconds
===================================================================
Test suite dlist succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

nashif, 2023-11-09:
> Ztest, compiler sections, mach-o, 16 characters, [#53123 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/53123#issuecomment-1803716402).

yeah, emphasises the need for your multiplatform CI workflow. We could have caught this breakage soon enough :)

tejlmand, 2023-11-09:
> there is no unit_testing board, this is just a dummy definition,

I know, but we are generally not testing host compiler on Windows, only cross-compiler.
We do go through part of the same CMake infrastructure for toolchains in `unittest` as we do for a regular build, but  I (and probably others as well) haven't been verifying the host part on Windows.

Where I was  in doubt on previous comment was whether we actually picked up some header from native_posix for unittesting, but I just verified that we do not do that.

So that means we should easily be able to get the `unit_testing` / `ztest` to work on Windows.

dabekjakub, 2024-01-08:
Hey, thanks for the help in the issue. Had a little downtime with ztest but the comments were useful and got what I needed for Windows. Closing.

closed 2024-01-08 by dabekjakub