zephyrproject-rtos/zephyr #64863: Settings: settings_delete API may return -ENOSPC error when use with the NVS backend

kapi-no, 2023-11-06:
**Describe the bug**

The `settings_delete` API may return an `ENOSPC` error when the Settings storage partition is full. The issue appears with the Settings NVS backend (`CONFIG_SETTINGS_NVS`) and when you try to delete the largest name ID in use (stored under the `NVS_NAMECNT_ID` ID).

An `ENOSPC` is returned in the `settings_nvs_save` function that is implemented in the `zephyr/subsys/settings/src
/settings_nvs.c` file:

```
found:
	if (delete) {
		if (name_id == NVS_NAMECNT_ID) {
			return 0;
		}

		if (name_id == cf->last_name_id) {
			cf->last_name_id--;
			rc = nvs_write(&cf->cf_nvs, NVS_NAMECNT_ID,
				       &cf->last_name_id, sizeof(uint16_t));
			if (rc < 0) {
				/* Error: can't to store
				 * the largest name ID in use.
				 */
				return rc;
			}
		}

		(...)
```

Before the actual delete operation, the NVS backend implementation attempts to update its metadata counter of name IDs. The operation fails as the storage partition is full (`Error: can't to store the largest name ID in use.`).

**To Reproduce**

Steps to reproduce the behavior:
1. Fill the Settings storage partition so there is no more room for the update of the name ID counter (`NVS_NAMECNT_ID`).
2. Attempt to delete the Settings name with the `settings_delete` API that has the largest name ID (see `cf->last_name_id`).
3. Observe that the `settings_delete` API returns an `ENOSPC` error.

**Expected behavior**

The `settings_delete` API should not return an `ENOSPC` error as this function is used to free space on the storage partition.

With The NVS file system, it is always possible to remove data via the `nvs_delete` API. The file system prevents data from being written to the memory fragment that is used to perform the delete operation (it always keeps 8 bytes unwritten for the delete ATE). The Settings NVS backend should preserve this property and make it always possible to remove data with the `settings_delete` API.

**Impact**

Bricked device (most devices need NVM storage for their operations).

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.16.3
 - Commit SHA or Version used: https://github.com/zephyrproject-rtos/zephyr/commit/b3447ec068bbcffb5386122c162fb4afd74a83d8 (latest)

Laczen, 2023-11-06:
@kapi-no, a simple fix would be to move the writing of the last_name_id after the delete of the entry name and value.

kapi-no, 2023-11-07:
@Laczen, yes, we thought about this change. However, I am not an expert in Settings and the Settings NVS backend, and I want to avoid introducing regressions in other edge cases. I am wondering why the current order of operations was selected in the first place.

It seems like the current order is also dangerous from the perspective of the power down / reset. You may decrement `last_name_id` and not be fast enough to call `nvs_delete` APIs during the delete operation due to the power down. In this case, the Settings subsystem will not be aware of the undeleted entry, and it will persist in the storage partition.

@Laczen, I am happy to propose the change and swap the order of operations as you suggested. However, if there is a more knowledgeable person in Settings who can think this issue through and consider all the `NVS_NAMECNT_ID` metadata writes (also in the `settings_load` context), I think it is better that they implement the fix for this issue.

kapi-no, 2023-11-07:
@Laczen, I opened a PR with some general improvements for handling name ID metadata Please take a look:

https://github.com/zephyrproject-rtos/zephyr/pull/64907

closed 2023-11-22 by fabiobaltieri