zephyrproject-rtos/zephyr #64871: LE Audio: Add API to request Broadcast Code without sync

mkasenberg, 2023-11-06:
**Is your enhancement proposal related to a problem? Please describe.**
There is a mandatory PTS BASS/SR/CP/BV-21-C [Set Broadcast Code without Sync] test case (from [BASS test spec](https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=524502)) with test purpose:

"Verify that the BASS server IUT handles writing of the Set Broadcast Code Opcode to the Broadcast Audio Scan Control Point when it does not support Sync."

I have not found a public function in LE Audio API that would allow to set BIG_Encryption parameter of Broadcast Receive State characteristic to 0x01 (Broadcast_Code required) without syncing to PA first.

**Describe the solution you'd like**
A function that updates Broadcast Receive State characteristic value so the BIG_Encryption parameter indicates Broadcast_Code required.

**Describe alternatives you've considered**
Any other solution that will allow to achieve this.

Thalley, 2024-03-06:
> I have not found a public function in LE Audio API that would allow to set BIG_Encryption parameter of Broadcast Receive State characteristic to 0x01 (Broadcast_Code required) without syncing to PA first.

Is that not possible with the `bt_bap_scan_delegator_mod_src`?

mkasenberg, 2024-03-13:
> Is that not possible with the `bt_bap_scan_delegator_mod_src`?

It seems that `bt_bap_scan_delegator_add_src` or `bt_bap_scan_delegator_mod_src` require to have PA Sync created.

The BASS/SR/CP/BV-21-C represents kind of a border use case . This test case goes like this:

1. There are only 2 devices, BASS server device + some client device (PTS) waiting for a notification from the Broadcast Audio Scan Control Point. Throughout the test case, there is no broadcast source to sync to nearby.

2. BASS server device should act as it does not support PA Sync.

3. BASS server device should expose a Broadcast Receive State with BIG_Encryption field indicating Broadcast_Code required and send the notification. There is no info in the test case description how to fill other fields like broadcaster address, broadcast id, etc., only that they should be valid. So I assume the BASS server should use some cached values from the app layer.

4. After the notification, the client will send a Set Broadcast Code command and the BASS server shall respond. End of the test case.

Thalley, 2024-03-13:
> It seems that bt_bap_scan_delegator_add_src or bt_bap_scan_delegator_mod_src require to have PA Sync created.

Ah, yes, you are right that `bt_bap_scan_delegator_add_src` seems to require a PA sync. I guess there are no reason for that really, and `bt_bap_scan_delegator_add_src` needs to be updated to support adding a local receive state without a PA sync

Thalley, 2024-06-25:
@mkasenberg Should we consider this a bug?

closed 2024-07-28 by nashif