zephyrproject-rtos/zephyr #64926: Reusable Zephyr setup GitHub workflow + multiple platform CI testing

fabiobaltieri, 2023-11-07:
## Introduction

Hi, some times ago I've put together a [reusable github action](https://github.com/fabiobaltieri/zephyr-setup) to prepare a stock GitHub CI image to build Zephyr projects. It really started as an exercise to learn how to use the workflow, then to figure out how fast it would be to prepare a Zephyr environment for building a single application from scratch, and lastly I added multi-OS support as well.

The thing turns out to work pretty decently and I'm thinking it may be nice for the project to support it officially, to facilitate running a CI on GitHub for end users.

### Proposed change

The first proposal is to move over the project under the Zephyr GitHub organization: that is creating a new repository for it under the org and then move over the code from https://github.com/fabiobaltieri/zephyr-setup.

After that I'm thinking we could reuse it for testing:
- in our example application: https://github.com/zephyrproject-rtos/example-application/pull/49, to show an actual example usage, it also happen to be faster than the current container-based solution
- in the main project CI, to test a build on all supported platforms (Linux, macOS and Windows, all SDKs): https://github.com/zephyrproject-rtos/zephyr/pull/64802

### Dependencies

The workflow right now has an optional property for specifying the SDK version, I think it'd be nice to be able to pull a "latest" one automatically that pulls whatever was currently used in the Zephyr checkout. This could be done by having the version in the main repository in a known file, so I'd like to propose to do that as well (something like dumping a `.github/current-sdk.txt` with `0.16.3` or something like that).

fabiobaltieri, 2023-11-07:
cc @kartben @nashif @stephanosio @carlescufi @gmarull

fabiobaltieri, 2023-11-23:
@carlescufi put this up for Arch WG. Feel free to put it at the bottom of the list, it's a bit of a side project of mine, don't have much time to do major rework at the moment but if we are happy with the current state and some minor tweaks we can go with it.

I think at this stage the request is to create the org repo and move the code there.

kartben, 2023-11-23:
As a random note, it's been working pretty smoothly for me since I added it to my artificial nose project, building with a couple different toolchains and all. https://github.com/kartben/artificial-nose-zephyr/actions

stephanosio, 2024-01-11:
@fabiobaltieri https://github.com/zephyrproject-rtos/action-zephyr-setup has been created.

fabiobaltieri, 2024-01-11:
Thanks @stephanosio, I've done an initial empty push and opened https://github.com/zephyrproject-rtos/action-zephyr-setup/pull/1 to import the files from my original repo.

closed 2024-01-31 by fabiobaltieri